openbmc/openbmc #2205: Unbind/Bind power supply driver on power supply remove/replace

spinler, 2017-08-24:
There is a phosphor-hwmon instance for the power supplies, so if the PS is removed and the driver isn't unbound (to stop phosphor-hwmon), the process will crash (I think).   Also, the power-supply-monitor code needs this too so it can talk to PSs that were added after the initial probe.

bjwyman, 2017-08-24:
Command used to manually bind driver after reinstalling the (top) power supply:
```echo 3-0068 > /sys/bus/i2c/drivers/p17531/bind```

spinler, 2017-08-31:
Task List
- [ ] Update phosphor-gpio-presence to take an optional 'drivers' argument with list of devices to bind/unbind when cards go present/missing
- [ ] When the presence state changes, do the binds/unbinds
- [ ] Update the systemd env files to pass in the power supply driver information
- [ ] Test

spinler, 2017-08-31:
# **STATUS**
- 08/31:

rfrandse, 2017-09-05:
<https://gerrit.openbmc-project.xyz/6410> Pass driver names to gpio-presence service
Resolves  #2205 Unbind/Bind power supply driver on power supply remove/replace

closed 2017-09-12 by williamspatrick