openbmc/openbmc #2208: Implement support for reset tolerance in Aspeed GPIO controller

amboar, 2017-08-25:
Reset tolerance holds the state of individual GPIO pins across SoC resets. This is essential for "reset/reload" support, and for maintaining LED state in some systems.

geissonator, 2017-09-29:
We good on this one @amboar ?

amboar, 2017-09-29:
I started looking at it, but it's not yet implemented

amboar, 2017-10-23:
I've sent patches upstream which have received some feedback: http://patchwork.ozlabs.org/project/openbmc/list/?series=9241&state=%2A&archive=both

I'll need to rework them, but interested parties might want to jump on and give their opinion on whether or not we'd like to have the reset tolerance be configurable from userspace for userspace-exported GPIOs. @bradbishop @shenki

amboar, 2018-04-16:
Implementation is upstream and also applied to the openbmc kernel

closed 2018-04-16 by amboar