openbmc/openbmc #2210: phosphor-host-state-manager is incorrectly requesting it's own state change on bmc reset

geissonator, 2017-08-25:
Not quite sure what's going on here yet but for some reason the host-state-manager code is requesting it's own state change and messing up the bmc r/r discovery at times:
```
Aug 24 19:51:06 witherspoon systemd[1]: Starting Check Host0 status on BMC reset...
Aug 24 19:51:06 witherspoon phosphor-host-check[1429]: Check if host is running
Aug 24 19:51:06 witherspoon mboxctl[1417]: Reset: Success
Aug 24 19:51:06 witherspoon systemd[1]: Started Reload mboxd during power on.
Aug 24 19:51:07 witherspoon systemd[1]: Started mapper subtree-remove /xyz/openbmc_project/software:xyz.openbmc_project.Software.ActivationBlocksTransition.
Aug 24 19:51:07 witherspoon ipmid[1382]: Pushing cmd on to queue
Aug 24 19:51:07 witherspoon ipmid[1382]: Pushing cmd on to queue
Aug 24 19:51:07 witherspoon ipmid[1382]: Asserting SMS Attention
Aug 24 19:51:07 witherspoon systemd[1]: Started Phosphor Time Manager daemon.
Aug 24 19:51:08 witherspoon phosphor-ledmanager[1193]: Asserting LEDs
Aug 24 19:51:08 witherspoon systemd[1]: Started Assert power_on LED.
Aug 24 19:51:08 witherspoon ipmid[1382]: SMS Attention asserted
Aug 24 19:51:08 witherspoon phosphor-host-check[1429]: Host control signal values
Aug 24 19:51:08 witherspoon phosphor-host-check[1429]: Host is running!
Aug 24 19:51:08 witherspoon systemd[1]: Started Check Host0 status on BMC reset.
Aug 24 19:51:08 witherspoon systemd[1]: Reached target Host0 running after reset.
Aug 24 19:51:08 witherspoon systemd[1]: Started Start host0 running after BMC reset.
Aug 24 19:51:08 witherspoon systemd[1]: Reached target Host0 (Reset Check).
Aug 24 19:51:08 witherspoon systemd[1]: Starting Phosphor Host State Manager...
Aug 24 19:51:08 witherspoon systemd[1]: Started Phosphor Cooling Type.
Aug 24 19:51:08 witherspoon systemd[1]: Starting Wait for /xyz/openbmc_project/watchdog/host0...
Aug 24 19:51:08 witherspoon systemd[1]: Created slice system-obmc\x2denable\x2dhost\x2dwatchdog.slice.
Aug 24 19:51:08 witherspoon systemd[1]: Created slice system-start_host.slice.
Aug 24 19:51:08 witherspoon systemd[1]: Starting mapper subtree-remove /xyz/openbmc_project/software:xyz.openbmc_project.Software.ActivationBlocksTransition...
Aug 24 19:51:08 witherspoon systemd[1]: Created slice system-phosphor\x2dgpio\x2dmonitor.slice.
Aug 24 19:51:08 witherspoon systemd[1]: Starting Wait for /org/openbmc/control/chassis0...
Aug 24 19:51:08 witherspoon systemd[1]: Created slice system-pcie\x2dslot\x2ddetect.slice.
Aug 24 19:51:09 witherspoon systemd[1]: Started Phosphor GPIO checkstop monitor.
Aug 24 19:51:09 witherspoon systemd[1]: Created slice system-phosphor\x2dwatchdog.slice.
Aug 24 19:51:09 witherspoon systemd[1]: Started Phosphor poweron watchdog.
Aug 24 19:51:10 witherspoon systemd[1]: Started Wait for /org/openbmc/control/chassis0.
Aug 24 19:51:10 witherspoon systemd[1]: Started mapper subtree-remove /xyz/openbmc_project/software:xyz.openbmc_project.Software.ActivationBlocksTransition.
Aug 24 19:51:11 witherspoon phosphor-host-state-manager[1439]: Initial Host State will be Off
Aug 24 19:51:11 witherspoon phosphor-host-state-manager[1439]: Host State transaction request
```

geissonator, 2017-08-25:
So, what I did to recreate this was the following:

Boot system
Request a host reboot via the host-state-manager interface
Once host is back up, issue a bmc reboot

The problem is that the host reboot request is persisted by phosphor-host-state-manager so when the BMC comes back from it's reboot, it loads that last requested state, and then executes it.  I have an email on the mailing list with a proposal on fixing this.

geissonator, 2017-08-29:
# **STATUS**
- 8/28: Emailed mailing list and made some initial code changes but not as easy as I'd hoped

rfrandse, 2017-09-05:
<https://gerrit.openbmc-project.xyz/6325> Move all restore state policy into discover_state
Resolves  #2210 phosphor-host-state-manager is incorrectly requesting it's own state change on bmc reset

closed 2017-09-08 by williamspatrick