openbmc/openbmc #2214: phosphor-dump-monitor[955]: Error in mapper call

gkeishin, 2017-08-25:
New master won't boot to standby
- [x] Update BMC to new Master build

State Host service also doesn't seems up
```
/xyz/openbmc_project/state/enumerate
{
    "/xyz/openbmc_project/state/bmc0": {
        "CurrentBMCState": "xyz.openbmc_project.State.BMC.BMCState.NotReady",
        "RequestedBMCTransition": "xyz.openbmc_project.State.BMC.Transition.None"
    },
    "/xyz/openbmc_project/state/chassis0": {
        "CurrentPowerState": "xyz.openbmc_project.State.Chassis.PowerState.Off",
        "RequestedPowerTransition": "xyz.openbmc_project.State.Chassis.Transition.Off"
    }
}
```

```
root@witherspoon:~# journalctl --no-pager | grep -i ERROR
Aug 18 15:22:02 witherspoon mboxd[1025]: terminate called after throwing an instance of 'sdbusplus::xyz::openbmc_project::Common::Error::InternalFailure'
Aug 18 15:22:02 witherspoon mboxd[1025]:   what():  xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally.
Aug 18 15:22:17 witherspoon mboxd[1061]: terminate called after throwing an instance of 'sdbusplus::xyz::openbmc_project::Common::Error::InternalFailure'
Aug 18 15:22:17 witherspoon mboxd[1061]:   what():  xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally.
Aug 18 15:22:21 witherspoon mboxd[1074]: terminate called after throwing an instance of 'sdbusplus::xyz::openbmc_project::Common::Error::InternalFailure'
Aug 18 15:22:21 witherspoon mboxd[1074]:   what():  xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally.
Aug 18 15:22:23 witherspoon mboxd[1081]: terminate called after throwing an instance of 'sdbusplus::xyz::openbmc_project::Common::Error::InternalFailure'
Aug 18 15:22:23 witherspoon mboxd[1081]:   what():  xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally.
Aug 18 15:22:30 witherspoon mboxd[1088]: terminate called after throwing an instance of 'sdbusplus::xyz::openbmc_project::Common::Error::InternalFailure'
Aug 18 15:22:30 witherspoon mboxd[1088]:   what():  xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally.
Aug 18 15:22:36 witherspoon mboxd[1095]: terminate called after throwing an instance of 'sdbusplus::xyz::openbmc_project::Common::Error::InternalFailure'
Aug 18 15:22:36 witherspoon mboxd[1095]:   what():  xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally.
Aug 18 15:22:38 witherspoon mboxd[1102]: terminate called after throwing an instance of 'sdbusplus::xyz::openbmc_project::Common::Error::InternalFailure'
Aug 18 15:22:38 witherspoon mboxd[1102]:   what():  xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally.
Aug 18 15:22:40 witherspoon mboxd[1109]: terminate called after throwing an instance of 'sdbusplus::xyz::openbmc_project::Common::Error::InternalFailure'
Aug 18 15:22:40 witherspoon mboxd[1109]:   what():  xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally.
Aug 18 15:22:43 witherspoon mboxd[1116]: terminate called after throwing an instance of 'sdbusplus::xyz::openbmc_project::Common::Error::InternalFailure'
Aug 18 15:22:43 witherspoon mboxd[1116]:   what():  xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally.
Aug 18 15:22:46 witherspoon mboxd[1123]: terminate called after throwing an instance of 'sdbusplus::xyz::openbmc_project::Common::Error::InternalFailure'
Aug 18 15:22:46 witherspoon mboxd[1123]:   what():  xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally.
Aug 18 15:22:49 witherspoon mboxd[1130]: terminate called after throwing an instance of 'sdbusplus::xyz::openbmc_project::Common::Error::InternalFailure'
Aug 18 15:22:49 witherspoon mboxd[1130]:   what():  xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally.
Aug 18 15:22:53 witherspoon phosphor-dump-monitor[955]: Error in mapper call
Aug 18 15:22:58 witherspoon phosphor-dump-monitor[955]: Error in mapper call
Aug 18 15:22:58 witherspoon phosphor-dump-monitor[955]: Error in mapper call
Aug 18 15:23:02 witherspoon phosphor-dump-monitor[955]: Error in mapper call
Aug 18 15:23:02 witherspoon phosphor-dump-monitor[955]: Error in mapper call
Aug 18 15:23:03 witherspoon phosphor-dump-monitor[955]: Error in mapper call
Aug 18 15:23:04 witherspoon phosphor-dump-monitor[955]: Error in mapper call
Aug 18 15:23:04 witherspoon mboxd[1147]: terminate called after throwing an instance of 'sdbusplus::xyz::openbmc_project::Common::Error::InternalFailure'
Aug 18 15:23:04 witherspoon mboxd[1147]:   what():  xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally.
Aug 18 15:23:08 witherspoon mboxd[1156]: terminate called after throwing an instance of 'sdbusplus::xyz::openbmc_project::Common::Error::InternalFailure'
Aug 18 15:23:08 witherspoon mboxd[1156]:   what():  xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally.
Aug 18 15:23:11 witherspoon phosphor-dump-monitor[955]: Error in mapper call
Aug 18 15:23:15 witherspoon phosphor-dump-monitor[955]: Error in mapper call
Aug 18 15:23:15 witherspoon phosphor-dump-monitor[955]: Error in mapper call
Aug 18 15:23:15 witherspoon phosphor-dump-monitor[955]: Error in mapper call
Aug 18 15:23:15 witherspoon phosphor-dump-monitor[955]: Error in mapper call
Aug 18 15:23:16 witherspoon mboxd[1173]: terminate called after throwing an instance of 'sdbusplus::xyz::openbmc_project::Common::Error::InternalFailure'
Aug 18 15:23:16 witherspoon mboxd[1173]:   what():  xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally.
Aug 18 15:23:22 witherspoon mboxd[1179]: terminate called after throwing an instance of 'sdbusplus::xyz::openbmc_project::Common::Error::InternalFailure'
Aug 18 15:23:22 witherspoon mboxd[1179]:   what():  xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally.
Aug 18 15:23:23 witherspoon mboxd[1186]: terminate called after throwing an instance of 'sdbusplus::xyz::openbmc_project::Common::Error::InternalFailure'
Aug 18 15:23:23 witherspoon mboxd[1186]:   what():  xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally.
Aug 18 15:23:24 witherspoon mboxd[1193]: terminate called after throwing an instance of 'sdbusplus::xyz::openbmc_project::Common::Error::InternalFailure'
Aug 18 15:23:24 witherspoon mboxd[1193]:   what():  xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally.
Aug 18 15:23:32 witherspoon phosphor-dump-monitor[955]: Error in mapper call
Aug 18 15:23:32 witherspoon phosphor-dump-monitor[955]: Error in mapper call
Aug 18 15:23:32 witherspoon phosphor-dump-monitor[955]: Error in mapper call
Aug 18 15:23:33 witherspoon mboxd[1204]: terminate called after throwing an instance of 'sdbusplus::xyz::openbmc_project::Common::Error::InternalFailure'
[... 16 lines cut ...]
Aug 18 15:23:50 witherspoon mboxd[1253]:   what():  xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally.
Aug 18 15:23:53 witherspoon phosphor-dump-monitor[955]: Error in mapper call
Aug 18 15:23:53 witherspoon phosphor-dump-monitor[955]: Error in mapper call
Aug 18 15:23:55 witherspoon phosphor-dump-monitor[955]: Error in mapper call
Aug 18 15:23:55 witherspoon phosphor-dump-monitor[955]: Error in mapper call
```

`Driver Info`
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.9-50"
VERSION_ID="v1.99.9-50-g870799b"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.9-50"
BUILD_ID="v1.99.9"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-ba30e9ed152bab62f72da24986ee5ca22d6c9b07 #1 Thu Aug 24 23:27:37 UTC 2017 armv6l GNU/Linux
root@witherspoon:~# uptime
 15:33:42 up 12 min,  load average: 0.31, 1.41, 1.56
root@witherspoon:~#
```

- [x] Journald 
[Master_v1.99.9-50-g870799b_wont_boot_standby.txt](https://github.com/openbmc/openbmc/files/1252207/Master_v1.99.9-50-g870799b_wont_boot_standby.txt)

dkodihal, 2017-08-25:
It also looks like dump did not capture the core dump files.

ojayanth, 2017-08-25:
@dkodihal This issue  need to use  for tracking busctl failure. 
busctl get-property  xyz.openbmc_project.State.Host   /xyz/openbmc_project/state/host0    xyz.openbmc_project.State.Host CurrentHostState
The name xyz.openbmc_project.State.Host was not provided by any .service files

As per @geissonator " If I start host-state, we get the states, but still hung up waiting for something for ready".

It will be good to track the secondary issue core dump file missing in dump separately.

dkodihal, 2017-08-26:
There is another issue for state. This one is for dump related

ojayanth, 2017-08-26:
@dkodihal mentioned there is already issue related to host-state check. 
This issue will use for fixing  "missing core file" in the dump.

rfrandse, 2017-08-26:
<https://gerrit.openbmc-project.xyz/6240> Fix for missing core file in the dump
Resolves  #2214 phosphor-dump-monitor[955]: Error in mapper call

ojayanth, 2017-08-31:
latest master with this fix   still not showing core file in the dump . 
Based on the initial investigation inotify reports inode number instead of actual file name.

root@witherspoon:/var/lib/systemd/coredump# ls -i
    619 core.ipmid.0.df10227843214e108eec954241adc919.1287.1503945744000000000000.xz

ojayanth, 2017-08-31:
I tested another system and Looks like it is working good
1503960974.tar.xz var/lib/phosphor-debug-collector/dumps/1# tar -xvf obmcdump_1_1
obmcdump_1_1503960974/
obmcdump_1_1503960974/top.log
obmcdump_1_1503960974/meminfo.log
obmcdump_1_1503960974/cpuinfo.log
obmcdump_1_1503960974/obmc_console.log
obmcdump_1_1503960974/failed_services.log
obmcdump_1_1503960974/hostnamectl.log
obmcdump_1_1503960974/HostState.log
obmcdump_1_1503960974/BMCState.log
obmcdump_1_1503960974/disk_usage.log
obmcdump_1_1503960974/uptime.log
obmcdump_1_1503960974/uname.log
obmcdump_1_1503960974/firmware_release.log
obmcdump_1_1503960974/proc_journal.log
obmcdump_1_1503960974/dreport.log
obmcdump_1_1503960974/core.phosphor-fru-fa.0.219c55ad75e64de1bb1035841780d859.1186.1503960969000000000000.xz
obmcdump_1_1503960974/summary.log

ojayanth, 2017-08-31:
Discussed with @williamspatrick  he suggested to looks the file system type.
                                   Filesystem           1K-blocks      Used Available Use% Mounted on
Working:  ubi0:rwfs                 2808      1540      1268  55% /var 
Not working: cow                       4096       560      3536  14% /

Enabled some more traces in file watch code , and noticed that , COW  file system based /var does lot of file access operation on "#inode" based core name and ubi all operation are specific to systemd-coredump based file format

ojayanth, 2017-09-01:
Created new RTC issue to track this filesystem specific issue 
https://github.com/openbmc/openbmc/issues/2240

closed 2017-09-01 by dkodihal