openbmc/openbmc #2215: [xCAT] Setting VLAN out-of-band via REST not working as expected

whowutwut, 2017-08-25:
@ratagupt  (@xuweibj FYI) 

I may be doing something wrong here, so please help... 

Using GHE build from 08/13  `BUILD_ID="ibm-v1.99.8-231-gdf827b69-r0"` 

Here's the steps that I have taken to restore the BMC to DHCP to start clean...
```
rm /etc/systemd/network/*
cp /etc/systemd/network00-bmc-eth0.network  /etc/systemd/network/00-bmc-eth0.network
reboot
```

When the BMC comes up, it obtains a DHCP IP address..., I do the following: 

1. Login 
```
+ curl -c cjar -k -X POST -H 'Content-Type: application/json' -d '{"data": [ "root", "0penBmc" ] }' https://172.12.253.111/login
{
  "data": "User 'root' logged in",
  "message": "200 OK",
  "status": "ok"
}
```

2. List to show NO VLAN 
```
[root@briggs01 mid05tor12cn05]# ./list.sh
+ curl -c cjar -b cjar -k -H 'Content-Type: application/json' -X GET https://172.12.253.111/xyz/openbmc_project/network/enumerate
{
  "data": {
    "/xyz/openbmc_project/network/config": {
      "DefaultGateway": "172.12.253.27",
      "HostName": "witherspoon"
    },
    "/xyz/openbmc_project/network/config/dhcp": {
      "DNSEnabled": 0,
      "HostNameEnabled": 0,
      "NTPEnabled": 0
    },
    "/xyz/openbmc_project/network/eth0": {
      "AutoNeg": 0,
      "DHCPEnabled": 1,
      "DomainName": [],
      "InterfaceName": "eth0",
      "MACAddress": "70:e2:84:14:28:d",
      "Nameservers": [],
      "Speed": 0
    },
    "/xyz/openbmc_project/network/eth0/ipv4/b318b9be": {
      "Address": "172.12.253.111",
      "Gateway": "0.0.0.0",
      "Origin": "xyz.openbmc_project.Network.IP.AddressOrigin.DHCP",
      "PrefixLength": 16,
      "Type": "xyz.openbmc_project.Network.IP.Protocol.IPv4"
    },
    "/xyz/openbmc_project/network/eth0/ipv6/28f8222a": {
      "Address": "fe80::72e2:84ff:fe14:280d",
      "Gateway": "",
      "Origin": "xyz.openbmc_project.Network.IP.AddressOrigin.DHCP",
      "PrefixLength": 64,
      "Type": "xyz.openbmc_project.Network.IP.Protocol.IPv6"
    },
    "/xyz/openbmc_project/network/host0/intf": {
      "MACAddress": "00:00:00:00:00:00"
    },
    "/xyz/openbmc_project/network/host0/intf/addr": {
      "Address": "0.0.0.0",
      "Gateway": "0.0.0.0",
      "Origin": "xyz.openbmc_project.Network.IP.AddressOrigin.Static",
[... 175 lines cut ...]
    inet6 fe80::72e2:84ff:fe14:280d/64 scope link
       valid_lft forever preferred_lft forever
```

Even with restart of networkd, nothing changes.    `systemctl restart systemd-networkd` 

```
└─/xyz
  └─/xyz/openbmc_project
    └─/xyz/openbmc_project/network
      ├─/xyz/openbmc_project/network/config
      │ └─/xyz/openbmc_project/network/config/dhcp
      ├─/xyz/openbmc_project/network/eth0
      │ ├─/xyz/openbmc_project/network/eth0/ipv4
      │ │ └─/xyz/openbmc_project/network/eth0/ipv4/b318b9be
      │ └─/xyz/openbmc_project/network/eth0/ipv6
      │   └─/xyz/openbmc_project/network/eth0/ipv6/28f8222a
      └─/xyz/openbmc_project/network/eth0_11
        └─/xyz/openbmc_project/network/eth0_11/ipv4
          └─/xyz/openbmc_project/network/eth0_11/ipv4/4cf442d5
```

Am i doing something wrong here?   Help! 

I can update to 8/24 GHE build on Monday... but I wanted to try this out since it was claimed to be included in the 8/13 GHE build first....

whowutwut, 2017-08-25:
@ratagupt  Using the 08/24 GHE build...  a little different... 

All the same steps as above... Set the IP
```
[root@briggs01 mid05tor12cn05]## ./set_ip.sh172.11.140.5 172.12.253.111
==> VLAN=11
==> Interface=eth0_11
==> Setting to 172.11.140.5 (16) Gateway: 172.11.254.2
++ curl -c cjar -b cjar -k -H 'Content-Type: application/json' -X POST -d '{"data": ["xyz.openbmc_project.Network.IP.Protocol.IPv4", "172.11.140.5", 16, "172.11.254.2"] }' https://172.12.253.111//xyz/openbmc_project/network/eth0_11/action/IP
{
  "data": null,
  "message": "200 OK",
  "status": "ok"
```

The `/etc/systemd/network` files are changed... but the ifconfig doesn't show it.  

I run a restart on the network, to see if it would pick up the file config and the eth0 gets a 169 IP address:

```
root@witherspoon:/etc/systemd/network# ifconfig
eth0      Link encap:Ethernet  HWaddr 70:E2:84:14:28:0D
          inet addr:169.254.178.193  Bcast:169.254.255.255  Mask:255.255.0.0  <----------- what IP is this? 
          inet6 addr: fe80::72e2:84ff:fe14:280d%1106588560/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:17220 errors:0 dropped:9 overruns:0 frame:0
          TX packets:13835 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:22283857 (21.2 MiB)  TX bytes:1117948 (1.0 MiB)
          Interrupt:20
```

The Tree does not have the VLAN info...

```
└─/xyz
  └─/xyz/openbmc_project
    └─/xyz/openbmc_project/network
      ├─/xyz/openbmc_project/network/config
      │ └─/xyz/openbmc_project/network/config/dhcp
      ├─/xyz/openbmc_project/network/eth0
      │ ├─/xyz/openbmc_project/network/eth0/ipv4
      │ │ └─/xyz/openbmc_project/network/eth0/ipv4/b318b9be
      │ └─/xyz/openbmc_project/network/eth0/ipv6
      │   └─/xyz/openbmc_project/network/eth0/ipv6/28f8222a
      └─/xyz/openbmc_project/network/eth0_11
```

But the files are there.. 
```
root@witherspoon:/etc/systemd/network# cat 00-bmc-eth0.11.network
[Match]
Name=eth0.11
[Network]
Address=172.11.140.5/16
Gateway=172.12.253.27
[Route]
Gateway=172.11.254.2
Destination=172.11.0.0
root@witherspoon:/etc/systemd/network# cat eth0.11.netdev
[NetDev]
Name=eth0.11
Kind=vlan
[VLAN]
Id=11
```

I did not set DHCPEnabled=0 on this run...

rfrandse, 2017-08-31:
<https://gerrit.openbmc-project.xyz/6285> DHCP value was not getting updated correctly
Resolves  #2215 [xCAT] Setting VLAN out-of-band via REST not working as expected

closed 2017-08-31 by williamspatrick