openbmc/openbmc #2219: Inventory Build Dated back to year 1996

gkeishin, 2017-08-28:
```
    "/xyz/openbmc_project/inventory/system/chassis/motherboard": {
        "BuildDate": "1996-01-01 - 00:00:00",
        "Cached": 0,
        "FieldReplaceable": 0,
        "Manufacturer": "0000000000000000",
        "Model": "",
        "PartNumber": "00VK614         ",
        "Present": 1,
        "PrettyName": "SYSTEM PLANAR   ",
        "SerialNumber": "YL30UF759012    "
    },

```

```
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu0": {
        "BuildDate": "1996-01-01 - 00:00:00",
        "Cached": 0,
        "FieldReplaceable": 1,
        "Functional": 1,
        "Manufacturer": "IBM",
        "Model": "",
        "PartNumber": "02AA765",
        "Present": 1,
        "PrettyName": "PROCESSOR MODULE",
        "SerialNumber": "YA1934007757",
        "Version": "20"
    },
```

```
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu1": {
        "BuildDate": "1996-01-01 - 00:00:00",
        "Cached": 0,
        "FieldReplaceable": 1,
        "Functional": 1,
        "Manufacturer": "IBM",
        "Model": "",
        "PartNumber": "02AA765",
        "Present": 1,
        "PrettyName": "PROCESSOR MODULE",
        "SerialNumber": "YA1934007753",
        "Version": "20"
    },
```

- [x] Journald 
[journald_v1.99.9-57-g8aa1.txt](https://github.com/openbmc/openbmc/files/1256471/journald_v1.99.9-57-g8aa1.txt)

gkeishin, 2017-08-28:
@tomjoseph83 Is this something to do with your recent changes.

gkeishin, 2017-08-28:
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.9-57"
VERSION_ID="v1.99.9-57-g8aa1141"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.9-57"
BUILD_ID="v1.99.9"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-ba30e9ed152bab62f72da24986ee5ca22d6c9b07 #1 Fri Aug 25 15:44:46 UTC 2017 armv6l GNU/Linux
root@witherspoon:~#
```

tomjoseph83, 2017-08-29:
@gkeishin I added changes to populate the Build Date in the dbus object. As per specification(platform-management-fru-document-rev-1-2-feb-2013) Page No 14. The 3 bytes mentioned below indicate the no of minutes from 1/1/96. If the build date is not populated in the Board VPD, hostboot would send down 0x00 x00 0x00 and would be shown as "1996-01-01 - 00:00:00".

```
Mfg. Date / Time
Number of minutes from 0:00 hrs 1/1/96.
LSbyte first (little endian)
00_00_00h = unspecified.
```

gkeishin, 2017-08-29:
Great.. Thanks. I ll check on it by updating the info and verify..

gkeishin, 2017-09-08:
I ll verify this on GHE as the MFG uses released build. closing and open a CQ if required

closed 2017-09-08 by gkeishin