openbmc/openbmc #2221: openpower-software-manager installs files into a shared area when those files already exist

amboar, 2017-08-29:
From Jenkins:

```
ERROR: openpower-software-manager-1.0-r1 do_populate_sysroot: The recipe openpower-software-manager is trying to install files into a shared area when those files already exist. Those files and their manifest location are:
   /tmp/openbmc/build/tmp/sysroots/witherspoon/usr/include/org/openbmc/Associations/server.hpp
 Matched in b'manifest-witherspoon-phosphor-software-manager.populate_sysroot'
Please verify which recipe should provide the above files.
The build has stopped as continuing in this scenario WILL break things, if not now, possibly in the future (we've seen builds fail several months later). If the system knew how to recover from this automatically it would however there are several different scenarios which can result in this and we don't know which one this is. It may be you have switched providers of something like virtual/kernel (e.g. from linux-yocto to linux-yocto-dev), in that case you need to execute the clean task for both recipes and it will resolve this error. It may be you changed DISTRO_FEATURES from systemd to udev or vice versa. Cleaning those recipes should again resolve this error however switching DISTRO_FEATURES on an existing build directory is not supported, you should really clean out tmp and rebuild (reusing sstate should be safe). It could be the overlapping files detected are harmless in which case adding them to SSTATE_DUPWHITELIST may be the correct solution. It could also be your build is including two different conflicting versions of things (e.g. bluez 4 and bluez 5 and the correct solution for that would be to resolve the conflict. If in doubt, please ask on the mailing list, sharing the error and filelist above.
```

amboar, 2017-08-29:
Hit by https://gerrit.openbmc-project.xyz/#/c/6271 and probably at least its parent commit.

williamspatrick, 2017-08-29:
I think that the generated header somehow got put in the automake list for install. We need to 'noinst' that header in both repos.

closed 2017-08-29 by williamspatrick