openbmc/openbmc #2222: Fan Control / Hwmon: Handle OCC temp sensor failure

msbarth, 2017-08-29:
When a temp sensor that is provided by the OCC times out having read failures, a 0xFF value is returned in the current temp reading byte. When a 0xFF is read by fan control, the fans for zone 0 should be set to full speed (10500rpms)
[image]

amboar, 2018-03-14:
Are we exposing these values via the sensor value attribute? I feel like this case should be translated into `-EAGAIN` or `-EIO` on reading the attribute.

msbarth, 2018-03-14:
Pre-4.13 these 0x00 & 0xFF values were being presented in the sensor value attribute. A driver update was made (and included in post-4.13) to provied EAGAIN for 0x00 and EIO on 0xFF, however the user side support in hwmon was not complete, hence this issue. This problem is in discussion right now and scheduling the work is underway.

rfrandse, 2018-04-27:
<https://gerrit.openbmc-project.xyz/10334> wspoon: Handle faulted OCC temp sensors
Resolves:  #2222 Fan Control / Hwmon(#2329): Handle OCC temp sensor failure

closed 2018-05-03 by geissonator