openbmc/openbmc #2225: /xyz/openbmc_project/logging/enumerate returns error

trajeswaran, 2017-08-29:
After biasing the VRMs one by one, listing the error log entries using the Rest API xyz/openbmc_project/logging/enumerate results in error. For biasing i2c commands are used. This sequence is used by memory characterization team.  

Please find attached the ffdc.

When I ran the top command in BMC at the time of executing the Rest API logging/enumerate, I noticed that the process "python /usr/sbin/phosphor-mapper --path_namespaces=/org/openbmc /xyz/openbmc_pr" utilized 100% cpu.

"data": {
    "description": "Internal Server Error", 
    "exception": "DBusException('Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.',)", 
    "traceback": [
      "Traceback (most recent call last):", 
      "  File \"/usr/lib/python2.7/site-packages/bottle.py\", line 858, in _handle", 
      "    route, args = self.router.match(environ)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 1124, in custom_router_match", 
      "    route.callback._setup(**args)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 187, in _setup", 
      "    return self.setup(**kw)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 301, in setup", 
      "    request.route_data['map'] = self.find(path)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 298, in find", 
      "    self.mapper.get_subtree, path=path)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 220, in try_mapper_call", 
      "    return f(**kw)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py\", line 72, in get_subtree", 
      "    retries, interval)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py\", line 42, in retry", 
      "    return func()", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py\", line 71, in <lambda>", 
      "    path, depth, interfaces, signature='sias'),", 
      "  File \"/usr/lib/python2.7/site-packages/dbus/proxies.py\", line 145, in __call__", 
      "    **keywords)", 
      "  File \"/usr/lib/python2.7/site-packages/dbus/connection.py\", line 651, in call_blocking", 
      "    message, timeout)", 
      "DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken."
    ]
  }, 
  "message": "500 Internal Server Error", 
  "status": "error"
}

[20170829122355037393_BMC_dmesg.txt](https://github.com/openbmc/openbmc/files/1261296/20170829122355037393_BMC_dmesg.txt)
[20170829122355037393_BMC_inventory.txt](https://github.com/openbmc/openbmc/files/1261297/20170829122355037393_BMC_inventory.txt)
[20170829122355037393_BMC_meminfo.txt](https://github.com/openbmc/openbmc/files/1261299/20170829122355037393_BMC_meminfo.txt)
[20170829122355037393_BMC_proc_list.txt](https://github.com/openbmc/openbmc/files/1261300/20170829122355037393_BMC_proc_list.txt)
[20170829122355037393_BMC_sensor_list.txt](https://github.com/openbmc/openbmc/files/1261301/20170829122355037393_BMC_sensor_list.txt)
[20170829122355037393_BMC_sensor_xyz_list.txt](https://github.com/openbmc/openbmc/files/1261302/20170829122355037393_BMC_sensor_xyz_list.txt)
[20170829122355037393_BMC_procinfo.txt](https://github.com/openbmc/openbmc/files/1261303/20170829122355037393_BMC_procinfo.txt)
[20170829122355037393_BMC_proc_fd_active_list.txt](https://github.com/openbmc/openbmc/files/1261305/20170829122355037393_BMC_proc_fd_active_list.txt)
[20170829122355037393_BMC_journalctl.txt](https://github.com/openbmc/openbmc/files/1261306/20170829122355037393_BMC_journalctl.txt)
[20170829122355037393_BMC_general.txt](https://github.com/openbmc/openbmc/files/1261307/20170829122355037393_BMC_general.txt)
[20170829122355037393_BMC_control_list.txt](https://github.com/openbmc/openbmc/files/1261309/20170829122355037393_BMC_control_list.txt)

gtmills, 2018-09-04:
@anoo1 I assume we haven't seen this problem recently? Can we close?

anoo1, 2018-09-04:
Yeah we haven't seen an issue recently, closing.

closed 2018-09-04 by anoo1