openbmc/openbmc #2227: Network:- Defer the systemd-networkd restart

ratagupt, 2017-08-30:
It is agreed on that instead of starting the systemd-networkd for each network operation,we would defer it for some time.
There are two approaches
1) we would wait for a certain amount of time and cache all the changes and then restart the systemd-networkd.
2) one more dbus method say "apply Changes".which will apply all the changes and restart the systemd-networkd.
Need to look at the pros and cons of both approaches.

ratagupt, 2017-09-04:
It is not a code review defect,It is enhancement.

ratagupt, 2017-09-07:
As discussed we will go with the approach 1

- [ ] Implement the timer which is already there.

- [ ] Start the timer whenever we get any dbus request,Need to make sure that only start the timer if it is not started.

- [ ] Once timer expired then restart the systemd-networkd.

rfrandse, 2017-09-23:
<https://gerrit.openbmc-project.xyz/6844> Reducing the restart of systemd-networkd by implementing timers
Resolves  #2227 Network:- Defer the systemd-networkd restart

sivassrr, 2017-09-25:
Not seeing much impact to existing network TCs. We need to rerun and do regression test. @prkatti1 FYI .

closed 2017-10-11 by geissonator