openbmc/openbmc #2228: NETDEV WATCHDOG: eth0 (ftgmac100): transmit queue 0 timed out

gkeishin, 2017-08-30:
```
[  438.150000] ------------[ cut here ]------------
[  438.150000] WARNING: CPU: 0 PID: 6 at /tmp/openbmc/build/tmp/work-shared/witherspoon/kernel-source/net/sched/sch_generic.c:316 dev_watchdog+0x230/0x248
[  438.150000] NETDEV WATCHDOG: eth0 (ftgmac100): transmit queue 0 timed out
[  438.150000] CPU: 0 PID: 6 Comm: ksoftirqd/0 Not tainted 4.10.17-ba30e9ed152bab62f72da24986ee5ca22d6c9b07 #1
[  438.150000] Hardware name: ASpeed SoC
[  438.150000] [<80010080>] (unwind_backtrace) from [<8000d934>] (show_stack+0x20/0x24)
[  438.150000] [<8000d934>] (show_stack) from [<801dbf2c>] (dump_stack+0x20/0x28)
[  438.150000] [<801dbf2c>] (dump_stack) from [<80018d54>] (__warn+0xe0/0x108)
[  438.150000] [<80018d54>] (__warn) from [<80018dc0>] (warn_slowpath_fmt+0x44/0x4c)
[  438.150000] [<80018dc0>] (warn_slowpath_fmt) from [<80353674>] (dev_watchdog+0x230/0x248)
[  438.150000] [<80353674>] (dev_watchdog) from [<80053be8>] (call_timer_fn+0x40/0x124)
[  438.150000] [<80053be8>] (call_timer_fn) from [<80053d60>] (expire_timers+0x94/0xcc)
[  438.150000] [<80053d60>] (expire_timers) from [<80053e74>] (run_timer_softirq+0x8c/0x170)
[  438.150000] [<80053e74>] (run_timer_softirq) from [<80009554>] (__do_softirq+0xc4/0x2b4)
[  438.150000] [<80009554>] (__do_softirq) from [<8001c284>] (run_ksoftirqd+0x34/0x4c)
[  438.150000] [<8001c284>] (run_ksoftirqd) from [<800375d8>] (smpboot_thread_fn+0x120/0x180)
[  438.150000] [<800375d8>] (smpboot_thread_fn) from [<800335ec>] (kthread+0x124/0x164)
[  438.150000] [<800335ec>] (kthread) from [<8000a5e8>] (ret_from_fork+0x14/0x2c)
[  438.150000] ---[ end trace d43d456a6fbfebd8 ]---
```

`Journald sinppet`
```
root@witherspoon:~# dmesg | grep -i net
[    0.150000] NET: Registered protocol family 16
[    0.240000] NET: Registered protocol family 2
[    0.240000] NET: Registered protocol family 1
[    3.580000] ftgmac100 1e660000.ethernet: Read MAC address 70:e2:84:14:28:76 from chip
[    3.580000] ftgmac100 1e660000.ethernet: Using NCSI interface
[    3.580000] ftgmac100 1e660000.ethernet eth0: irq 20, mapped at f0060000
[    3.960000] Driver for 1-wire Dallas network protocol.
[    6.390000] NET: Registered protocol family 10
[    6.400000] NET: Registered protocol family 17
[    6.450000] console [netcon0] enabled
[    6.450000] netconsole: network logging started
[   12.290000] systemd[1]: Listening on Network Service Netlink Socket.
[   27.450000] ftgmac100 1e660000.ethernet eth0: NCSI interface down
[  438.150000] WARNING: CPU: 0 PID: 6 at /tmp/openbmc/build/tmp/work-shared/witherspoon/kernel-source/net/sched/sch_generic.c:316 dev_watchdog+0x230/0x248
[  438.150000] NETDEV WATCHDOG: eth0 (ftgmac100): transmit queue 0 timed out
root@witherspoon:~#
```
`Driver Info`
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.9-84"
VERSION_ID="v1.99.9-84-g13138e1"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.9-84"
BUILD_ID="v1.99.9"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-ba30e9ed152bab62f72da24986ee5ca22d6c9b07 #1 Tue Aug 29 19:02:49 UTC 2017 armv6l GNU/Linux
root@witherspoon:~#
```
- [x] dmesg log 
[dmesg_kernel.txt](https://github.com/openbmc/openbmc/files/1262547/dmesg_kernel.txt)

gkeishin, 2017-08-30:
- [x] Journald log 
[NET_oops_journald.txt](https://github.com/openbmc/openbmc/files/1262559/NET_oops_journald.txt)

gkeishin, 2017-08-30:
@amboar ^^^

amboar, 2017-08-30:
@ozbenh @sammj this looks like a network related oops (`dev_watchdog`).

amboar, 2017-08-30:
Some preliminary results:

Running `top` showed:

```
Mem: 147640K used, 284608K free, 14428K shrd, 11092K buff, 52688K cached
CPU:   0% usr   0% sys   0% nic   0% idle   0% io   0% irq 100% sirq
Load average: 1.05 1.08 1.02 7/143 1397
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
    6     2 root     RW       0   0% 100% [ksoftirqd/0]
 1115     1 root     S    71580  17%   0% mboxd --flash 64M --window-size 1M
```
Interrupt status:
```
root@witherspoon:~# cat /proc/interrupts
           CPU0
 17:       6678      AVIC  16 Edge      timer
 18:          0      AVIC  19 Edge      aspeed-smc
 19:          0      AVIC  46 Edge      aspeed-mbox
 20:     218175      AVIC   2 Edge      eth0
 22:          0      AVIC   8 Edge      ipmi-bt-host, serial
 25:      27051      AVIC  10 Edge      serial
 26:          0     dummy   2 Edge      1e78a0c0.i2c-bus
 27:    3221262     dummy   3 Edge      1e78a100.i2c-bus
 28:    1163972     dummy   4 Edge      1e78a140.i2c-bus
 29:    1164695     dummy   5 Edge      1e78a180.i2c-bus
 30:          4     dummy   9 Edge      1e78a380.i2c-bus
 31:          0     dummy  10 Edge      1e78a3c0.i2c-bus
 32:      13363     dummy  11 Edge      1e78a400.i2c-bus
 33:          0     dummy  12 Edge      1e78a440.i2c-bus
 34:          0     dummy  13 Edge      1e78a480.i2c-bus
 48:          0  aspeed-gpio  13 Edge      air-water
102:          0  aspeed-gpio  67 Edge      gpiolib
108:          0  aspeed-gpio  73 Edge      gpiolib
109:          0  aspeed-gpio  74 Edge      checkstop
139:          0  aspeed-gpio 104 Edge      ps1-presence
162:          0  aspeed-gpio 127 Edge      ps0-presence
Err:          0
```
Disabling the network with `ifconfig eth0 down` then running `top` gave:
```
root@witherspoon:~# top
Mem: 147456K used, 284792K free, 14436K shrd, 11092K buff, 52708K cached
CPU:   7% usr  28% sys   0% nic  64% idle   0% io   0% irq   0% sirq
Load average: 1.08 1.10 1.04 4/144 1418
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
  972     1 root     R    16572   4%  10% python /usr/sbin/phosphor-mapper --path_namespaces=/xyz/openbmc_project /org/openbmc /org/open_power --interface_namespaces=org.open_
 1137     1 root     R     6084   1%  10% phosphor-hwmon-readd -o /ahb/apb/i2c@1e78a000/i2c-bus@100/max31785@52
 1339     1 root     R     6084   1%  10% phosphor-hwmon-readd -o /ahb/apb/i2c@1e78a000/i2c-bus@140/ir35221@71
```
Checking the clocks and reset registers for mac0 didn't show any obvious problems, though mac0 is being clocked at HPLL/10 = 79.2MHz, where the minimum clock rate for NCSI is 25MHz.

Trying to reboot the box invoked the oomkiller:
```
root@witherspoon:~# reboot
Failed to wait for response: Bad message
root@witherspoon:~# [  OK  ] Started /dev/initctl Compatibility Daemon.
[  OK  ] Removed slice system-op\x2dreset\x2dchassis\x2drunning.slice.
         Stopping Phosphor sysfs LED controller...
[ 6205.810000] systemd-cgroups invoked oom-killer: gfp_mask=0x0(), nodemask=0, order=0, oom_score_adj=0
[ 6205.810000] CPU: 0 PID: 1429 Comm: systemd-cgroups Tainted: G        W       4.10.17-ba30e9ed152bab62f72da24986ee5ca22d6c9b07 #1
[ 6205.850000] Hardware name: ASpeed SoC
[ 6205.850000] [<80010080>] (unwind_backtrace) from [<8000d934>] (show_stack+0x20/0x24)
[ 6205.850000] [<8000d934>] (show_stack) from [<801dbf2c>] (dump_stack+0x20/0x28)
[ 6205.850000] [<801dbf2c>] (dump_stack) from [<800def04>] (dump_header.constprop.5+0x80/0x1b0)
[ 6205.850000] [<800def04>] (dump_header.constprop.5) from [<800aa948>] (oom_kill_process+0x2a0/0x46c)
[ 6205.850000] [<800aa948>] (oom_kill_process) from [<800aaf18>] (out_of_memory+0xe8/0x418)
[ 6205.850000] [<800aaf18>] (out_of_memory) from [<800ab290>] (pagefault_out_of_memory+0x48/0x5c)
[ 6205.900000] [<800ab290>] (pagefault_out_of_memory) from [<80011e30>] (do_page_fault+0x29c/0x2c8)
[ 6205.900000] [<80011e30>] (do_page_fault) from [<80009244>] (do_DataAbort+0x48/0xc8)
[... 70 lines cut ...]
[ 6206.480000] [ 1167]     0  1167     1321      704       6       0        0             0 phosphor-fru-fa
[ 6206.480000] [ 1172]     0  1172     1470      951       7       0        0             0 phosphor-gpio-p
[ 6206.480000] [ 1174]     0  1174     1470      913       6       0        0             0 phosphor-gpio-p
[ 6206.480000] [ 1182]     0  1182     1508      829       6       0        0             0 openpower-occ-c
[ 6206.480000] [ 1184]     0  1184      942      408       6       0        0             0 netman_watch_dn
[ 6206.480000] [ 1192]     0  1192     4455     2867      12       0        0             0 python
[ 6206.590000] [ 1204]     0  1204     1437      912       7       0        0             0 phosphor-bmc-st
[ 6206.590000] [ 1205]     0  1205     1437      931       7       0        0             0 phosphor-chassi
[ 6206.590000] [ 1206]     0  1206     8154     1068      11       0        0             0 button_power.ex
[ 6206.590000] [ 1207]     0  1207     8156     1098      11       0        0             0 power_control.e
[ 6206.590000] [ 1208]     0  1208     8154     1076       9       0        0             0 button_reset.ex
[ 6206.590000] [ 1225]     0  1225     1829     1232       7       0        0             0 ipmid
[ 6206.590000] [ 1237]     0  1237     1468      933       6       0        0             0 timemanager
[ 6206.590000] [ 1249]     0  1249     1484     1015       7       0        0             0 phosphor-host-s
[... cut ...]
[ 6206.720000] [ 1427]     0  1427     6685      416      11       0        0             0 systemd
[ 6206.720000] [ 1428]     0  1428     1241      467       5       0        0             0 systemd-cgroups
[ 6206.720000] [ 1429]     0  1429     1243      262       5       0        0             0 systemd-cgroups
[ 6206.720000] Out of memory: Kill process 1192 (python) score 25 or sacrifice child
```
The oomkiller proceeded to stomp around killing other processes for the remainder of the reboot process

gkeishin, 2017-08-31:
We are hit badly on this particular system, it was working good last week, not sure what has caused this havoc. This is the ONLY DD2.0 system we got, any help to bring it up or direction would be of great help

amboar, 2017-08-31:
Was it DD2.0 last week when it was working?

gkeishin, 2017-08-31:
Yes @amboar it was DD2.0 last week.

ozbenh, 2017-08-31:
Any idea what might have changed ?

Cheers,
Ben.

gkeishin, 2017-08-31:
The only thing which may not be related but that I can remember of is, that we were running GHE builds earlier on that system for testing and were working ok.. and then we happened to load the Master public build to test and then thing started not good. 

Now even if we switch to GHE build as well, the same remains. For a while, we thought it's could be because of the Austin, Tx network issues due to bad weather and cabling NW issue there. ... but thats being sorted out, we are here now with this system eth0 interface going down.

gkeishin, 2017-08-31:
Looks like we still hit it.
```
root@witherspoon:~# dmesg | grep -i net
[    0.140000] NET: Registered protocol family 16
[    0.230000] NET: Registered protocol family 2
[    0.240000] NET: Registered protocol family 1
[    3.590000] ftgmac100 1e660000.ethernet: Read MAC address 70:e2:84:14:28:76 from chip
[    3.590000] ftgmac100 1e660000.ethernet: Using NCSI interface
[    3.590000] ftgmac100 1e660000.ethernet eth0: irq 20, mapped at f0060000
[    3.950000] Driver for 1-wire Dallas network protocol.
[    6.400000] NET: Registered protocol family 10
[    6.410000] NET: Registered protocol family 17
[    6.460000] console [netcon0] enabled
[    6.460000] netconsole: network logging started
[   32.880000] ftgmac100 1e660000.ethernet eth0: NCSI interface down
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-g05ea6be #114 Thu Aug 31 17:18:59 AEST 2017 armv6l GNU/Linux
root@witherspoon:~#
```

gkeishin, 2017-09-01:
I have put a GHE build yesterday and it hasn't gone down so far till today.. so the Master public build has some feature which must be causing this.

The GHE build that I loaded.

```
root@witherspoon:~# uptime
 03:44:50 up  8:09,  load average: 0.16, 0.19, 0.17
root@witherspoon:~#

root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v1.99.9"
VERSION_ID="ibm-v1.99.9-0-r3-0-g8bb9b44"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v1.99.9"
BUILD_ID="ibm-v1.99.9-0-r3"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-ba30e9ed152bab62f72da24986ee5ca22d6c9b07 #1 Wed Aug 30 15:30:59 CDT 2017 armv6l GNU/Linux
root@witherspoon:~# cat /media/pnor-ro-8764baa3/VERSION
IBM-witherspoon-sequoia-ibm-OP9_v1.18_1.54
        op-build-v1.7-1321-g6eef393-dirty
        buildroot-2017.05-163-g8035745
        skiboot-v5.8-rc1-pd9d2012
        hostboot-c68be97
        linux-4.12.8-openpower1-p919e663
        petitboot-v1.5.1-p848cad7
        machine-xml-c1e49a6-pbb9f626
        occ-854999a
        hostboot-binaries-67a15fd
        capp-ucode-9c73e9f
        sbe-4a62126-p180ac2a
root@witherspoon:~#
```

geissonator, 2017-09-01:
We hit this on a different system today (w50) so bumping up the priority and marking phase 5 due to the fall out when we hit this (no network to system works).

Interestingly enough, after a few hours, the system seemed to fix itself and network started working, which is how we get this FFDC capture.  Some interesting failure about vlan before we hit the fail, this system was doing no vlan testing so interesting data point

```
11416 Aug 28 20:51:26 witherspoon kernel: ftgmac100 1e660000.ethernet eth0: no vlan ids left to set
11417 Aug 28 20:51:26 witherspoon kernel[951]: [ 9845.730000] ftgmac100 1e660000.ethernet eth0: no vlan ids left to set
...
11430 Aug 28 20:51:34 witherspoon kernel: WARNING: CPU: 0 PID: 8116 at /tmp/anoo/obmc/witherspoon/build/tmp/work-shared/witherspoon/kernel-source/net/ncsi/ncsi-manage.c:256 ncsi_start_channel_monitor+0x7c/0      x8c
11431 Aug 28 20:51:34 witherspoon kernel: CPU: 0 PID: 8116 Comm: kworker/0:5 Not tainted 4.10.17-76ff93cf4565068e97b802e41748eb6745500606 #1
11432 Aug 28 20:51:34 witherspoon kernel: Hardware name: ASpeed SoC
11433 Aug 28 20:51:34 witherspoon kernel: Workqueue: events ncsi_dev_work
11434 Aug 28 20:51:34 witherspoon kernel: [<80010080>] (unwind_backtrace) from [<8000d934>] (show_stack+0x20/0x24)
11435 Aug 28 20:51:34 witherspoon kernel: [<8000d934>] (show_stack) from [<801dbf2c>] (dump_stack+0x20/0x28)
11436 Aug 28 20:51:34 witherspoon kernel: [<801dbf2c>] (dump_stack) from [<80018d54>] (__warn+0xe0/0x108)
11437 Aug 28 20:51:34 witherspoon kernel: [<80018d54>] (__warn) from [<80018e48>] (warn_slowpath_null+0x30/0x38)
11438 Aug 28 20:51:34 witherspoon kernel: [<80018e48>] (warn_slowpath_null) from [<803f6940>] (ncsi_start_channel_monitor+0x7c/0x8c)
11439 Aug 28 20:51:34 witherspoon kernel: [<803f6940>] (ncsi_start_channel_monitor) from [<803f74f4>] (ncsi_configure_channel+0xd0/0x5c4)
11440 Aug 28 20:51:34 witherspoon kernel: [<803f74f4>] (ncsi_configure_channel) from [<803f7f64>] (ncsi_dev_work+0xac/0x494)
11441 Aug 28 20:51:34 witherspoon kernel: [<803f7f64>] (ncsi_dev_work) from [<8002d21c>] (process_one_work+0x1e0/0x450)
11442 Aug 28 20:51:34 witherspoon kernel: [<8002d21c>] (process_one_work) from [<8002d4e8>] (worker_thread+0x5c/0x570)
11443 Aug 28 20:51:34 witherspoon kernel: [<8002d4e8>] (worker_thread) from [<800335ec>] (kthread+0x124/0x164)
11444 Aug 28 20:51:34 witherspoon kernel: [<800335ec>] (kthread) from [<8000a5e8>] (ret_from_fork+0x14/0x2c)
11445 Aug 28 20:51:34 witherspoon kernel: ---[ end trace d825f693b2118269 ]---
```

[w50.170901.063946.BMC_journalctl_nopager.txt](https://github.com/openbmc/openbmc/files/1270703/w50.170901.063946.BMC_journalctl_nopager.txt)

geissonator, 2017-09-01:
Hit again on w50, just getting this over and over in dmesg:
```
[ 6330.110000] ftgmac100 1e660000.ethernet eth0: no vlan ids left to set
[ 6339.370000] ftgmac100 1e660000.ethernet eth0: NCSI interface down
[ 6348.870000] ftgmac100 1e660000.ethernet eth0: NCSI interface down
[ 6351.630000] ftgmac100 1e660000.ethernet eth0: no vlan ids left to set
[ 6361.450000] ftgmac100 1e660000.ethernet eth0: NCSI interface down
[ 6371.510000] ftgmac100 1e660000.ethernet eth0: NCSI interface down
```

top doesn't show anything significant happening

geissonator, 2017-09-01:
A BMC reboot doesn't fix it.  We had to do a PDU cycle to get the network going again.

geissonator, 2017-09-01:
@sammj - this patch seemed to be what introduced this error statement:
https://patchwork.kernel.org/patch/9894999/
```
+	if (!vlan || index >= 0) {
+		netdev_printk(KERN_DEBUG, ndp->ndev.dev,
+			      "no vlan ids left to set\n");
+		return -1;
+	}
```

sammj, 2017-09-05:
Yep, you'll actually see that message on every configuration change. It's one of the states of the larger NCSI config state in the driver, so whenever a configuration of the NCSI channel occurs we'll see this once.

The fact we're seeing this repeat over and over indicates that the channel is for some reason getting put back into the configuration state. This is pretty odd but could be because of repeated AEN packets from the Broadcom controller. I'll need to sit down and find exactly what's changing the state.

sammj, 2017-09-08:
I've not been lucky enough to recreate this myself yet, but I've pushed a kernel patch to gerrit that includes my debugging. If we can recreate with that kernel we should have a much better idea how the NCSI stack is behaving here.
https://gerrit.openbmc-project.xyz/#/c/6545/

sammj, 2017-09-08:
> A BMC reboot doesn't fix it. We had to do a PDU cycle to get the network going again.

This is still very interesting - something more must be going on than just the NCSI stack stepping on its own toes for example.

If we get a recreate please post the entire kernel log, the context will be important.

gkeishin, 2017-09-08:
We have been trying with debug builds and fixes and tested, so far all recreated and the log is shared for further analysis. Hope we see this one fixed soon.

gkeishin, 2017-09-09:
There is no hard and fast rule to recreate this issue. it will show up either on Power on/ Reset reload those test.

geissonator, 2017-09-12:
@gkeishin and I finally got to root cause on this today by just testing builds in between working and non-working.  The root cause for the issue was in https://gerrit.openbmc-project.xyz/#/c/6158/ and the solution is in https://gerrit.openbmc-project.xyz/#/c/6629/.  I opened #2288 to see if there's any pci reset we actually need to do on witherspoon.  We verified that neither zaius or romulus have these set so pretty sure we don't need on witherspoon either.

rfrandse, 2017-09-12:
<https://gerrit.openbmc-project.xyz/6629> Remove pci_reset from witherspoon config
Resolves  #2228 NETDEV WATCHDOG: eth0 (ftgmac100): transmit queue 0 timed out

closed 2017-09-12 by williamspatrick