openbmc/openbmc #2229: [xCAT] On Factory Reset, hostname is not wiped out

whowutwut, 2017-08-30:
When running the command to FactoryReset, 

```
POST https: //172.11.140.5/xyz/openbmc_project/network/action/Reset -d {"data":[]}
```

If the hostname is changed, it is not reset to "witherspoon"

mtritz, 2017-08-30:
@ratagupt Is the hostname intended to be reset? My understanding has been that the factory reset simply resets existing interfaces to DHCP.

whowutwut, 2017-08-30:
The hostname is part of the `/network/config/` object, so why wouldn't all attributes get restored to a "factory" setting?

williamspatrick, 2017-08-30:
The hostname is used both as part of DHCP advertisement ("I would like to be this hostname") and internally for any case a 'hostname' would be used, such as FFDC identifiers.  systemd actually uses the words "static hostname" and "transient hostname" to define the advertised / desired hostname vs the assigned hostname (received from DHCP server).

We did not include the "static hostname" in the set of network settings that we cleared on network reset.  We only considered the network settings themselves as the items to be cleared.  I am not suggesting this is right or wrong, but just what was implemented.

There is also an IPMI command that sets the "hostname", part of the DCMI command set, which happens to not be part of networking.  If we cleared this as part of a network reset, it would also clear this IPMI setting.

I don't have a strong preference which way we go one this, except that we'd have to write some additional code to clear out the hostname because we are not currently doing it.

whowutwut, 2017-08-31:
Is this "hostname" planned to persist a firmware flash?  The reason I'm hitting this is that I'm starting to get confused with BMCs all being `witherspoon` and sometimes not sure if I'm on the right guy running the commands...  So I implemented support to set hostname via REST.  Would also be useful to customers to know they are on the correct BMC.  

 Resetting to DHCP and reboot, caused it to take the prio hostname that I set.

ratagupt, 2017-08-31:
@williamspatrick : we set the static hostname.

mtritz, 2017-08-31:
In the design call this morning, we determined that, although the hostname attribute is part of the /network/config/ object, it is considered a BMC setting and should be reset to "witherspoon" when a new image is flashed, not during the network factory reset.

It would be better if the documentation was clearer on what exactly the scope of a network factory reset is.

whowutwut, 2017-09-01:
@mtritz thanks for the update, sounds reasonable. Feel free to close out unless using this to track documentation

geissonator, 2017-09-06:
Moving out of phase 5 since this is just a doc thing now.

rfrandse, 2017-09-07:
<https://gerrit.openbmc-project.xyz/6542> Add README for FactoryReset interface
Resolves  #2229 [xCAT] On Factory Reset, hostname is not wiped out

closed 2017-10-02 by geissonator