openbmc/openbmc #2230: [xCAT] On FactoryReset, default network config file is not created

whowutwut, 2017-08-30:
When running the command to FactoryReset, 

```
POST https: //172.11.140.5/xyz/openbmc_project/network/action/Reset -d {"data":[]}
```

The `/etc/systemd/network/` directory is blank and there's no default config file for `00-bmc-eth0.network`

Although DHCP is enabled, so this is OK

geissonator, 2017-08-30:
Is this working as designed @ratagupt ?  We'd then create the file when someone comes in and configures a new IP?

mdmillerii, 2017-08-30:
Two questions: (1) does it change the rest objects? (2) Does default.network need [updating](https://github.com/openbmc/openbmc/issues/865#issuecomment-325794210)?

whowutwut, 2017-08-30:
@mdmillerii  The reset is OK and rest seems OK.  After the reboot of BMC, the interface does in fact come up and request DHCP IP address. 

The only thing is that there is no configuration file that we would expect in /etc/systemd/network.    If this is correct, then we can just close the issue.  I just wanted to make sure something is not missed here... 

For example, before, here's VLAN config...
```
root@xyz123:/tmp# uptime
 20:12:37 up  1:09,  load average: 0.89, 0.82, 0.73
root@xyz123:/tmp# ls -ltr /etc/systemd/network
total 2
-rw-r--r--    1 root     root            45 Aug 30 20:11 eth0.11.netdev
-rw-r--r--    1 root     root            70 Aug 30 20:11 00-bmc-eth0.network
-rw-r--r--    1 root     root            84 Aug 30 20:11 00-bmc-eth0.11.network
root@xyz123:/tmp#
```

Now I use REST to Reset...
```
[root@briggs01 ~]# rspconfig node-70e28414280d ip=dhcp
node-70e28414280d: [openbmc_debug] POST https: //172.11.140.5/login -d { "data": [ "root", "0penBmc" ] }
node-70e28414280d: [openbmc_debug] login_response 200 OK
node-70e28414280d: [openbmc_debug] POST https: //172.11.140.5/xyz/openbmc_project/network/action/Reset -d {"data":[]}
node-70e28414280d: [openbmc_debug] rspconfig_dhcp_response 200 OK
node-70e28414280d BMC: Setting IP to DHCP...
node-70e28414280d: [openbmc_debug] PUT https: //172.11.140.5/xyz/openbmc_project/state/bmc0/attr/RequestedBMCTransition -d {"data":"xyz.openbmc_project.State.BMC.Transition.Reboot"}
node-70e28414280d: [openbmc_debug] rpower_reset_response 200 OK
node-70e28414280d BMC: reboot
```

When the BMC comes back up...., it does get DHCP IP... but no config file is created ...    
```
root@xyz123:~# uptime
 20:14:59 up 1 min,  load average: 5.92, 1.61, 0.54
root@xyz123:~# ls -ltr /etc/systemd/network
total 0
root@xyz123:~#
```

rfrandse, 2017-08-31:
<https://gerrit.openbmc-project.xyz/6285> DHCP value was not getting updated correctly
Resolves  #2230 [xCAT] On FactoryReset, default network config file is not created

closed 2017-08-31 by williamspatrick