openbmc/openbmc #2231: [BMC Code update] Failed to start Create Volatile Files and Directories

gkeishin, 2017-08-30:
`NOTE: ` This is with new BMC code update 
`Driver Info`
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.9-94"
VERSION_ID="v1.99.9-94-gf9c946b-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.9-94"
BUILD_ID="v1.99.9"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-ba30e9ed152bab62f72da24986ee5ca22d6c9b07 #5 Wed Aug 30 12:00:23 CDT 2017 armv6l GNU/Linux
root@witherspoon:~#
```

```
root@witherspoon:~# systemctl status systemd-tmpfiles-setup.service | cat
● systemd-tmpfiles-setup.service - Create Volatile Files and Directories
   Loaded: loaded (/lib/systemd/system/systemd-tmpfiles-setup.service; static; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2017-08-28 18:07:40 UTC; 4min 27s ago
     Docs: man:tmpfiles.d(5)
           man:systemd-tmpfiles(8)
  Process: 1042 ExecStart=/bin/systemd-tmpfiles --create --remove --boot --exclude-prefix=/dev (code=exited, status=1/FAILURE)
 Main PID: 1042 (code=exited, status=1/FAILURE)

Aug 28 18:07:40 witherspoon systemd[1]: Starting Create Volatile Files and Directories...
Aug 28 18:07:40 witherspoon systemd-tmpfiles[1042]: /srv does not exist and cannot be created as the file system is read-only.
Aug 28 18:07:40 witherspoon systemd[1]: systemd-tmpfiles-setup.service: Main process exited, code=exited, status=1/FAILURE
Aug 28 18:07:40 witherspoon systemd[1]: Failed to start Create Volatile Files and Directories.
Aug 28 18:07:40 witherspoon systemd[1]: systemd-tmpfiles-setup.service: Unit entered failed state.
Aug 28 18:07:40 witherspoon systemd[1]: systemd-tmpfiles-setup.service: Failed with result 'exit-code'.
root@witherspoon:~#
```

- [x] Journald 
[New_BMC_code_update_journald.txt](https://github.com/openbmc/openbmc/files/1264629/New_BMC_code_update_journald.txt)

rfrandse, 2017-08-31:
<https://gerrit.openbmc-project.xyz/6329> Add dir "srv" to the root file system
Resolves  #2231 [BMC Code update] Failed to start Create Volatile Files and Directories

saqibkh, 2017-09-06:
This bug should have been closed since https://gerrit.openbmc-project.xyz/#/c/6329/ was merged.
@rfrandse Can you take a look at this?
Thanks

closed 2017-09-06 by williamspatrick