openbmc/openbmc #2232: occ pass through command failing and causing occ-control application failure

geissonator, 2017-08-30:
This works in older code but something is causing a fail in our latest code.

```
[witherspoon]# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v1.99.9"
VERSION_ID="ibm-v1.99.9-0-r2-0-g54c707f"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v1.99.9"
BUILD_ID="ibm-v1.99.9-0-r2"
```

Verified both occ's were active and hwmon sensors were working fine for them.  Issued this curl command which I got from @dkodihal:

curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST https://${BMC_IP}/org/open_power/control/occ0/action/Send -d '{"data": [[100,200]]}'
```
Aug 28 20:43:47 witherspoon openpower-occ-control[3962]: Writing to OCC failed.
Aug 28 20:43:47 witherspoon openpower-occ-control[3962]: terminate called after throwing an instance of 'sdbusplus::org::open_power::OCC::Device::Error::WriteFailure'
Aug 28 20:43:47 witherspoon openpower-occ-control[3962]:   what():  org.open_power.OCC.Device.Error.WriteFailure: Writing to OCC failed.
Aug 28 20:43:47 witherspoon systemd[1]: org.open_power.OCC.Control.service: Main process exited, code=dumped, status=6/ABRT
Aug 28 20:43:47 witherspoon systemd[1]: org.open_power.OCC.Control.service: Unit entered failed state.
Aug 28 20:43:47 witherspoon systemd[1]: org.open_power.OCC.Control.service: Failed with result 'core-dump'.
Aug 28 20:43:47 witherspoon systemd[1]: org.open_power.OCC.Control.service: Service hold-off time over, scheduling restart.
Aug 28 20:43:47 witherspoon systemd[1]: Stopped OpenPOWER OCC controller.
Aug 28 20:43:47 witherspoon systemd[1]: Starting OpenPOWER OCC controller...
Aug 28 20:43:47 witherspoon phosphor-gevent[1204]: Traceback (most recent call last):
Aug 28 20:43:47 witherspoon phosphor-gevent[1204]:   File "/usr/lib/python2.7/site-packages/bottle.py", line 862, in _handle
Aug 28 20:43:47 witherspoon phosphor-gevent[1204]:     return route.call(**args)
Aug 28 20:43:47 witherspoon phosphor-gevent[1204]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 800, in __call__
Aug 28 20:43:47 witherspoon phosphor-gevent[1204]:     return self.callback(*a, **kw)
Aug 28 20:43:47 witherspoon phosphor-gevent[1204]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 857, in wrap
Aug 28 20:43:47 witherspoon phosphor-gevent[1204]:     return callback(*a, **kw)
Aug 28 20:43:47 witherspoon phosphor-gevent[1204]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 1040, in __call__
Aug 28 20:43:47 witherspoon phosphor-gevent[1204]:     return self.callback(*a, **kw)
Aug 28 20:43:47 witherspoon phosphor-gevent[1204]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 1014, in wrap
Aug 28 20:43:47 witherspoon phosphor-gevent[1204]:     return self.to_jsonp(callback(*a, **kw))
Aug 28 20:43:47 witherspoon phosphor-gevent[1204]:   File "/usr/lib/python2.7/site-packages/bottle.py", line 1734, in wrapper
Aug 28 20:43:47 witherspoon phosphor-gevent[1204]:     rv = callback(*a, **ka)
Aug 28 20:43:47 witherspoon phosphor-gevent[1204]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 979, in wrap
Aug 28 20:43:47 witherspoon phosphor-gevent[1204]:     data = callback(*a, **kw)
Aug 28 20:43:47 witherspoon phosphor-gevent[1204]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 908, in wrap
Aug 28 20:43:47 witherspoon phosphor-gevent[1204]:     return callback(*a, **kw)
Aug 28 20:43:47 witherspoon phosphor-gevent[1204]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 939, in wrap
Aug 28 20:43:47 witherspoon phosphor-gevent[1204]:     return callback(*a, **kw)
Aug 28 20:43:47 witherspoon phosphor-gevent[1204]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 200, in __call__
Aug 28 20:43:47 witherspoon phosphor-gevent[1204]:     return getattr(self, 'do_' + request.method.lower())(**kw)
Aug 28 20:43:47 witherspoon phosphor-gevent[1204]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 337, in do_post
Aug 28 20:43:47 witherspoon phosphor-gevent[1204]:     return request.route_data['method'](*request.parameter_list)
Aug 28 20:43:47 witherspoon phosphor-gevent[1204]:   File "/usr/lib/python2.7/site-packages/dbus/proxies.py", line 145, in __call__
Aug 28 20:43:47 witherspoon phosphor-gevent[1204]:     **keywords)
Aug 28 20:43:47 witherspoon phosphor-gevent[1204]:   File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 651, in call_blocking
Aug 28 20:43:47 witherspoon phosphor-gevent[1204]:     message, timeout)
Aug 28 20:43:47 witherspoon phosphor-gevent[1204]: DBusException: org.freedesktop.DBus.Error.NoReply: Message recipient disconnected from message bus without replying
Aug 28 20:43:48 witherspoon phosphor-gevent[1204]: ::8009:be76:8e91:f402%1307749140 - - [2017-08-28 20:43:48] "POST /org/open_power/control/occ0/action/Send HTTP/1.1" 500 2438 0.919577
Aug 28 20:43:50 witherspoon systemd[1]: Started OpenPOWER OCC controller.
```

```
{
	"__CURSOR" : "s=ae7314c8394a40249cb28462feee8cf2;i=11c3;b=85d58cfa515a4907abb8a01dacb58903;m=2bc844b07;t=557d551a95087;x=27f5cbbbc4ce8e19",
	"__REALTIME_TIMESTAMP" : "1503948593254535",
	"__MONOTONIC_TIMESTAMP" : "11752721159",
	"_BOOT_ID" : "85d58cfa515a4907abb8a01dacb58903",
	"_MACHINE_ID" : "bf6c557a12b6419894ce4df23cf45864",
	"_HOSTNAME" : "witherspoon",
	"_UID" : "0",
	"_GID" : "0",
	"_SYSTEMD_SLICE" : "system.slice",
	"_TRANSPORT" : "journal",
	"_CAP_EFFECTIVE" : "3fffffffff",
	"PRIORITY" : "3",
	"CODE_FILE" : "/usr/include/phosphor-logging/log.hpp",
	"CODE_LINE" : "100",
	"CODE_FUNC" : "helper_log",
	"MESSAGE" : "Writing to OCC failed.",
	"TRANSACTION_ID" : "3414503068",
	"CALLOUT_ERRNO" : "22",
	"CALLOUT_DEVICE_PATH" : "/dev/occ1",
	"SYSLOG_IDENTIFIER" : "openpower-occ-control",
	"_PID" : "1193",
	"_COMM" : "openpower-occ-c",
	"_EXE" : "/usr/sbin/openpower-occ-control",
	"_CMDLINE" : "/usr/sbin/openpower-occ-control",
	"_SYSTEMD_CGROUP" : "/system.slice/org.open_power.OCC.Control.service",
	"_SYSTEMD_UNIT" : "org.open_power.OCC.Control.service",
	"_SYSTEMD_INVOCATION_ID" : "75f3549ecd7d487b861745b0ba574129",
	"_SOURCE_REALTIME_TIMESTAMP" : "1503948593253719"
}
```

I tried disabling hwmon since that was new in the code but hit the same issue with the pass through

dkodihal, 2017-08-31:
@geissonator when you say this works with the old code, did the same curl command work?

I doubt if there's a real issue here. Looking at https://github.com/openbmc/openpower-occ-control/blob/master/occ_pass_through.cpp#L81, an unhandled exception is thrown when the write fails in terms of not writing anything or writing less than expected.

This probably is happening because the data we're sending via 'Send' is probably invalid for the OCC, or there's a driver issue. Assigning to @vishwabmc to check if the exception should be handled. As it was not handled, the app restarted with the OCCs being not active.

vishwabmc, 2017-08-31:
`"CALLOUT_ERRNO" : "22"`

Errno: 22 is `Invalid Argument` and I suspect it may be an issue with the payload that is being passed -or- something with the driver. I will need @eddiejames to comment on this part.

The other part of app starting at Runtime and hence Active marked `False` need to be solved in the occ-control app.

My solution is : `See if the Host is up as part of app start and if so, make all active`

dkodihal, 2017-08-31:
'See if the host is up' - How can you do that reliably? The host may go down after you figured out it was up. Can't we just handle the exception with the write() fail instead of killing the app each time the write fails? I can understand doing that on an open() fail, but not write() fail.

vishwabmc, 2017-08-31:
So what if host goes down ?. We do set the Inactive anyway on power off path. I am trying to see some OperationalState + host state etc.. Need to talk to @geissonator 

But I am not understanding why the write must fail, if it fails, then it must be some real error.

geissonator, 2017-08-31:
Lets use this issue to specifically track why the amester pass through is not working.  We can create a separate bug to deal with the occ-control application restarting on error.

geissonator, 2017-08-31:
So first here's a valid pass through command (the first one we were using was not valid):
```
curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST https://${BMC_IP}/org/open_power/control/occ0/action/Send -d '{"data": [[0,0,0,1,20]]}'
```

The first run of this on wsbmc008 worked ok, but the host logged an error, and further runs of the same command failed.

Here's the output in the journal from the first run of the command (which succeeded)
```
Aug 30 23:05:27 witherspoon phosphor-gevent[1325]: ::80b9:bf76:8e91:f402%1154132756 - - [2017-08-30 23:05:27] "POST /org/open_power/control/occ0/action/Send HTTP/1.1" 200 235 7.501427
Aug 30 23:05:28 witherspoon ipmid[1331]: A host system event was received
Aug 30 23:05:28 witherspoon dbus[1069]: [system] Activating via systemd: service name='org.freedesktop.journal1' unit='dbus-org.freedesktop.journal1.service'
Aug 30 23:05:28 witherspoon systemd[1]: Starting Journal Dbus Service...
Aug 30 23:05:28 witherspoon dbus[1069]: [system] Successfully activated service 'org.freedesktop.journal1'
Aug 30 23:05:28 witherspoon systemd[1]: Started Journal Dbus Service.
Aug 30 23:05:30 witherspoon phosphor-fru-fault-monitor[1319]: /xyz/openbmc_project/logging/entry/58
```
 
The error from the host seems to map to these 1 of these 2 OCC errors:
```
|------------------------------------------------------------------------------|
 |                       Platform Event Log - 0x89000043                        |
 |------------------------------------------------------------------------------|
 |                                Private Header                                |
 |------------------------------------------------------------------------------|
 | Section Version          : 1                                                 |
 | Sub-section type         : 0                                                 |
 | Created by               : occc                                              |
 | Created at               : 00/4A/0000 99:EA:F5                               |
 | Committed at             : 00/4A/0000 99:EB:E5                               |
 | Creator Subsystem        : HostBoot                                          |
 | CSSVER                   :                                                   |
 | Platform Log Id          : 0x89000043                                        |
 | Entry Id                 : 0x89000043                                        |
 | Total Log Size           : 1976                                              |
 |------------------------------------------------------------------------------|
 |                                 User Header                                  |
 |------------------------------------------------------------------------------|
 | Section Version          : 1                                                 |
 | Sub-section type         : 0                                                 |
 | Log Committed by         : tmgt                                              |
 | Subsystem                : HostBoot                                          |
 | Event Scope              : Single Platform                                   |
 | Event Severity           : Unrecoverable Error                               |
 | Event Type               : Not Applicable                                    |
 | Return Code              : 0x00000000                                        |
 | Action Flags             : None declared                                     |
 |------------------------------------------------------------------------------|
 |                        Primary System Reference Code                         |
 |------------------------------------------------------------------------------|
 | Section Version          : 1                                                 |
 | Sub-section type         : 1                                                 |
 | Created by               : occc                                              |
 | SRC Format               : 0xE0                                              |
 | SRC Version              : 0x02                                              |
 | Virtual Progress SRC     : False                                             |
 | I5/OS Service Event Bit  : False                                             |
 | Hypervisor Dump Initiated: False                                             |
 | Power Control Net Fault  : False                                             |
 | Deconfigured             : False                                             |
 | Garded                   : False                                             |
 | Acknowledged             : False                                             |
 |                                                                              |
 | Valid Word Count         : 0x09                                              |
 | Reference Code           : BC8A2A03                                          |
 | Hex Words 2 - 5          : 000000E0 00000400 00000000 00200000               |
 | Hex Words 6 - 9          : 00000100 00000013 00000000 0E040041               |
 |                                                                              |
 |------------------------------------------------------------------------------|
 |                              User Defined Data                               |
[... 290 lines cut ...]
[... cut ...]
 | Section Version          : 1                                                 |
 | Sub-section type         : 1                                                 |
 | Created by               : errl                                              |
 | String data              : Hostboot Build ID: hostboot-c68be97/hbirt.bin     |
 |------------------------------------------------------------------------------|
```

A different system worked fine, so may be specific to this system.  Trying to get a 3rd system to be the tie breaker.

geissonator, 2017-09-01:
Turns out we had 2 issues here.  wsbmc008 just has some hardware issues.  Lots of logs from the host about both chips, and the OCC is not in a good state because of this.

The second issue was that bmc code added a new interface to /org/open_power/control/.  The amester script had assume it would only provide occ pass through interfaces and did not handle that new interface.  Team made a quick change to the amester script to support this and all was good.  Going to close this issue now.

closed 2017-09-01 by geissonator