openbmc/openbmc #2233: How to tell bitbake not to fetch?

FrankHome, 2017-08-31:
Hi All
Every time bitbake will to do_fetch for the internet,but now I need work in a place whithout network.  What Can I do ,Help me,Thanks a lot !

mdmillerii, 2017-09-02:
Look at [the yocto documentation](http://www.yoctoproject.org/docs/2.1/bitbake-user-manual/bitbake-user-manual.html#the-download-fetch) or in the raw xml chapter source of [the bitbake-user-manual-fetching.xml](https://github.com/openbmc/openbmc/blob/master/import-layers/yocto-poky/bitbake/doc/bitbake-user-manual/bitbake-user-manual-fetching.xml) about `PREMIRRORS` and `BB_NO_NETWORK` for a way to redirect the download an internal mirror.

There's probably more examples around, as `bitbake` supports shared caching.

FrankHome, 2017-09-04:
Thanks a lot

closed 2017-09-04 by FrankHome