openbmc/openbmc #2234: phosphor-image-updater: Error in mapper GetSubTreePath

gkeishin, 2017-08-31:
```
Aug 28 19:34:12 witherspoon phosphor-image-updater[1102]: Error in mapper GetSubTreePath
```
On boot to standby post BMC and PNOR code update.

`Driver Info`
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.9-99"
VERSION_ID="v1.99.9-99-geb86447"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.9-99"
BUILD_ID="v1.99.9"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-76ff93cf4565068e97b802e41748eb6745500606 #1 Wed Aug 30 20:33:11 UTC 2017 armv6l GNU/Linux
root@witherspoon:~# cat /media/pnor-ro-fd2cbe15/VERSION
IBM-witherspoon-sequoia-ibm-OP9_v1.18_1.32
        op-build-v1.18-74-g53f4235-dirty
        buildroot-2017.05-163-g8035745
        skiboot-v5.7-92-g0818df625e35
        hostboot-be278ac
        linux-4.12.5-openpower1-pc06e0a6
        petitboot-v1.5.1-p641e963
        machine-xml-9a28007-p5e6f272
        occ-71b5f68
        hostboot-binaries-cf51511
        capp-ucode-9c73e9f
        sbe-bcb079b
root@witherspoon:~#
```

- [x] Journald 
[journald_v1.99.9-99-geb86447_codeudate.txt](https://github.com/openbmc/openbmc/files/1265838/journald_v1.99.9-99-geb86447_codeudate.txt)

saqibkh, 2017-08-31:
Seems to be related to ItemUpdater::setBMCInventoryPath()
https://github.com/openbmc/phosphor-bmc-code-mgmt/blob/1eef62de893e619a8bc0c250e0b439178da11e8f/item_updater.cpp#L419

gtmills, 2017-08-31:
So the BMC inventory association would have been empty due to an error in GetSubTreePath.

gtmills, 2017-08-31:
Have you seen this more than once?

gkeishin, 2017-08-31:
So far latest Master build its pervasive on every boot or update.

rfrandse, 2017-08-31:
<https://gerrit.openbmc-project.xyz/6338> Add "After" mapper to BMC updater
Resolves  #2234 phosphor-image-updater: Error in mapper GetSubTreePath

gkeishin, 2017-09-23:
Good in Master too.

closed 2017-09-23 by gkeishin