openbmc/openbmc #2235: witherspoon-pseq-monitor.service: Main process exited on OS reboot test

gkeishin, 2017-08-31:
- [x] Boot to OS
- [x] Issue reboot from OS

`Journald snippet`
```
Aug 28 19:52:52 witherspoon systemd[1]: Started Witherspoon Power Sequencer Runtime Monitor.
Aug 28 19:52:52 witherspoon systemd[1]: Starting Power on bind ir35221 device driver devs...
Aug 28 19:52:52 witherspoon witherspoon-pseq-monitor[2350]: Invalid action
Aug 28 19:52:52 witherspoon witherspoon-pseq-monitor[2350]: Usage: witherspoon-pseq-monitor [options]
Aug 28 19:52:52 witherspoon witherspoon-pseq-monitor[2350]: Options:
Aug 28 19:52:52 witherspoon witherspoon-pseq-monitor[2350]:     --help                Print this menu
Aug 28 19:52:52 witherspoon witherspoon-pseq-monitor[2350]:     --action=<action>     Action: pgood-monitor
Aug 28 19:52:52 witherspoon witherspoon-pseq-monitor[2350]:     --interval=<interval> Time to allow PGOOD to come up
Aug 28 19:52:52 witherspoon systemd[1]: witherspoon-pseq-monitor.service: Main process exited, code=exited, status=1/FAILURE
Aug 28 19:52:52 witherspoon systemd[1]: witherspoon-pseq-monitor.service: Unit entered failed state.
Aug 28 19:52:52 witherspoon systemd[1]: witherspoon-pseq-monitor.service: Failed with result 'exit-code'.
```

`Systemctl snippet`
```

● witherspoon-pseq-monitor.service - Witherspoon Power Sequencer Runtime Monitor
   Loaded: loaded (/lib/systemd/system/witherspoon-pseq-monitor.service; static; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2017-08-28 19:52:52 UTC; 2min 44s ago
  Process: 2350 ExecStart=/usr/bin/env witherspoon-pseq-monitor -a runtime-monitor -i 1 (code=exited, status=1/FAILURE)
 Main PID: 2350 (code=exited, status=1/FAILURE)

Aug 28 19:52:52 witherspoon systemd[1]: Started Witherspoon Power Sequencer Runtime Monitor.
Aug 28 19:52:52 witherspoon witherspoon-pseq-monitor[2350]: Invalid action
Aug 28 19:52:52 witherspoon witherspoon-pseq-monitor[2350]: Usage: witherspoon-pseq-monitor [options]
Aug 28 19:52:52 witherspoon witherspoon-pseq-monitor[2350]: Options:
Aug 28 19:52:52 witherspoon witherspoon-pseq-monitor[2350]:     --help                Print this menu
Aug 28 19:52:52 witherspoon witherspoon-pseq-monitor[2350]:     --action=<action>     Action: pgood-monitor
Aug 28 19:52:52 witherspoon witherspoon-pseq-monitor[2350]:     --interval=<interval> Time to allow PGOOD to come up
Aug 28 19:52:52 witherspoon systemd[1]: witherspoon-pseq-monitor.service: Main process exited, code=exited, status=1/FAILURE
Aug 28 19:52:52 witherspoon systemd[1]: witherspoon-pseq-monitor.service: Unit entered failed state.
Aug 28 19:52:52 witherspoon systemd[1]: witherspoon-pseq-monitor.service: Failed with result 'exit-code'.
root@witherspoon:~#
```

`Driver Info`
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.9-99"
VERSION_ID="v1.99.9-99-geb86447"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.9-99"
BUILD_ID="v1.99.9"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-76ff93cf4565068e97b802e41748eb6745500606 #1 Wed Aug 30 20:33:11 UTC 2017 armv6l GNU/Linux
root@witherspoon:~# cat /media/pnor-ro-fd2cbe15/VERSION
IBM-witherspoon-sequoia-ibm-OP9_v1.18_1.32
        op-build-v1.18-74-g53f4235-dirty
        buildroot-2017.05-163-g8035745
        skiboot-v5.7-92-g0818df625e35
        hostboot-be278ac
        linux-4.12.5-openpower1-pc06e0a6
        petitboot-v1.5.1-p641e963
        machine-xml-9a28007-p5e6f272
        occ-71b5f68
        hostboot-binaries-cf51511
        capp-ucode-9c73e9f
        sbe-bcb079b
root@witherspoon:~#
```

- [x] Journald 
[journald_v1.99.9-99-geb86447_os_reboot.txt](https://github.com/openbmc/openbmc/files/1265890/journald_v1.99.9-99-geb86447_os_reboot.txt)

spinler, 2017-08-31:
Yea, unfortunately https://gerrit.openbmc-project.xyz/#/c/6185/ merged before the 12 commits that contain the code to supply this new service.  Fortunately it's just in .wants instead of .requires and won't break the power on.

geissonator, 2017-09-06:
Good to go on this one now that we got all merge @spinler ?

spinler, 2017-09-06:
None of those commits have been merged yet.  will be several more days

gkeishin, 2017-09-07:
Sure thing.. Let me know once merged and I ll verify and close this one up..

spinler, 2017-09-07:
The plan is I'm going to revert the https://gerrit.openbmc-project.xyz/#/c/6185/ commit to remove the service for now until the code behind it is all merged.  Will do that today.

spinler, 2017-09-07:
# **STATUS**

rfrandse, 2017-09-07:
<https://gerrit.openbmc-project.xyz/6534> Revert "Add service to monitor the UCD90160 at runtime"
Resolves  #2235 witherspoon-pseq-monitor.service: Main process exited on OS reboot test

closed 2017-09-08 by williamspatrick