openbmc/openbmc #2238: openpower-occ-control does not recover correctly if reset once host booted

geissonator, 2017-08-31:
See #2232 for details.  This is to track a recovery mechanism in openpower-occ-control for when it gets reset (especially on an error) when system is up and running.  Need to rediscover occ active.

vishwabmc, 2017-09-01:
I was talking to @williamspatrick yesterday about my initial code of returning error on failure and he seemed to be okay on that but wanted to discuss as a team.

Now we know that write() calls are not truly HW issues and can be injected by incorrect payload, we do not want to crash on such scenarios.

So, a failure on `read` and `write` can throw a `D-Bus error back to the caller` and a failure on `open` could result in crash.

If open has to fail, then it must be something real on the machine. Also, /dev/occ<> is only opened as a response to setting `OccActive`, so I think we may not get a crash at `open()` at all.

If a crash happens at open(), then it would still have a very tiny window where we will no longer get the `occActive` set by `Host` and we never set it. But will we ever hit this ?.

Other solution would be to send a HeartBeat command to host and see if it responds , but then it would delay the app start if other commands are already queued in the host bound command buffer.

Probably, we can serialize this property and be fine ?.  When the app starts, it can check if host is not up, then it can set to `false`. But what is even the reliable way to define host is off ?., We can key off of HostState but then we may end up keeping it `Active` if we fall in a window where Host has already made it `Inactive` while app was down and Host is still not in Off state.

williamspatrick, 2017-09-01:
I do not want to serialize the occ-active property.

I thought we talked before about automatically setting the OCCs active after a BMC-reset if the heartbeat passed?

vishwabmc, 2017-09-01:
Right.. But this is about app restarting at runtime

geissonator, 2017-09-01:
Yeah, on the r/r we get the heartbeat check for free via service files.  May be able to find a way to call the service and have it call the occ-active.sh if it passes, or may just need to call the heartbeat directly.

vishwabmc, 2017-09-01:
Okay, settle on HB command ?. but the only thing is it may delay the app start if there are more commands queued up. But that's a very remote possibility. So may be we shall settle on sending Heartbeat command ?

geissonator, 2017-09-01:
Yeah, should be fine

vishwabmc, 2017-09-02:
@geissonator @williamspatrick Do we agree on returning D-Bus error on `read` and `write` and only crash on `open` ?.

Also, if occ control app is anyway sending a Heartbeat command, we can remove occ-enable.sh. Agree ?

vishwabmc, 2017-09-04:
I have coded a fix for now to take care of setting OccActive on start. We do not have to wait on the conclusion on whether to return D-Bus error or not to solve this.

rfrandse, 2017-09-04:
<https://gerrit.openbmc-project.xyz/6398> Set OccActive property on application start
Fixes  #2238 openpower-occ-control does not recover correctly if reset once host booted

vishwabmc, 2017-09-12:
We had agreed that counting on `HeartBeat` should be good but then there is another concern raised now.

`I should not be setting the OccActive even if host responds to us if we have not booted enough and wait for the set operation by host`. This is almost like going back to start of the discussion on `how do I close all windows`. The HB was the option selected as a means to close all windows.

vishwabmc, 2017-09-19:
Perfect.. We discussed this at design meeting today and here is what it is : 
```
1/. Return the exception on error than crashing
2/. As part of OCC app start, check if it's already bound. If so, set the OccActive, else No
```

closed 2017-09-29 by williamspatrick