openbmc/openbmc #2239: Networkd:- After adding vlan and reboot the BMC,Networkd doesn't start.

ratagupt, 2017-09-01:
We have a strange problem,If I create the Vlan while the system is up,Everything works fine..Even I restart the openbmc network service during that time,it works fine..
Restarting the systemd-networkd works fine till the system is up.

But If I reboot the bmc..systemd-networkd doesn't come up,  there is some problem while calling the (ncsi_vlan_rx_add_vid)..Here are few interesting kernel traces.

```[   27.150000] Unable to handle kernel NULL pointer dereference at virtual address 000001c4
[   27.150000] pgd = 9cbdc000
[   27.150000] [000001c4] *pgd=00000000
[   27.150000] Internal error: Oops: 5 [#1] ARM
[   27.150000] CPU: 0 PID: 951 Comm: systemd-network Not tainted 4.10.17-ba30e9ed152bab62f72da24986ee5ca22d6c9b07 #1
[   27.150000] Hardware name: ASpeed SoC
[   27.150000] task: 9d44e480 task.stack: 9d144000
[   27.150000] PC is at ncsi_vlan_rx_add_vid+0xec/0x1c0
[   27.150000] LR is at vlan_vid_add+0x1e8/0x218
[   27.150000] pc : [<803f8260>]    lr : [<803f0cb4>]    psr: 60000013
[   27.150000] sp : 9d145b70  ip : 9d145b98  fp : 9d145b94
[   27.150000] r10: 9d0c89e0  r9 : 00000081  r8 : 9e61f000
[   27.150000] r7 : 00000028  r6 : 9e5aa000  r5 : 9e61f074  r4 : 9e61c000
[   27.150000] r3 : 00000000  r2 : 00000000  r1 : 80600b78  r0 : 9e5aa000
[   27.150000] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[   27.150000] Control: 00c5387d  Table: 9cbdc008  DAC: 00000055
[   27.150000] Process systemd-network (pid: 951, stack limit = 0x9d144188)
[   27.150000] Stack: (0x9d145b70 to 0x9d146000)
[   27.150000] 5b60:                                     9d12b0d0 00000081 00000028 9d12b080
[   27.150000] 5b80: 9e5aa000 00000001 9d145bcc 9d145b98 803f0cb4 803f8180 80433358 9e5aa000
[   27.150000] 5ba0: 00000001 9d06c800 9d06cc80 9e5aa000 00000028 805fb4e8 00000000 9d145c28
[   27.150000] 5bc0: 9d145bfc 9d145bd0 803f1ae4 803f0ad8 00000000 9d06c800 9d145c28 9e5aa000
[   27.150000] 5be0: 9d145cc8 805fb4e8 00000000 9d145c28 9d145c1c 9d145c00 803f35ec 803f1abc
[   27.150000] 5c00: 9d06c800 803f3518 9d117800 805d3c40 9d145da4 9d145c20 8034480c 803f3524
[   27.150000] 5c20: 9d145cc8 00000000 00000000 9d117848 00000000 00000000 00000000 00000000
[   27.150000] 5c40: 80448408 800dc980 9d145c58 9d117810 9d17ac00 9d145c40 9d145ca4 9d145c68
[   27.150000] 5c60: 8015be74 30687465 0030342e 00000000 00000000 00000000 9d117838 9d117844
[   27.150000] 5c80: 00000000 00000000 00000000 6e616c76 00000000 00000000 00000000 00000000
[   27.150000] 5ca0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   27.150000] 5cc0: 00000000 00000000 00000000 00000000 00000000 9d117820 00000000 9d11782c
[   27.150000] 5ce0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   27.150000] 5d00: 00000000 00000000 00000000 00000000 9d117834 00000000 00000000 00000000
[   27.150000] 5d20: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   27.150000] 5d40: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   27.150000] 5d60: 00000000 00000000 00000000 00000000 00000000 00000000 80357540 00000000
[   27.150000] 5d80: 9d117800 9d17ac00 00000000 9d145e1c 00000001 00000000 9d145ddc 9d145da8
[   27.150000] 5da0: 803429fc 80344274 9d145e27 014000c0 9d145df4 9d145dc0 800ddcd4 800c6ca0
[   27.150000] 5dc0: 9d117800 00000000 80342920 9d17ac00 9d145dfc 9d145de0 80359180 8034292c
[   27.150000] 5de0: 9d17ac00 9c966400 00000050 9d17ac00 9d145e14 9d145e00 80341d4c 803590c0
[   27.150000] 5e00: 9e4f9400 9c966400 9d145e4c 9d145e18 80357a34 80341d2c 8031b0c8 7fffffff
[   27.150000] 5e20: 9d144000 9d145ed8 9c966400 9d17ac00 00000000 00000050 00000000 00000000
[   27.150000] 5e40: 9d145eac 9d145e50 803581cc 803578c0 566474d8 00000058 801351e0 80051790
[   27.150000] 5e60: 9d145e84 00000008 800fbcac 9cb7e2c0 00000000 000003b7 000003e4 000003e1
[   27.150000] 5e80: 800fc2c4 00000000 00000000 9e1f2e20 00000122 8000a6e4 9d144000 00000000
[   27.150000] 5ea0: 9d145ebc 9d145eb0 803138fc 80357f50 9d145fa4 9d145ec0 80314c38 803138e4
[   27.150000] 5ec0: 9d145ee0 9d145ed0 00000000 00000000 5664b9a0 00000050 9d145f08 00000010
[   27.150000] 5ee0: 00000001 00000000 00000000 9d145ed8 00000000 9d145f00 00000000 00000000
[   27.150000] 5f00: 00000000 800573f0 00000010 00000000 00000000 00000000 2838cd66 00000000
[   27.150000] 5f20: 0c571a78 0093d1cc 28375434 00000000 a65a687a ffffffff 9d145f88 7eb039c4
[   27.150000] 5f40: 00000000 00000107 8000a6e4 9d144000 9d145f84 9d145f60 8005e3ec 800573a8
[   27.150000] 5f60: 0000001b 7eb039c4 00000000 00000107 8000a6e4 7eb039c4 00000008 00000000
[   27.150000] 5f80: 7eb039c4 8005f554 0000001b 7eb039ac 00000010 000003b7 00000000 9d145fa8
[   27.150000] 5fa0: 8000a540 80314b7c 7eb039ac 00000010 00000003 5664b9a0 00000050 00000000
[   27.150000] 5fc0: 7eb039ac 00000010 000003b7 00000122 566470a0 54b60168 00000000 56645780
[   27.150000] 5fe0: 7eb039a0 7eb03994 76ec8758 76deaa74 60000010 00000003 00000000 00000000
[   27.150000] [<803f8260>] (ncsi_vlan_rx_add_vid) from [<803f0cb4>] (vlan_vid_add+0x1e8/0x218)
[   27.150000] [<803f0cb4>] (vlan_vid_add) from [<803f1ae4>] (register_vlan_dev+0x34/0x1c8)
[   27.150000] [<803f1ae4>] (register_vlan_dev) from [<803f35ec>] (vlan_newlink+0xd4/0xe8)
[   27.150000] [<803f35ec>] (vlan_newlink) from [<8034480c>] (rtnl_newlink+0x5a4/0x68c)
[   27.150000] [<8034480c>] (rtnl_newlink) from [<803429fc>] (rtnetlink_rcv_msg+0xdc/0x1f8)
[   27.150000] [<803429fc>] (rtnetlink_rcv_msg) from [<80359180>] (netlink_rcv_skb+0xcc/0xe8)
[   27.150000] [<80359180>] (netlink_rcv_skb) from [<80341d4c>] (rtnetlink_rcv+0x2c/0x3c)
[   27.150000] [<80341d4c>] (rtnetlink_rcv) from [<80357a34>] (netlink_unicast+0x180/0x214)
[   27.150000] [<80357a34>] (netlink_unicast) from [<803581cc>] (netlink_sendmsg+0x288/0x348)
[   27.150000] [<803581cc>] (netlink_sendmsg) from [<803138fc>] (sock_sendmsg+0x24/0x34)
[   27.150000] [<803138fc>] (sock_sendmsg) from [<80314c38>] (SyS_sendto+0xc8/0xf0)
[   27.150000] [<80314c38>] (SyS_sendto) from [<8000a540>] (ret_fast_syscall+0x0/0x34)
[   27.150000] Code: e5922000 e1520005 1afffff8 e5942024 (e59221c4)```

ratagupt, 2017-09-01:
@sammj can take a look in this.

ratagupt, 2017-09-01:
Steps to reproduce the problem.
===========================
1) busctl call xyz.openbmc_project.Network /xyz/openbmc_project/network xyz.openbmc_project.Network.VLAN.Create VLAN su "eth0" 40

which internally creates the network file,and net dev file for the vlan.
2) reboot the bmc.

sammj, 2017-09-05:
NULL deference in ncsi_vlan_rx_add_vid() is interesting. Most likely it's this;
```
          ndp = TO_NCSI_DEV_PRIV(nd);
          ncf = ndp->hot_channel->filters[NCSI_FILTER_VLAN];      
```
The posted trace doesn't have it but in another trace I have we see the NULL dereference occurs a fair way into the `ndp` struct:
`[   27.150000] Unable to handle kernel NULL pointer dereference at virtual address 000001c4`
Which would match up with the position of the `hot_channel` field which we try retrieve the filter from.

I'll test and confirm a quick fix for that, although it may raise some issues with tracking the maximum allowed vlan ids.

sammj, 2017-09-06:
Proposed patch available here: http://patchwork.ozlabs.org/patch/810407/

shenki, 2017-09-06:
Patch made available for testing here: https://gerrit.openbmc-project.xyz/#/c/6427/

rfrandse, 2017-09-08:
<https://gerrit.openbmc-project.xyz/6514> kernel: MSN flash layout, iio and tach dts, ncsi bugfix
Fixes:  #2239 Networkd:- After adding vlan and reboot the BMC,Networkd doesn't start.

closed 2017-09-11 by williamspatrick