openbmc/openbmc #2240: ubi0:rwfs: Missing core file in BMC Dump

ojayanth, 2017-09-01:
BMC Dump manager watch core file created by systemd-coredump for core base BMV dump creation . In the system which have ubi0:rwfs based /var/lib/systemd/coredump directory , noticed that  existing inotify-event structure provide  only core file “#<inode>” format instead of actual file name for IN_CLOSE_WRITE event watch.  Also this file  name is getting changed systemd-coredump format with different events. This is causing dreport fail to copy the file in to BMC dump

ojayanth, 2017-09-01:
Refer https://github.com/openbmc/openbmc/issues/2214 for more details

ojayanth, 2017-09-08:
Proposed fix , till e identified the actual fix  from systemd-coredump is to mount the core path in tmpfs filesystems.

ojayanth, 2017-09-08:
# **Status**
09/08: Coding is in progress.

amboar, 2017-09-11:
Isn't the actual fix a kernel patch? Is there some discussion of why UBI behaves this way? I dug through this with @ojayanth but I wouldn't think the actual solution is to just use a tmpfs instead.

ojayanth, 2017-09-12:
@amboar this is temporary fix . Actual fix should be related Kernel or systemd-core dump side . Planing to open new issue to track the actual fix.

ojayanth, 2017-09-12:
https://github.com/openbmc/openbmc/issues/2287 used to track the actual fix.

rfrandse, 2017-09-12:
<https://gerrit.openbmc-project.xyz/6626> phosphor-debug-collector: Create /var/lib/systemd/coredump mount point
Resolves  #2240 ubi0:rwfs: Missing core file in BMC Dump

geissonator, 2017-09-14:
Workaround did not work, reopening

geissonator, 2017-09-18:
Core dumps disabled for now, moving to 5.1

ojayanth, 2017-09-18:
# **Status**

09/18: Code changes are ready. tmpfs based core path , inotify reporting incorrect core file name for dir level events. As per inotify documentation optional data is valid only for file events. Waiting for driver which have ubfi filesystem enabed.

ojayanth, 2017-09-22:
In ubfi filesystem inotify notification events behaves very diffrently like 
   - IN_CREATE notification event structure provides the core file name 
   - IN_CLOSE_WRITE notification event structure provides the INODE number instead of file name 
    - IN_CLOSE_NOWRITE  notification triggers "directory" level and  "Size of name field " shows 0 . inthis case file name trucates most of the cases. 
As a temporary solution we need to change the watch to IN_CREATE instead of IN_CLOSE_WRITE. 
In my testing, dreports package conatins the good core file with this event.

rfrandse, 2017-09-23:
<https://gerrit.openbmc-project.xyz/6835> Fix for missing core file in the Applicationcore type BMC dump
Resolves  #2240 ubi0:rwfs: Missing core file in BMC Dump

closed 2017-10-12 by geissonator