openbmc/openbmc #2241: [xCAT] Got curl(52) when run curl command to reboot BMC

xuweibj, 2017-09-01:
I tried serval times to reboot BMC. Most I can get "200 OK", but sometime I got response as below:

```
# curl -k -b cjar -X PUT -H "Content-Type: application/json" -d '{"data":"xyz.openbmc_project.State.BMC.Transition.Reboot"}' https: //10.6.9.254/xyz/openbmc_project/state/bmc0/attr/RequestedBMCTransition -d 
curl: (52) Empty reply from server 
```

Is this a bug?

mdmillerii, 2017-09-02:
An empty response likely means the socket was closed because the http server terminated before writing the response to the socket.

A small delay before requesting `systemd` reboot the machine may be in order?

xuweibj, 2017-09-11:
Hi @mdmillerii , I think it's better to return the response first and then close socket. For us, we did not know whether the request be received and successful. Could this be modified?

anoo1, 2017-09-12:
The current reboot interface in the org namespace waits 3s before reboot, we can probably add a sleep of 1s(?) to the xyz one @geissonator  ?

https://github.com/openbmc/skeleton/blob/dfda936510a56421370267ef5fc54b10b739c67c/bmcctl/control_bmc_obj.c#L25

```
    /* Wait a while before reboot, so the caller can be responded.
     */
    const char *reboot_command = "/bin/sh -c 'sleep 3;reboot'&";
```

xuweibj, 2017-09-20:
Hi @rfrandse , when this issue could be fixed and which version will it be merged in?

whowutwut, 2017-10-19:
This issue should resolve: https://github.com/xcat2/xcat-core/issues/3832 by fixing in the BMC , xCAT should respond with OK message

gtmills, 2018-08-14:
@geissonator Are we going to do anything here or can this be closed?

gtmills, 2018-08-14:
Both xcat issues were closed. 
xcat2/xcat-core#3832 and xcat2/xcat-core#4105

closed 2019-10-28 by stale[bot]