openbmc/openbmc #2244: kernel: Hangs during reboot, watchdog bites

amboar, 2017-09-01:
On:
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v1.99.8"
VERSION_ID="ibm-v1.99.8-98-g06734f31-r4-1-g01a1ad683df4"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v1.99.8"
BUILD_ID="ibm-v1.99.8-98-g06734f31-r4"
```
With kernel:
```
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-03afe5775baf06c3037d62cc2a15628079af38c9 #2 Fri Sep 1 10:15:30 ACST 2017 armv6l GNU/Linux
```

Having brought the BMC up, I issued `obmcutil poweron; reboot` at a root prompt. The subsequent startup hung in the kernel, and the watchdog bit:

```
U-Boot 2016.07 (Jul 26 2017 - 22:40:56 +0930)

       Watchdog enabled
DRAM:  496 MiB
Flash: 32 MiB
In:    serial
Out:   serial
Err:   serial
Net:   aspeednic#0
Error: aspeednic#0 address not set.

Hit any key to stop autoboot:  0
## Loading kernel from FIT Image at 20080000 ...
   Using 'conf@1' configuration
   Trying 'kernel@1' kernel subimage
     Description:  Linux kernel
     Type:         Kernel Image
     Compression:  uncompressed
     Data Start:   0x2008012c
     Data Size:    2074216 Bytes = 2 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: 0x80001000
     Entry Point:  0x80001000
     Hash algo:    sha1
     Hash value:   96e68ae9f980917ec16c7cae6caa940533947357
   Verifying Hash Integrity ... sha1+ OK
## Loading ramdisk from FIT Image at 20080000 ...
   Using 'conf@1' configuration
   Trying 'ramdisk@1' ramdisk subimage
     Description:  obmc-phosphor-initramfs
     Type:         RAMDisk Image
     Compression:  lzma compressed
     Data Start:   0x20282668
     Data Size:    1583904 Bytes = 1.5 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: unavailable
     Entry Point:  unavailable
     Hash algo:    sha1
     Hash value:   653c7f8133e97e827951704a3bb650112ff95c19
   Verifying Hash Integrity ... sha1+ OK
## Loading fdt from FIT Image at 20080000 ...
   Using 'conf@1' configuration
   Trying 'fdt@1' fdt subimage
     Description:  Flattened Device Tree blob
     Type:         Flat Device Tree
     Compression:  uncompressed
     Data Start:   0x2027a888
     Data Size:    32031 Bytes = 31.3 KiB
     Architecture: ARM
[... 164 lines cut ...]
[    4.490000] aspeed-i2c-bus 1e78a480.i2c-bus: i2c bus 13 registered, irq 34
[    4.490000] Driver for 1-wire Dallas network protocol.
[    4.490000] lm75 9-004a: hwmon0: sensor 'tmp275'

U-Boot 2016.07 (Jul 26 2017 - 22:40:56 +0930)

       Watchdog enabled
DRAM:  496 MiB
Flash: 32 MiB
In:    serial
Out:   serial
Err:   serial
Net:   aspeednic#0
Error: aspeednic#0 address not set.

Hit any key to stop autoboot:  0
```
Further, once userspace came up it reported:
```
root@witherspoon:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Off
```
The desired host state is Running.

chinaridinesh, 2017-09-22:
root@witherspoon:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Running
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v1.99.10"
VERSION_ID="ibm-v1.99.10-0-r7-0-gcce0f1b"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v1.99.10"
BUILD_ID="ibm-v1.99.10-0-r7"

I didnt see this problem any more. Has this got fixed in any other commit?

amboar, 2017-09-22:
As it involves the watchdog, the timing is heavily dependent on what's happening across u-boot and kernel initialisation. There are patches [here](http://patchwork.ozlabs.org/project/openbmc/list/?series=4020&state=%2A&archive=both) that help, but they are yet to be applied.