openbmc/openbmc #2248: openpower-occ-control detecting occ failure and crashing

geissonator, 2017-09-01:
We did some IPL testing with the latest code that now checks for OCC reported errors.  Looks like an error was detected.

```
1737 Aug 28 18:13:44 witherspoon openpower-occ-control[1199]: Error file updated
1738 Aug 28 18:13:44 witherspoon ipmid[1246]: Failed to call the method: Input/output error
1739 Aug 28 18:13:44 witherspoon ipmid[1246]: Dest: :1.13, Path: /org/openbmc/HostIpmi/1
1740 Aug 28 18:13:44 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@gpio\x2dfsi-cfam\x400\x2c0-sbefifo\x402400-occ\x401-occ\x2dhwmon\x401.service: Unit not needed anymore. Stopping.
1741 Aug 28 18:13:44 witherspoon systemd[1]: Stopping Phosphor Hwmon Poller...
1742 Aug 28 18:13:44 witherspoon systemd[1]: Stopped Phosphor Hwmon Poller.
1743 Aug 28 18:13:44 witherspoon systemd[1]: Started Phosphor Hwmon Poller.
1744 Aug 28 18:13:45 witherspoon openpower-occ-control[1199]: ERROR in getting service
1745 Aug 28 18:13:45 witherspoon openpower-occ-control[1199]: terminate called after throwing an instance of 'sdbusplus::xyz::openbmc_project::Common::Error::InternalFailure'
1746 Aug 28 18:13:45 witherspoon openpower-occ-control[1199]:   what():  xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally.
1747 Aug 28 18:13:45 witherspoon openpower-occ-control[1199]: The operation failed internally.
1748 Aug 28 18:13:45 witherspoon systemd[1]: org.open_power.OCC.Control.service: Main process exited, code=dumped, status=6/ABRT
1749 Aug 28 18:13:45 witherspoon systemd[1]: org.open_power.OCC.Control.service: Unit entered failed state.
1750 Aug 28 18:13:45 witherspoon systemd[1]: org.open_power.OCC.Control.service: Failed with result 'core-dump'.
1751 Aug 28 18:13:45 witherspoon ipmid[1246]: Failed to call the method: Input/output error
```

An error reported by the occ shouldn't cause occ-control app to fail right?

vishwabmc, 2017-09-01:
This was discovered before and the code is all up for merge review

https://gerrit.openbmc-project.xyz/#/c/6272/

geissonator, 2017-09-01:
Resolved in https://gerrit.openbmc-project.xyz/#/c/6272/

closed 2017-09-01 by geissonator