openbmc/openbmc #2249: BMC new code Activation: Error: "rwreset" not defined

gkeishin, 2017-09-01:
```
$ curl -b cjar -k -H "Content-Type: application/json" -X PUT -d '{"data":"xyz.openbmc_project.Software.Activation.RequestedActivations.Active"}' https://$BMC_IP//xyz/openbmc_project/software/6cf4227a/attr/RequestedActivation
{
  "data": null,
  "message": "200 OK",
  "status": "ok"
}
```

```
Aug 28 18:37:30 witherspoon phosphor-gevent[1936]: ::6019:c276:8e91:f402%1200925460 - - [2017-08-28 18:37:30] "PUT //xyz/openbmc_project/software/6cf4227a/attr/RequestedActivation HTTP/1.1" 200 169 0.879705
Aug 28 18:37:30 witherspoon obmc-flash-bmc[4028]: ## Error: "rwreset" not defined
Aug 28 18:37:31 witherspoon systemd[1]: Started Create BMC read-write ubi volume.
Aug 28 18:38:47 witherspoon phosphor-gevent[1936]: ::6019:c276:8e91:f402%1200925460 - - [2017-08-28 18:38:47] "GET /xyz/openbmc_project/software/6cf4227a HTTP/1.1" 200 649 0.129851
Aug 28 18:38:59 witherspoon obmc-flash-bmc[4027]: Volume ID 0, size 257 LEBs (16809856 bytes, 16.0 MiB), LEB size 65408 bytes (63.9 KiB), static, name "rofs-6cf4227a", alignment 1
Aug 28 18:39:45 witherspoon kernel: block ubiblock4_0: created from ubi4:0(rofs-6cf4227a)
Aug 28 18:39:45 witherspoon kernel[1125]: [ 1945.150000] block ubiblock4_0: created from ubi4:0(rofs-6cf4227a)
Aug 28 18:39:48 witherspoon obmc-flash-bmc[4113]: Volume ID 3, size 65 LEBs (4251520 bytes, 4.1 MiB), LEB size 65408 bytes (63.9 KiB), static, name "kernel-6cf4227a", alignment 1
Aug 28 18:39:56 witherspoon obmc-flash-bmc[4151]: [130B blob data]
Aug 28 18:39:57 witherspoon obmc-flash-bmc[4151]: [1.9K blob data]
Aug 28 18:39:57 witherspoon obmc-flash-bmc[4151]: [2.0K blob data]
Aug 28 18:39:57 witherspoon obmc-flash-bmc[4151]: [149B blob data]
Aug 28 18:39:57 witherspoon systemd[1]: Started Store read-only images 6cf4227a to BMC storage.
```

`Driver Info`
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.9-100"
VERSION_ID="v1.99.9-100-g11db66a"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.9-100"
BUILD_ID="v1.99.9"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-76ff93cf4565068e97b802e41748eb6745500606 #1 Thu Aug 31 11:12:47 CDT 2017 armv6l GNU/Linux
root@witherspoon:~#
```

gkeishin, 2017-09-01:
```
root@witherspoon:~# df
Filesystem           1K-blocks      Used Available Use% Mounted on
/dev/root                14592     14592         0 100% /
devtmpfs                215980         0    215980   0% /dev
tmpfs                   216120         0    216120   0% /dev/shm
tmpfs                   216120      8900    207220   4% /run
tmpfs                   216120         0    216120   0% /sys/fs/cgroup
tmpfs                     2808      1264      1544  45% /etc/machine-id
tmpfs                   216120     17836    198284   8% /tmp
tmpfs                   216120         0    216120   0% /media
ubi0:rwfs                 2808      1264      1544  45% /var
tmpfs                   216120         0    216120   0% /var/volatile
overlay                   2808      1264      1544  45% /etc
/dev/ubiblock8_0         19712     19712         0 100% /media/pnor-ro-fd2cbe15
ubi8:pnor-rw-8729dbf     13816       256     12812   2% /media/pnor-rw-8729dbf
ubi8:pnor-prsv             972        32       872   4% /media/pnor-prsv
ubi8:pnor-rw-fd2cbe15
                         13816       284     12788   2% /media/pnor-rw-fd2cbe15
root@witherspoon:~# mount
/dev/ubiblock0_0 on / type squashfs (ro,relatime)
devtmpfs on /dev type devtmpfs (rw,relatime,size=215980k,nr_inodes=53995,mode=755)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
proc on /proc type proc (rw,relatime)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
devpts on /dev/pts type devpts (rw,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /run type tmpfs (rw,nosuid,nodev,mode=755)
tmpfs on /sys/fs/cgroup type tmpfs (ro,nosuid,nodev,noexec,mode=755)
cgroup on /sys/fs/cgroup/systemd type cgroup (rw,nosuid,nodev,noexec,relatime,xattr,release_agent=/lib/systemd/systemd-cgroups-agent,name=systemd)
tmpfs on /etc/machine-id type tmpfs (ro,mode=755)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev)
debugfs on /sys/kernel/debug type debugfs (rw,relatime)
tmpfs on /media type tmpfs (rw,relatime)
ubi0:rwfs on /var type ubifs (rw,relatime)
tmpfs on /var/volatile type tmpfs (rw,relatime)
overlay on /etc type overlay (rw,relatime,lowerdir=/etc,upperdir=/var/persist/etc,workdir=/var/persist/etc-work)
/dev/ubiblock8_0 on /media/pnor-ro-fd2cbe15 type squashfs (ro,relatime)
ubi8:pnor-rw-8729dbf on /media/pnor-rw-8729dbf type ubifs (rw,relatime)
ubi8:pnor-prsv on /media/pnor-prsv type ubifs (rw,relatime)
ubi8:pnor-rw-fd2cbe15 on /media/pnor-rw-fd2cbe15 type ubifs (rw,relatime)
root@witherspoon:~#
```

gkeishin, 2017-09-01:
@saqibkh ^^^

gkeishin, 2017-09-01:
The image we were tying to activate is
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v1.99.9"
VERSION_ID="ibm-v1.99.9-94-g7d7329ba-test0-0-g7d7329b"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v1.99.9"
BUILD_ID="ibm-v1.99.9-94-g7d7329ba-test0"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-ba30e9ed152bab62f72da24986ee5ca22d6c9b07 #1 Wed Aug 30 13:49:17 CDT 2017 armv6l GNU/Linux
root@witherspoon:~#
```

rfrandse, 2017-09-07:
<https://gerrit.openbmc-project.xyz/6543> Suppress undefined rwreset error in obmc-flash-bmc
Resolves  #2249 BMC new code Activation: Error: "rwreset" not defined

closed 2017-09-12 by williamspatrick