openbmc/openbmc #2250: /root/ needs an overlay so ssh keys can be written to it

geissonator, 2017-09-01:
The new BMC filesystem layout made most things read-only now.  Similar to /etc/ we need an overlay on /root/ so that ssh keys can still be uploaded and written to it.

mdmillerii, 2017-09-01:
`/etc/passwd` says root's home is `/home/root` on w51 `ibm-v1.99.9-0-r4-0-gf0746a0`  

The location is movable, but we also need to consider policing the storage usage (there was at least one bug where the system failed to boot because `rwfs` was full from a file in `root`'s home directory).

One alternative is make the home a `tmpfs` and have a ~~script~~ unit  symlink  the `.ssh` directory from `/var`.

rfrandse, 2017-09-05:
<https://gerrit.openbmc-project.xyz/6423> Add /home tmpfs mount and symlink /home/root/.ssh to /var
Resolves  #2250 /root/ needs an overlay so ssh keys can be written to it

closed 2017-09-11 by williamspatrick

rfrandse, 2018-04-20:
<https://gerrit.openbmc-project.xyz/6372> Add /root/ overlayfs mount
Resolves  #2250 /root/ needs an overlay so ssh keys can be written to it

rfrandse, 2018-04-20:
<https://gerrit.openbmc-project.xyz/6371> Add /root/ overlayfs mount
Resolves  #2250 /root/ needs an overlay so ssh keys can be written to it