openbmc/openbmc #2254: Implement event logging mechanism

dkodihal, 2017-09-05:
Implement the design proposal chosen via #1856. A couple of proposals have been sent out to the community today for feedback.

dkodihal, 2017-09-08:
Proposals were discussed and very briefly, the idea is to map interesting d-bus signals to event objects.

A new app, phosphor-event-monitor could be implemented in the phosphor-dbus-monitor repo.

- [ ] Split the existing Entry interface into two interfaces - one should serve as the event interface, the combination of both should correspond to the existing Entry interface. The event interface would not have severity and resolved.
- [ ] Come up with a YAML based structure to map d-bus signals to event objects. The event d-bus objects should be in appropriate path namespaces, such as /xyz/openbmc_project/logging/events/boot/1 or /xyz/openbmc_project/logging/events/thermal/1. It should be possible to supply the yaml via a bitbake recipe.
A script will consume the code above to do the following. The script should be able to use existing mako templates in phosphor-dbus-monitor.
- [ ] The script should generate code to subscribe to signals of interest.
- [ ] The script should in the signal handler, create event d-bus objects, by creating corresponding event objects. Properties in the signal's message would have to be transformed to the event interface's metadata (the AdditionalData property).

Question - what events are we interested in?

I think we'd need to plan two more stories,
- for persisting events, to be able to have a history.
- add rate-limiting (process signal only is a certain time has passed since the last processing) capabilities to yaml and code.

rfrandse, 2017-10-14:
<https://gerrit.openbmc-project.xyz/7107> Implement event rotating
Resolves  #2254 Implement event logging mechanism

ratagupt, 2017-12-14:
Status

14/12 :- Started incorporating the review comments.

closed 2018-02-23 by geissonator