openbmc/openbmc #2255: Implement the ability to delete all BMC dumps in one shot

dkodihal, 2017-09-05:
Implement the ability to delete all BMC dumps in one shot.

ngorugan, 2017-09-07:
- []  Add a new interface Remove.interface.yaml at /xyz/openbmc_project/Dump with a method RemoveAll, which is suppose to delete all BMC dumps in one shot.

- [] Implement Remove interface

dkodihal, 2017-09-07:
@ngorugan based on your current "delete all error logs", the interface we finalize there should be usable here as well. We don't need a new interface.

ngorugan, 2017-09-07:
@dkodihal Okay, let me wait until Patrick replays back to your comments on my "delete all error logs" related yaml file changes, then I will update tasks of this Issue accordingly.

ngorugan, 2017-09-20:
# **STATUS**
09/20
Cod is under review

rfrandse, 2017-09-20:
<https://gerrit.openbmc-project.xyz/6611> Add implementation for delete all BMC Dumps in one shot
Resolves  #2255 Implement the ability to delete all BMC dumps in one shot

closed 2017-09-29 by williamspatrick