openbmc/openbmc #2257: phosphor-log-manager: Failed to find metadata

gkeishin, 2017-09-05:
```
Aug 28 19:35:18 witherspoon dbus[962]: [system] Successfully activated service 'org.freedesktop.journal1'
Aug 28 19:35:18 witherspoon systemd[1]: Started Journal Dbus Service.
Aug 28 19:35:19 witherspoon phosphor-log-manager[972]: Failed to find metadata
Aug 28 19:35:19 witherspoon phosphor-fru-fault-monitor[1221]: /xyz/openbmc_project/logging/entry/1
```

```
------------------------------------------------------------------------ 
BMC DATA : BMC Uptime	Executed : uptime
------------------------------------------------------------------------ 
 19:55:50 up 22 min,  load average: 0.57, 0.47, 0.64

------------------------------------------------------------------------ 
APPLICATION DATA : BMC state	Executed : /usr/sbin/obmcutil state
------------------------------------------------------------------------ 
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Running

------------------------------------------------------------------------ 
DRIVER INFO : FW Level	Executed : cat /etc/os-release
------------------------------------------------------------------------ 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.9-104"
VERSION_ID="v1.99.9-104-g243b1bd"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.9-104"
BUILD_ID="v1.99.9"

------------------------------------------------------------------------ 
DRIVER INFO : FW Timestamp	Executed : cat /etc/timestamp
------------------------------------------------------------------------ 
20170901222438
```
- [x] Journald 
[20170905144133653155_BMC_journalctl_nopager.txt](https://github.com/openbmc/openbmc/files/1277801/20170905144133653155_BMC_journalctl_nopager.txt)

anoo1, 2017-09-05:
From the journal log, the journal1 dbus object (which performs the journal sync during a commit) is started when the first commit is requested. The journal1 dbus object should probably be started when bmc boots so that it's ready to perform the sync on the first error log.

anoo1, 2018-02-05:
Per Brad, there may be a systemd setting in the journal service file that directs it to stop if it's not being used.

rfrandse, 2018-04-03:
<https://gerrit.openbmc-project.xyz/9910> log_manager: Create journal sync function
Closes  #2257  phosphor-log-manager: Failed to find metadata

closed 2018-04-18 by geissonator