openbmc/openbmc #2258: [xCAT] Changing DHCPEnabled from 0 to 1 obtains a NEW IP, REST and Ip addr show out of sync

cdeadmin, 2017-09-06:


rfrandse, 2017-09-06:
<https://gerrit.openbmc-project.xyz/6399> Network: Add network Monitor
Resolves  #2258 [xCAT] Changing DHCPEnabled from 0 to 1 obtains a NEW IP, REST and Ip addr show out of sync

prkatti1, 2017-09-07:
Some observations found:
1.
Pre-Condition: DHCP server configured in LAN
When we do factory reset: BMC comes up with 2 IP addresses, one from DHCP and one default address 169.254.81.215.

2. 
- Enable DHCP, BMC gets IP address from DHCP 10.5.5.2
- Disbale DHCP
- Check IP, there is no IP address
- Configure static IP 10.6.6.2
5. Check IP addresses on BMC, it'll list 2 IPs
    10.5.5.2 and 10.6.6.2
But REST and ip addr both show same output.

closed 2017-09-13 by williamspatrick