openbmc/openbmc #2259: phosphor-hwmon: enhancement to support adjust/offset values

mine260309, 2017-09-06:
In legacy python hwmon, `adjust` and `offset` are supported as hwmon dbus objects' interfaces.

In current `phosphor-hwmon`, there is no such support. But some systems do need to adjust the values read from hwmon sensors.

As discussed, `phosphor-hwmon` can be enhanced to support this:
1. There is no need to export adjust/offset to dbus;
2. The value is calculated internally, dubs object provides the calculated value.

For example:
```
OFFSET_temp1 = 1000
GAIN_in1=1.01
```
In the code, when it reads `OFFSET` or `GAIN`, it adds or multiple with the value:
```
value = value * gain + offset
```

mine260309, 2017-09-06:
Felix from MSI is working on this.

rfrandse, 2017-09-14:
<https://gerrit.openbmc-project.xyz/6428> Enhancement to support GAIN and OFFSET value for sensor reading
Resolves:  #2259 phosphor-hwmon: enhancement to support adjust/offset values

closed 2017-09-18 by williamspatrick