openbmc/openbmc #2260: Loss of bus arbitration: irq handled != irq. expected 0x00000008

gkeishin, 2017-09-06:
```
Aug 28 20:17:49 witherspoon kernel: aspeed-i2c-bus 1e78a100.i2c-bus: irq handled != irq. expected 0x00000008, but was 0x00000000
Aug 28 20:17:49 witherspoon kernel: leds-pca955x 3-0060: pca955x_read_ls: reg 0x3, err -11
Aug 28 20:17:49 witherspoon kernel[1071]: [ 2650.750000] aspeed-i2c-bus 1e78a100.i2c-bus: irq handled != irq. expected 0x00000008, but was 0x00000000
Aug 28 20:17:49 witherspoon kernel[1071]: [ 2650.760000] leds-pca955x 3-0060: pca955x_read_ls: reg 0x3, err -11
Aug 28 20:17:49 witherspoon kernel: aspeed-i2c-bus 1e78a100.i2c-bus: irq handled != irq. expected 0x00000008, but was 0x00000000
Aug 28 20:17:49 witherspoon kernel: bmp280 3-0077: failed to read temperature
Aug 28 20:17:49 witherspoon kernel[1071]: [ 2650.810000] aspeed-i2c-bus 1e78a100.i2c-bus: irq handled != irq. expected 0x00000008, but was 0x00000000
Aug 28 20:17:49 witherspoon kernel[1071]: [ 2650.810000] bmp280 3-0077: failed to read temperature
Aug 28 20:17:52 witherspoon kernel: aspeed-i2c-bus 1e78a100.i2c-bus: irq handled != irq. expected 0x00000008, but was 0x00000000
Aug 28 20:17:52 witherspoon kernel[1071]: [ 2653.600000] aspeed-i2c-bus 1e78a100.i2c-bus: irq handled != irq. expected 0x00000008, but was 0x00000000
Aug 28 20:17:56 witherspoon kernel: aspeed-i2c-bus 1e78a100.i2c-bus: irq handled != irq. expected 0x00000008, but was 0x00000000
Aug 28 20:17:56 witherspoon kernel[1071]: [ 2657.980000] aspeed-i2c-bus 1e78a100.i2c-bus: irq handled != irq. expected 0x00000008, but was 0x00000000
```

`Driver Info`
```
		[ Detailed Logs Captured Section ]
------------------------------------------------------------------------ 
APPLICATION DATA : BMC state	Executed : /usr/sbin/obmcutil state
------------------------------------------------------------------------ 
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Running

------------------------------------------------------------------------ 
DRIVER INFO : FW Level	Executed : cat /etc/os-release
------------------------------------------------------------------------ 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.9-104"
VERSION_ID="v1.99.9-104-g243b1bd"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.9-104"
BUILD_ID="v1.99.9"

------------------------------------------------------------------------ 
DRIVER INFO : FW Timestamp	Executed : cat /etc/timestamp
------------------------------------------------------------------------ 
20170901222438

------------------------------------------------------------------------ 
BMC DATA : BMC Uptime	Executed : uptime
------------------------------------------------------------------------ 
 20:37:20 up  1:03,  load average: 8.84, 5.61, 3.22

------------------------------------------------------------------------ 
BMC DATA : BMC OS	Executed : uname -a
------------------------------------------------------------------------ 
Linux witherspoon 4.10.17-76ff93cf4565068e97b802e41748eb6745500606 #1 Fri Sep 1 22:23:19 UTC 2017 armv6l GNU/Linux
```

```
        "/xyz/openbmc_project/software/8764baa3": {
            "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
            "ExtendedVersion": "op-build-v1.7-1321-g6eef393-dirty,buildroot-2017.05-163-g8035745,skiboot-v5.8-rc1-pd9d2012,hostboot-c68be97,linux-4.12.8-openpower1-p919e663,petitboot-v1.5.1-p848cad7,machine-xml-c1e49a6-pbb9f626,occ-854999a,hostboot-binaries-67a15fd,capp-ucode-9c73e9f,sbe-4a62126-p180ac2a",
            "Path": "",
            "Priority": 0,
            "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.Host",
            "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
            "Version": "IBM-witherspoon-sequoia-ibm-OP9_v1.18_1.54",
            "associations": [
                [
                    "activation",
                    "inventory",
                    "/xyz/openbmc_project/inventory/system/chassis"
                ]
            ]
        },
```

- [x] Journald log 
[20170906060058060349_BMC_journalctl_nopager.txt](https://github.com/openbmc/openbmc/files/1280277/20170906060058060349_BMC_journalctl_nopager.txt)
- [x] dmesg log 
[20170906060058060349_BMC_dmesg.txt](https://github.com/openbmc/openbmc/files/1280287/20170906060058060349_BMC_dmesg.txt)

gkeishin, 2017-09-06:
@amboar I couldn't see Eric in the list to assign this one. Hope u could help me bridge to him, Thanks

amboar, 2017-09-06:
@gkeishin I highly doubt Erich is on github.

amboar, 2017-09-06:
So unpacking this a bit, we have a hardware problem with devices on this bus. Specifically, problems with the topology of the fan card.

The errors that you have observed here are reported by the Aspeed bus master for hardware I2C4 (i2c3 in Linux). The master driver is reporting:

* Loss of arbitration (0x8)
* Abnormal Start/Stop condition (0x20)

The bus is meant to be single-master. Both of these conditions indicate that another device is attempting to master the bus in conflict with the Aspeed.

We need the hardware guys to investigate and determine what chip(s) on the bus is triggering this.

This problem will cause all manner of instability of we do not have a concrete answer for why it is happening.

@geissonator this is exactly the broader problem I was discussing with you yesterday.

amboar, 2017-09-09:
I feel like the bug label is somewhat misleading because the cause is not openbmc's fault; the software (kernel) is not misbehaving, rather it's reporting an error condition that appears to be ongoing. This cause appears to be a hardware problem associated with a particular fan control design.

amboar, 2017-09-09:
On the flip side, the issue isn't describing a feature either, but I think that's a false dichotomy.

gkeishin, 2017-09-18:
So anything we need to do on this one @amboar  or I can close this up..

amboar, 2017-09-19:
The investigation is ongoing. I don't see any harm in leaving it open.

gkeishin, 2017-09-19:
sure thing,,, thanks

gtmills, 2018-10-12:
@amboar Can we close?

amboar, 2018-11-08:
Well, I'm not aware of any current issues with the ASPEED I2C master driver; we've got work-arounds in place for the MAX31785 which seem to be working for whatever is wrong with it or the bus it's on. Given the lack of bugs/complaints, I'm going to close this.

closed 2018-11-08 by amboar