openbmc/openbmc #2261: New BMC code update: same version install on uploading different new image

gkeishin, 2017-09-06:
When we upload different images, why does it says same version installed
```
Aug 30 19:18:31 witherspoon phosphor-gevent[1168]: ::60a9:ba76:8e91:f402%1235725076 - - [2017-08-30 19:18:31] "POST /upload/image HTTP/1.1" 200 169 6.606602
Aug 30 19:18:31 witherspoon phosphor-version-software-manager[1125]: Untaring
Aug 30 19:18:32 witherspoon phosphor-image-updater[1080]: Software Object with the same version already exists
```

```
  "data": {
    "/xyz/openbmc_project/software/48bfac44": {
      "Activation": "xyz.openbmc_project.Software.Activation.Activations.Ready",
      "Path": "/tmp/images/48bfac44",
      "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.BMC",
      "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
      "Version": "v1.99.9-100-g11db66a",
      "associations": [
        [
          "activation",
          "inventory",
          ""
        ]
      ]
    },
    "/xyz/openbmc_project/software/48bfac44/activation": {
      "endpoints": [
        ""
      ]
    },
    "/xyz/openbmc_project/software/6cf4227a": {
      "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
      "Path": "",
      "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.BMC",
      "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
      "Version": "ibm-v1.99.9-94-g7d7329ba-test0-0-g7d7329b",
      "associations": [
        [
          "activation",
          "inventory",
          ""
        ]
      ]
    },
    "/xyz/openbmc_project/software/6cf4227a/activation": {
      "endpoints": [
        ""
      ]
    },
	
```

- [x] Journald 
[journald_BMC_update.txt](https://github.com/openbmc/openbmc/files/1281247/journald_BMC_update.txt)

saqibkh, 2017-10-18:
This is how the code should behave once the code referenced below gets merged.

-  If a user upload an image with version that doesn't exist, then there will be any log in the journal
-  If a user uploads an image with a version that is already set to Ready, then there will be a log in the journal stating that an image with same version already exists.
-  If a user uploads an image with a version that is already set to Active, then were will be no log.
   This is the expected behavior since we are thinking of adding the feasibility (in near future) to be able 
   to update to the same version over and over again (as was the case with old BMC).

gkeishin, 2017-10-18:
Thanks @saqibkh

rfrandse, 2017-10-18:
<https://gerrit.openbmc-project.xyz/7384> Revert "item_updater: Add a log to indicate duplicate image upload."
Resolves  #2261 New BMC code update: same version install on uploading different new image

closed 2017-10-25 by geissonator