openbmc/openbmc #2263: phosphor mapper failure

gkeishin, 2017-09-06:
```
Aug 28 18:08:24 witherspoon phosphor-mapper[949]: Traceback (most recent call last):
Aug 28 18:08:24 witherspoon phosphor-mapper[949]:   File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 604, in msg_reply_handler
Aug 28 18:08:24 witherspoon phosphor-mapper[949]:     reply_handler(*message.get_args_list(**get_args_opts))
Aug 28 18:08:24 witherspoon phosphor-mapper[949]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/server.py", line 143, in <lambda>
Aug 28 18:08:24 witherspoon phosphor-mapper[949]:     path, x),
Aug 28 18:08:24 witherspoon phosphor-mapper[949]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/server.py", line 107, in _gmo_callback
Aug 28 18:08:24 witherspoon phosphor-mapper[949]:     self.check_done()
Aug 28 18:08:24 witherspoon phosphor-mapper[949]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/server.py", line 84, in check_done
Aug 28 18:08:24 witherspoon phosphor-mapper[949]:     callback(service, self.results)
Aug 28 18:08:24 witherspoon phosphor-mapper[949]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/server.py", line 289, in discovery_callback
Aug 28 18:08:24 witherspoon phosphor-mapper[949]:     self.add_items(owner, items)
Aug 28 18:08:24 witherspoon phosphor-mapper[949]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/server.py", line 460, in add_items
Aug 28 18:08:24 witherspoon phosphor-mapper[949]:     self.update_interfaces(path, str(owner), old=[], new=items)
Aug 28 18:08:24 witherspoon phosphor-mapper[949]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/server.py", line 452, in update_interfaces
Aug 28 18:08:24 witherspoon phosphor-mapper[949]:     new_assoc = new[iface]['associations']
Aug 28 18:08:24 witherspoon phosphor-mapper[949]: KeyError: 'associations'
```

```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.9-111"
VERSION_ID="v1.99.9-111-g972e482"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.9-111"
BUILD_ID="v1.99.9"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-76ff93cf4565068e97b802e41748eb6745500606 #2 Wed Sep 6 08:42:36 CDT 2017 armv6l GNU/Linux
root@witherspoon:~#

```

- [x] Journald 
[boot_standby_crash_v1.99.9-111-g972e482.txt](https://github.com/openbmc/openbmc/files/1281781/boot_standby_crash_v1.99.9-111-g972e482.txt)

gkeishin, 2017-09-06:
Master is also broken now in latest build
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.9-109"
VERSION_ID="v1.99.9-109-ge993d48"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.9-109"
BUILD_ID="v1.99.9"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-76ff93cf4565068e97b802e41748eb6745500606 #1 Wed Sep 6 16:34:27 UTC 2017 armv6l GNU/Linux
root@witherspoon:~#
```

rfrandse, 2017-09-07:
<https://gerrit.openbmc-project.xyz/6531> Fix bug in mapper for associations KeyError
Resolves  #2263 phosphor mapper failure on boot to standby

closed 2017-09-08 by williamspatrick