openbmc/openbmc #2264: Software Object: Provide method to delete all versions

anoo1, 2017-09-06:
Currently a factory reset removes the read-write portion of versions, but does not delete all the versions on the system.
The user can individually delete versions but manufacturing for example would want to be able to 'clean' up a machine of all versions uploaded and activated on the system before shipping.

mtritz, 2017-09-07:
This needs a bit of design discussion. In my opinion, it would make sense to simply...

- [ ] Implement the generic FactoryReset under xyz.openbmc_project.Software.Version
- [ ] Write a Reset override function that, when called, clears all versions except the active one

rfrandse, 2017-09-21:
<https://gerrit.openbmc-project.xyz/6737> BMC Updater: Add function to delete all versions
Resolves  #2264 Software Object: Provide method to delete all versions

closed 2017-09-29 by williamspatrick