openbmc/openbmc #2265: Network:- Enable Zeroconf

ratagupt, 2017-09-07:
We need to make sure while enabling the zeroconf that IPMI should behave correctly.
If on the system there are multiple ip address objects then we should not return the Zeroconfing address.
But if there is no other IP address on the system except zeroconfig then we can return the zeroconf address.

ZeroConf can be enabled on the system by writing the Following parameter under Nerwork Section in network file.
LinkLocalAddressing = yes.

ngorugan, 2017-09-07:
- [ ]  Enable zeroconf in network file,While we write the configuration file.
- [ ] Make the changes in IPMI so that we should not return the zeroconfig address while there are other IP address.
- [ ] Write a test case in phosphor-networkd to verify that parameter is correctly written.

ngorugan, 2017-09-21:
# **STATUS**
09/21
Cod is under review

rfrandse, 2017-09-21:
<https://gerrit.openbmc-project.xyz/6810> Don't return zeroconfig address if other IP address is present
Resolves  #2265 Network:- Enable Zeroconf

closed 2017-10-12 by geissonator