openbmc/openbmc #2269: OCC object attributes inaccessible via REST

gkeishin, 2017-09-07:
```
/org/open_power/control/enumerate
{
    "/org/open_power/control/host0": {},
    "/org/open_power/control/occ0": {
        "OccActive": 0
    },
    "/org/open_power/control/occ1": {
        "OccActive": 1
    }
}
```
Enumerate shows the OCC objects

Accessing the occ attribute

```
/org/open_power/control/occ1/attr/OccActive
{
    "description": "The specified property cannot be found: 'OccActive'"
}

```

And 
```
/org/open_power/control/occ1/
[]
```

vishwabmc, 2017-09-08:
@gkeishin `OccActive` is supposed to be set only by the Host via IPMI anyway and I don't see any reason why an external client has any interest in it.

This does not defend the `I can not see attribute via REST` but I am saying the current behaviour does not affect the production code.

bradbishop, 2017-09-11:
this seems like a rest server bug if the properties show up with enumerate but not when you access them directly.

bradbishop, 2017-09-12:
fix here: https://gerrit.openbmc-project.xyz/#/c/6602/

gkeishin, 2017-09-12:
With the @bradbishop fix.. looking good, Tanks
```
$ ./rest_cmd -i xx.xx.xx.xx -o GET -u /org/open_power/control/enumerate
Executing: GET  /org/open_power/control/enumerate

/org/open_power/control/enumerate
{
    "/org/open_power/control/host0": {},
    "/org/open_power/control/occ0": {
        "OccActive": 0
    },
    "/org/open_power/control/occ1": {
        "OccActive": 0
    }
}

$ ./rest_cmd -i xx.xx.xx.xx -o GET -u /org/open_power/control/occ0
Executing: GET  /org/open_power/control/occ0

/org/open_power/control/occ0
{
    "OccActive": 0
}

$ ./rest_cmd -i xx.xx.xx.xx -o GET -u /org/open_power/control/occ1
Executing: GET  /org/open_power/control/occ1

/org/open_power/control/occ1
{
    "OccActive": 0
}

$ ./rest_cmd -i xx.xx.xx.xx -o GET -u /org/open_power/control/occ0/attr/OccActive
Executing: GET  /org/open_power/control/occ0/attr/OccActive

/org/open_power/control/occ0/attr/OccActive
0

$ ./rest_cmd -i xx.xx.xx.xx -o GET -u /org/open_power/control/occ1/attr/OccActive
Executing: GET  /org/open_power/control/occ1/attr/OccActive

/org/open_power/control/occ1/attr/OccActive
0

```

closed 2017-09-12 by williamspatrick