openbmc/openbmc #2270: [xCAT] Host version is not displayed after reboot of BMC with ubi layout.

gurevichmark, 2017-09-07:
After rebooting BMC with ubi layout observed that only BMC versions are displayed, no previously activated HOST versions are displayed.

If a new HOST image is uploaded, it is missing most of its attributes and is displayed as :
```
 "/xyz/openbmc_project/software/32f129cd": {
      "Path": "/tmp/images/32f129cd", 
      "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.Host", 
      "Version": "IBM-witherspoon-redbud-ibm-OP9_v1.17_1.68"
    },
```

anoo1, 2017-09-12:
@gurevichmark is this still an issue? There has been a few fixes merged in the last couple days.
@saqibkh could you take a look?

gurevichmark, 2017-09-12:
Works on the latest build I got.

closed 2017-09-12 by gurevichmark