openbmc/openbmc #2272: Enable QEMU runs against witherspoon in devops pipeline

geissonator, 2017-09-07:
There's a script, https://github.com/openbmc/openbmc-build-scripts/blob/master/run-qemu-robot-test.sh, which launches QEMU and then runs a set of automated robot tests against that QEMU session.  It stopped working a while ago for a variety of reasons, this story is to bring it back and make it better.

- [ ] Build a QEMU image that support witherspoon (or romulus)
- [ ] Boot up that QEMU session and work with @gkeishin to get a robot test suite in place that will verify the basics (REST api's, IPMI api's, bmc, chassis, and host states, ...).
- [ ] Integrate the 2 steps above into run-qemu-robot-test.sh so we can launch it against all generated CI images on openpower.xyz

We've had a lot of painful regressions make it through into master that would have been caught with the above items in place.

chinaridinesh, 2017-09-08:
Working with George for robot test suite.

gkeishin, 2017-09-08:
The old CI has the tags which were removed.. so we need come up with a new CI_QEMU list for it I ll work on that and we'll need the Qemu script to run the CI run on it.

amboar, 2017-09-09:
Please make use of the palmetto-bmc and romulus-bmc qemu machines as part of this effort

legoater, 2017-09-11:
Adding a witherspoon-bmc machine  would be interesting as there is a second flash on the witherspoon system. This is a very small extension to QEMU that I have already done for testing purpose. If this is of some interest, we can merge the patch.

amboar, 2017-09-11:
@legoater send it to the list? I'm keen to put it in (along with a witherspoon-bmc).

geissonator, 2017-09-11:
Ahh, good point on the witherspoon support.  And yes, having it support witherspoon would be awesome.  Until we get that in, @chinaridinesh - please focus on making the scripts generic to support whatever the user inputs for a systems, and get romulus working first.  Ideally we'd start up QEMU for all supported systems during CI.

legoater, 2017-09-11:
Please check : http://patchwork.ozlabs.org/patch/812293/

we lack models for some I2C devices : 
 * ir35221
 * dps310
 * max31785a
 * bmp280
 * ucd90160
 * ...

Simple I2C models would be nice to have some basic interaction from the kernel. 

FSI support would be great of course.

amboar, 2017-09-12:
I've updated openbmc/qemu@master to 2.10.0 and applied the witherspoon patch @legoater sent to the list. Building openbmc/qemu@61d5eae7e2f89a2f4c1b2398e9c2366344dbf1ce will now provide a `witherspoon-bmc` argument to the `-machine` option.

legoater, 2017-09-12:
thx. I did some updates on the openbmc/QEMU wki :  https://github.com/openbmc/qemu/wiki

ojayanth, 2017-10-16:
@amboar 
Able to get BMC statndby with latest openbmc image and qemu.
./qemu-system-arm -m 256 -M witherspoon-bmc -nographic -drive file=< build_path>/tmp/deploy/images/witherspoon/obmc-phosphor-image-witherspoon-20171006023754.ubi.mtd,format=raw,if=mtd -net nic -net user,hostfwd=:127.0.0.1:2222-:22,hostfwd=:127.0.0.1:2443-:443,hostname=qemu

amboar, 2017-10-16:
@ojayanth are you voluntering to be our CI system? :D

geissonator, 2018-01-03:
@legoater (or anyone) - is there any specific reason we don't add a .bbappend into our meta-phosphor or meta-openbmc-machines layer and just point to our openbmc repo (i.e. https://github.com/openbmc/qemu).  It seems like our normal flow in openbmc (like with the kernel) is to have our own repo for faster development feedback, while things slowly get upstreamed.

I'm asking, because it would be a lot easier to use your stuff in our automation env if it was just included in the default bitbake of the image.  The alternative is I'll need to swizzle up a jenkins job that extracts your code, generates a docker image with it contained within it, and then executes the qemu tests using this special docker container with the images from the jobs (i.e.  witherspoon, palmetto, romulus)

geissonator, 2018-01-04:
Although our intern did write a nice script to build a docker container and build the tool out of the openbmc/qemu repo (https://github.com/openbmc/openbmc-build-scripts/blob/master/qemu-build.sh).  I could just configure a jenkins job to run this periodically and have the jobs that need the tool just wget the latest successful artifact from it.

amboar, 2018-01-04:
It's not quite as easy as what you suggest above, which is why it hasn't been done. This has been on my radar forever; maybe I can cook something up in the next couple of weeks. You need to configure qemu appropriately, which is the main challenge. Yocto revamped their qemu support in the 2.2 release to make life easier for BSPs, which was why I hadn't done it prior. Now that we're on 2.2 it has just been a matter of trying to find the time to configure and integrate the necessary pieces.

rfrandse, 2018-01-23:
<https://gerrit.openbmc-project.xyz/8431> Launch appropriate QEMU based on input image
Resolves  #2272 Enable QEMU runs against System in devops pipeline

closed 2018-01-29 by geissonator