openbmc/openbmc #2274: Software Version ID Enhancements

anoo1, 2017-09-07:
The software version id is a hash calculated over the version string of the image.

Some requested enhancements have been the ability for the user to be able to calculate
externally that id so that there's no need to enumerate the dbus object to find it after uploading an
image.

Also calculate the hash over perhaps the whole tar file so that the version id is different even if
the version string is the same, like in the case of dirty git commits.

amboar, 2017-09-21:
Why did this go to 3.0? Why can't we do it earlier? It's not like we'd break things by enabling users to generate their own hashes.

zahrens, 2017-11-13:
This will be implemented with #2290 and #2291

closed 2017-11-13 by zahrens