openbmc/openbmc #2277: BMC version property missing inSoftware Manager object

gkeishin, 2017-09-08:
With the latest build
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.9-123"
VERSION_ID="v1.99.9-123-gdbf5de0"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.9-123"
BUILD_ID="v1.99.9"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-76ff93cf4565068e97b802e41748eb6745500606 #1 Fri Sep 8 16:23:02 UTC 2017 armv6l GNU/Linux
root@witherspoon:~# uptime
 19:40:52 up 7 min,  load average: 0.39, 0.89, 0.61
root@witherspoon:~#

root@witherspoon:~# df
Filesystem           1K-blocks      Used Available Use% Mounted on
dev                     215208         0    215208   0% /dev
tmpfs                   216124     14356    201768   7% /run
/dev/mtdblock4           14720     14720         0 100% /run/initramfs/ro
/dev/mtdblock5            4096       308      3788   8% /run/initramfs/rw
cow                       4096       308      3788   8% /
tmpfs                   216124         0    216124   0% /dev/shm
tmpfs                   216124         0    216124   0% /sys/fs/cgroup
tmpfs                   216124         0    216124   0% /tmp
tmpfs                   216124        72    216052   0% /var/volatile
tmpfs                   216124         0    216124   0% /usr/local
ubi7:pnor-rw-729a501a
                         13816       252     12820   2% /media/pnor-rw-729a501a
ubi7:pnor-rw-73731af2
                         13816       284     12788   2% /media/pnor-rw-73731af2
ubi7:pnor-prsv             972        32       872   4% /media/pnor-prsv
/dev/ubiblock7_4         20224     20224         0 100% /media/pnor-ro-729a501a
ubi7:pnor-patch          13816        16     13056   0% /usr/local/share/pnor
root@witherspoon:~#
```

`enumerate`
```
/xyz/openbmc_project/software/enumerate
{
    "/xyz/openbmc_project/software/729a501a": {
        "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
        "ExtendedVersion": "op-build-v1.7-1321-g53c41e3-dirty,buildroot-2017.05-163-g8035745,skiboot-v5.8-rc1-p2f3f436,hostboot-c68be97,linux-4.12.8-openpower1-p5e99141,petitboot-v1.5.1-p6b4d24f,machine-xml-c1e49a6-p1c7cc59,occ-854999a,hostboot-binaries-67a15fd,capp-ucode-9c73e9f,sbe-4a62126-pd206ccb",
        "Path": "",
        "Priority": 0,
        "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.Host",
        "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
        "Version": "IBM-witherspoon-redbud-ibm-OP9_v1.19_1.2",
        "associations": [
            [
                "inventory",
                "activation",
                "/xyz/openbmc_project/inventory/system/chassis"
            ]
        ]
    },
    "/xyz/openbmc_project/software/729a501a/inventory": {
        "endpoints": [
            "/xyz/openbmc_project/inventory/system/chassis"
        ]
    }
}
```

gkeishin, 2017-09-08:
- [x] Journald 
[build_100_software_error.txt](https://github.com/openbmc/openbmc/files/1288615/build_100_software_error.txt)

gkeishin, 2017-09-08:
- [x] We still need to support the old non-Witherspoon system to show up the version

saqibkh, 2017-09-11:
With the NEW BMC Code Update the entries in priority are wrong.

# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="0.1.0 (morty)"
VERSION_ID="0.1.0"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 (morty)"
BUILD_ID="20170803014838"

a0d9ba0d --> Set to priority 0
VERSION_ID="v1.99.9-143-g69cab69"

f3b29aa8 --> Set to priority 255                  <-------- This one should be set to 0
VERSION_ID="0.1.0"

rfrandse, 2017-09-18:
<https://gerrit.openbmc-project.xyz/6734> Restore BMC version on systems that doesn't have new BMC ubi layout.
Resolves  #2277 BMC version property missing in Software Manager object

closed 2017-10-04 by geissonator