openbmc/openbmc #2278: Need error log if fsi scan service fails to processors

geissonator, 2017-09-08:
During power on, the bmc software runs multiple services that attempt to scan the FSI hardware.  If you have a bad chip, then the scan can fail, and the power on will fail because of that.

BMC firmware needs to find a way to log an error if this happens so the user konws a fail occurred and what part needs to be replaced.

Here's an example on a system
```
Sep 08 16:26:05 witherspoon systemd[1]: Started Scan FSI devices.
Sep 08 16:26:05 witherspoon systemd[1]: Starting Scan FSI hub device...
Sep 08 16:26:05 witherspoon systemd[1]: fsi-hub-scan@0.service: Main process exited, code=exited, status=1/FAILURE
Sep 08 16:26:06 witherspoon sh[2342]: /bin/sh: /sys/devices/platform/gpio-fsi/fsi0/slave@00:00/00:00:00:0a/fsi1/rescan: No such file or directory
Sep 08 16:26:05 witherspoon systemd[1]: Failed to start Scan FSI hub device.
Sep 08 16:26:05 witherspoon systemd[1]: Dependency failed for Chassis0 (Power On).
Sep 08 16:26:05 witherspoon systemd[1]: Dependency failed for Start Host0.
Sep 08 16:26:05 witherspoon systemd[1]: obmc-host-start@0.target: Job obmc-host-start@0.target/start failed with result 'dependency'.
Sep 08 16:26:05 witherspoon systemd[1]: obmc-host-start@0.target: Triggering OnFailure= dependencies.
Sep 08 16:26:05 witherspoon systemd[1]: obmc-chassis-poweron@0.target: Job obmc-chassis-poweron@0.target/start failed with result 'dependency'.
Sep 08 16:26:05 witherspoon systemd[1]: obmc-chassis-poweron@0.target: Triggering OnFailure= dependencies.
Sep 08 16:26:06 witherspoon systemd[1]: fsi-hub-scan@0.service: Unit entered failed state.
Sep 08 16:26:06 witherspoon systemd[1]: fsi-hub-scan@0.service: Failed with result 'exit-code'.
```

spinler, 2017-09-14:
The failure above happens when the master scan has a problem and can't create that hub scan file.  So really the master processor needs to be called out.

After disussion in the design call, the following needs to be done:

1. Prepopulate the FSI master processor in the inventory so it can be called out on the genesis boot (for when MFG built a system with a bad proc).  Method TBD
2. Move the FSI scan and hub-scan calls from doing an echo  into a new code procedure in openpower-proc-control
3. Before running the hub scan, check for the existence of that file.  If it isn't there, create an error log and call out the master processor (can do this using the FSI path)
4. Modify phosphor-logging to handle the FSI path to do the callout.  Right now the callout only works for the path used for a get/putCFAM

spinler, 2017-09-14:
- [ ] Create a new openfsi/scan procedure in openpower-proc control
- [ ] Procedure will do hub and master scans.  will check for hub scan file after master scan and create appropriate error log to do a master processor callout if not there
- [ ] Remove the fsi-scan and fsi-hub-scan services from the 'host' bitbake dir and create a new fsi-scan service under op-host-control.  Run this service in the same place as the old fsi-scan.
- [ ] Copy the error log yaml out to phosphor-dbus-yaml so phosphor-logging can find it (was missed before)

rfrandse, 2017-09-19:
<https://gerrit.openbmc-project.xyz/6750> Use openpower-proc-control for fsi-scan
Resolves  #2278 Need error log if fsi scan service fails to processors

closed 2017-10-06 by geissonator