openbmc/openbmc #2285: openpower-occ-control reports error and tries to reboot occ on 2nd boot

geissonator, 2017-09-11:
A few things here, I loaded master v1.99.9-142-g73c92ac, on wsbmc011 and ran some tests.  First boot was fine, I then powered down and back up, I noticed the below in the journal.

May need a new issue for all the fan stuff below but we can ignore that for now.  The main issue was the openpower-occ-control app issues.  Looks like it detects an error and requests a reset for both occ's?  I think we are only supposed to request one reset (which will reset all occ's).  Also, I don't really think there was an occ error?  Wonder if we've got something happening with the disable/enable of occ in the IPL path.

2nd issue was that the reset request failed, this was early in the OPAL boot, I'm wondering if they even support this request during their boot, or if it's a runtime only thing.

```
Aug 28 19:43:41 witherspoon openpower-occ-control[1160]: Error file updated
Aug 28 19:43:41 witherspoon systemd[1]: Started Phosphor Hwmon Poller.
Aug 28 19:43:41 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@gpio\x2dfsi-cfam\x400\x2c0-sbefifo\x402400-occ\x401-occ\x2dhwmon\x401.service: Unit not needed anymore. Stopping.
Aug 28 19:43:41 witherspoon systemd[1]: Stopping Phosphor Hwmon Poller...
Aug 28 19:43:41 witherspoon systemd[1]: Stopped Phosphor Hwmon Poller.
Aug 28 19:43:43 witherspoon phosphor-fan-control[2307]: Failed to invoke DBus method.
Aug 28 19:43:43 witherspoon phosphor-fan-control[2307]: The operation failed internally.
Aug 28 19:43:43 witherspoon phosphor-fan-control[2307]: Unable to find property
Aug 28 19:43:44 witherspoon ipmid[1185]: Pushing cmd on to queue
Aug 28 19:43:44 witherspoon ipmid[1185]: Pushing cmd on to queue
Aug 28 19:43:44 witherspoon ipmid[1185]: Asserting SMS Attention
Aug 28 19:43:45 witherspoon phosphor-fan-control[2307]: Failed to invoke DBus method.
Aug 28 19:43:45 witherspoon phosphor-fan-control[2307]: The operation failed internally.
Aug 28 19:43:45 witherspoon phosphor-fan-control[2307]: Unable to find property
Aug 28 19:43:46 witherspoon ipmid[1185]: SMS Attention asserted
Aug 28 19:43:46 witherspoon phosphor-fan-control[2307]: Failed to invoke DBus method.
Aug 28 19:43:46 witherspoon phosphor-fan-control[2307]: The operation failed internally.
Aug 28 19:43:46 witherspoon phosphor-fan-control[2307]: Unable to find property
Aug 28 19:43:47 witherspoon phosphor-fan-control[2307]: Failed to invoke DBus method.
Aug 28 19:43:47 witherspoon phosphor-fan-control[2307]: The operation failed internally.
Aug 28 19:43:47 witherspoon phosphor-fan-control[2307]: Unable to find property
Aug 28 19:43:47 witherspoon phosphor-fan-control[2307]: Failed to invoke DBus method.
Aug 28 19:43:47 witherspoon phosphor-fan-control[2307]: The operation failed internally.
Aug 28 19:43:47 witherspoon phosphor-fan-control[2307]: Unable to find property
Aug 28 19:43:47 witherspoon phosphor-watchdog[1848]: watchdog: disabled
Aug 28 19:43:47 witherspoon phosphor-fan-control[2307]: Failed to invoke DBus method.
Aug 28 19:43:47 witherspoon phosphor-fan-control[2307]: The operation failed internally.
Aug 28 19:43:47 witherspoon phosphor-fan-control[2307]: Unable to find property
Aug 28 19:43:47 witherspoon phosphor-fan-control[2307]: Failed to invoke DBus method.
Aug 28 19:43:47 witherspoon phosphor-fan-control[2307]: The operation failed internally.
Aug 28 19:43:47 witherspoon phosphor-fan-control[2307]: Unable to find property
Aug 28 19:43:47 witherspoon phosphor-fan-control[2307]: Failed to invoke DBus method.
Aug 28 19:43:47 witherspoon phosphor-fan-control[2307]: The operation failed internally.
Aug 28 19:43:47 witherspoon phosphor-fan-control[2307]: Unable to find property
Aug 28 19:43:47 witherspoon phosphor-fan-control[2307]: Failed to invoke DBus method.
Aug 28 19:43:47 witherspoon phosphor-fan-control[2307]: The operation failed internally.
Aug 28 19:43:47 witherspoon phosphor-fan-control[2307]: Unable to find property
Aug 28 19:43:47 witherspoon phosphor-fan-control[2307]: Failed to invoke DBus method.
Aug 28 19:43:47 witherspoon phosphor-fan-control[2307]: The operation failed internally.
Aug 28 19:43:47 witherspoon phosphor-fan-control[2307]: Unable to find property
Aug 28 19:43:48 witherspoon phosphor-fan-control[2307]: Failed to invoke DBus method.
Aug 28 19:43:48 witherspoon phosphor-fan-control[2307]: The operation failed internally.
Aug 28 19:43:48 witherspoon phosphor-fan-control[2307]: Unable to find property
Aug 28 19:43:48 witherspoon phosphor-fan-control[2307]: Failed to invoke DBus method.
Aug 28 19:43:48 witherspoon phosphor-fan-control[2307]: The operation failed internally.
Aug 28 19:43:48 witherspoon phosphor-fan-control[2307]: Unable to find property
Aug 28 19:43:48 witherspoon phosphor-fan-control[2307]: Failed to invoke DBus method.
Aug 28 19:43:48 witherspoon phosphor-fan-control[2307]: The operation failed internally.
Aug 28 19:43:48 witherspoon phosphor-fan-control[2307]: Unable to find property
Aug 28 19:43:48 witherspoon phosphor-fan-control[2307]: Failed to invoke DBus method.
Aug 28 19:43:48 witherspoon phosphor-fan-control[2307]: The operation failed internally.
Aug 28 19:43:48 witherspoon phosphor-fan-control[2307]: Unable to find property
Aug 28 19:43:49 witherspoon ipmid[1185]: Host control timeout hit!
Aug 28 19:43:49 witherspoon ipmid[1185]: Timer expired
Aug 28 19:43:49 witherspoon openpower-occ-control[1160]: Host control signal values
Aug 28 19:43:49 witherspoon openpower-occ-control[1160]: Error resetting the OCC.
Aug 28 19:43:49 witherspoon openpower-occ-control[1160]: Host control signal values
Aug 28 19:43:49 witherspoon openpower-occ-control[1160]: Error resetting the OCC.
```

Priority 5 until we can figure out if this is just a one-off thing on wsbmc011 or pervasive in master.  We can't have the occ monitoring not work (and the occ potentially in safe mode) on other boots then the first one.

vishwabmc, 2017-09-12:
Couple things on how OCC error is handled : 

* On detecting the first error, : 
  - Stops the error watch on `all` the OCCs
  - Does a unbind on the error occ
  - Asks for OCC reset

Since the watch on other OCCs have already been stopped, we should not get into the `I reset OCC twice` condition. But I see that's how its shown at-least in that trace.
 
The only way I see this happening is when both the OCCs report error at the same time

journal meta would have helped here.

geissonator, 2017-09-12:
Please just put master on a system and recreate Vishwa.  I think George reported a similar issue.  May actually happen on the first boot.  Our primary focus right now is on #2228 but we'll need this fixed in parallel.

vishwabmc, 2017-09-12:
Okay... I put that comment in the morning before we discussed this in openbmc

gkeishin, 2017-09-12:
1. Load the Master 
2. boot to Host
3. r/r 

That should recreate this issue, either it will manifest the error reported here and one time we saw OCC core dumping..

gkeishin, 2017-09-13:
Recreated and system provided for debugging.

vishwabmc, 2017-09-20:
When I looked at this machine, saw that `-6` error was being reported even after binding all the OCCs. Left a message with @eddiejames

geissonator, 2017-10-05:
Eddie to take this one over with plan we made from discussion with LTC team last night

rfrandse, 2017-10-07:
<https://gerrit.openbmc-project.xyz/7118> Add OCC present count detection and watch
Resolves  #2285 openpower-occ-control reports error and tries to reboot occ on 2nd boot

rfrandse, 2017-11-18:
<https://gerrit.openbmc-project.xyz/7843> Re-enable OCC error watching
Resolves  #2285 openpower-occ-control reports error and tries to reboot occ on 2nd boot

closed 2017-12-13 by geissonator