openbmc/openbmc #2287: ubi0:rwfs: Incorrect behaviour of notify event on systemd-coredump created core files

ojayanth, 2017-09-12:
In the system which have ubi0:rwfs based /var/lib/systemd/coredump directory , noticed that existing inotify-event structure provide only core file “#” format instead of actual file name for IN_CLOSE_WRITE event watch.

Detail available in https://github.com/openbmc/openbmc/issues/2240.

ojayanth, 2020-07-27:
same issue seeing in the ext4 filsystem also , Need to investigate this issue.

chiragsibm, 2020-07-27:
On ext4 based systems, noticed that existing inotify-event structure notifies IN_CLOSE_WRITE instead of IN_CREATE; as a result of which core file is coming in “#<inode number>” format instead of the core actual file name.
Example trace:-
Jul 21 02:00:25 rainier phosphor-dump-monitor[601]: #262337

rsnair, 2020-08-04:
[image]

Table in the picture above captures the INOTIFY events generated for a systemd coredump with compression disabled & enabled. 

Looking at the ubifs & ext4 flow, it appears the systemd coredump binary is creating an unnamed temporary file via O_TMPFILE. Once all the data pertaining to the core is written, it does an O_CREAT to associate a name to the temporary file.

From manpage of open, 
>    O_TMPFILE (since Linux 3.11)
>               Create an unnamed temporary regular file.  The pathname argu‐
>               ment specifies a directory; an unnamed inode will be created
>               in that directory's filesystem.  Anything written to the
>               resulting file will be lost when the last file descriptor is
>               closed, unless the file is given a name.
>               <snip>
>               <snip>
>              O_TMPFILE requires support by the underlying filesystem; only
>               a subset of Linux filesystems provide that support.  In the
>               initial implementation, support was provided in the ext2,
>               ext3, ext4, UDF, Minix, and shmem filesystems.  Support for
>               other filesystems has subsequently been added as follows: XFS
>               (Linux 3.15); Btrfs (Linux 3.16); F2FS (Linux 3.16); and ubifs
>               (Linux 4.9)

jffs2 probably doesn't implement this and the code falls back to do a regular named temporary file create via O_CREAT, perform the necessary writes and finally, issues *mv* operation to rename the temporary file to the standard **core.<pattern>** name.

Based on the analysis, a proposed solution would be to use IN_CREATE as the default implementation for systemd coredump. For the outlier in jffs2, a workaround can be added to watch IN_CLOSE_WRITE.

Another recommendation is to disable the compression of coredump by default. Enabling the same, creates additional overhead in terms of event generation and the associated time required for compression - but adds no value as the coredump file is anyway compressed in a subsequent step during BMC dump creation.  
> /etc/systemd/coredump.conf
> Compress=no

ojayanth, 2020-08-05:
Good findings, and thanks for the detailed updates.

Existing coredump watch code already ignoring temporary file inotify signals.
https://github.com/openbmc/phosphor-debug-collector/blob/master/core_manager.cpp#L30
   /*
          As per coredump source code systemd-coredump uses below format
          https://github.com/systemd/systemd/blob/master/src/coredump/coredump.c
          /var/lib/systemd/coredump/core.%s.%s." SD_ID128_FORMAT_STR “
          systemd-coredump also creates temporary file in core file path prior
          to actual core file creation. Checking the file name format will help
          to limit dump creation only for the new core files.
        */

Can you add the File name/inode information during IN_CREATE and IN_CLOSE_WRITE signal user data.

ojayanth, 2020-08-05:
Regarding , Disable compression in systemd-core dump : This default behaviour enabled to support low flash size ( 32MB) support systems.   If required system owners can control this config based on flash size.

rsnair, 2020-08-05:
> Can you add the File name/inode information during IN_CREATE and IN_CLOSE_WRITE signal user data.

For ext4 & ubifs as the O_TMPFILE is implemented, the IN_CREATE is invoked only towards the end of writes. Hence you will get the actual core file name as part of the inotify event structure. 

In case of jffs2, due to the limitation of not having O_TMPFILE, the systemd-coredump code falls back to creating a named temporary file via O_CREAT. But this file is eventually renamed to the actual core.xxxx and hence the filename from IN_CREATE inotify event cannot be used. You should instead use the IN_CLOSE_WRITE or IN_MOVED_TO for the jffs2 case.

For additional clarity also sharing the snippet code from sytemd-coredump -

> **File:** **systemd/src/coredump/coredump.c**
>  325 static int save_external_coredump(
>  ...
>  ...
>  379         fd = open_tmpfile_linkable(fn, O_RDWR|O_CLOEXEC, &tmp);
> 
> 
> **File: systemd/src/basic/tmpfile-util.c**
> 263         **fd = open_parent(target, O_TMPFILE|flags, 0640);**
> 264         if (fd >= 0) {
> 265                 *ret_path = NULL;
> 266                 return fd;
> 267         }
> 268
> 269         log_debug_errno(fd, "Failed to use O_TMPFILE for %s: %m", target);
> 270
> 271         r = tempfn_random(target, NULL, &tmp);
> 272         if (r < 0)
> 273                 return r;
> 274
> 275         **fd = open(tmp, O_CREAT|O_EXCL|O_NOFOLLOW|O_NOCTTY|flags, 0640);**
> 276         if (fd < 0)
> 277                 return -errno;

geissonator, 2020-08-05:
> Based on the analysis, a proposed solution would be to use IN_CREATE as the default implementation for systemd coredump. For the outlier in jffs2, a workaround can be added to watch IN_CLOSE_WRITE.

What would this workaround look like? A compile flag option passed in by the bitbake recipe?

rsnair, 2020-08-05:
Right, similar to what was done earlier to add the ubifs workaround. Also, get rid of the ubifs workaround flags as that's probably not required anymore.

ojayanth, 2020-08-05:
Since  same behaviors is seeing in  ext4/ubifs am ok with changing default behavior to IN_CREATE.   We should be forcing this workaround ( changing to IN_CLOSE_WRITE)  for jffs2  based systems.

closed 2021-07-08 by stale[bot]