openbmc/openbmc #2288: Get to root cause for pci_reset_outs causing bmc network failure on witherspoon

geissonator, 2017-09-12:
During investigation of bug #2228 it was found that https://gerrit.openbmc-project.xyz/#/c/6158/ was root cause for the issue.  For now we're going to revert it.  This issue is to get to root cause for why it caused this problem on witherspoon.

geissonator, 2017-09-12:
Basically the code before this commit was causing us to do nothing with the PCI gpio's.  When this change went in on witherspoon, we started doing the following:
```
356        'pci_reset_outs': [
357            # net name, polarity, reset hold
358            ('BMC_VS1_PERST_N', False, False),
359            ('BMC_CP0_PERST_ENABLE_R', False, False),
360        ],
```

mdmillerii, 2017-09-12:
On many power8 systems the pci resets needed to be asserted until the processor was able to control it.  See issue #315 for details.

Regardless, this points to an issue where the broadcom is reacting to PERST for the PCI interface to affect the NCSI interface in some way that causes our stack to stop sending on the channel.

netdev watchdog transmit timeout says that the network stack has detected a packet has been scheduled for transmit for an extended period of time (5 seconds from memory).

Erich-Hauptli, 2017-09-12:
So the BCM5719 on this system should have the A chip revision which contains a fix that prevents the NCSI link from dropping during a PERST to the Broadcom controller, except for the first instance of PERST being released after an AC cycle.  On P8, we had an issue were PERST would be released when the P8 chip was not on, so we had to have workarounds that prevented this line from releasing until after we hit a certain point in the system IPL.  As I understand it the PERST releasing early issue should no longer be an issue for P9, so it should be directly driven without the extra wonky code.  Thus, the BMC should not ever need to gate this PERST line.  It should only be controlled through the P9 via Opal.  From the looking at the schematics, it seems this may not have been communicated to the card team and they left the complicated PERST gating logic in the designs.

Looking at the logic

BMC_VS1_PERST_N is there to allow us to gate the Virtual Swith 1 of the PLX chip, I suspect for now we should leave this high as well and let Opal control through the proc.
BMC_CP0_PERST_ENABLE_R I suspect should always remain high, to allow the P9 processor to directly control PERST.

If the BMC code does nothing with these GPIOs, that would have the effect of holding these lines high, which I think might be what you want.

mdmillerii, 2017-09-13:
I agree with the desire to leave the GPIOs alone.  The issue here is "why does transitioning perst cause the network tx to stop".   @sammyj @amboar Now that we have a recreate I hope we can find why the stack stops transmiting.  Note that perst will still cycle each time the host boots.

amboar, 2017-09-13:
Ping @sammj, @mdmillerii typo'ed

gtmills, 2018-10-12:
@geissonator Still a problem?

closed 2019-10-27 by stale[bot]