openbmc/openbmc #2291: No way to determine image ID you just uploded

cdeadmin, 2017-09-12:


stewart-ibm, 2017-09-26:
https://gerrit.openbmc-project.xyz/#/c/6758/ at least partially addresses this as you can calculate the image id.

anoo1, 2017-09-28:
**STATUS**
9/28 - Discussion on having the REST upload return the id are in progress, after the change Stewart mentions above is completed.

zahrens, 2017-11-13:
The software version id is a hash calculated over the version string of the image.

Some requested enhancements have been the ability for the user to be able to calculate
externally that id so that there's no need to enumerate the dbus object to find it after uploading an
image.

Also calculate the hash over perhaps the whole tar file so that the version id is different even if
the version string is the same, like in the case of dirty git commits.

gtmills, 2019-01-23:
The bug internal to IBM was closed. There is still a related issue here: https://github.com/openbmc/openbmc/issues/2290

closed 2019-01-23 by gtmills