openbmc/openbmc #2292: PrepareForUpdate: Support for redundant environment variables

anoo1, 2017-09-12:
With the change to add redundant environment variables, need to call fw_setenv twice to set the variable for PrepareForUpdate, because the initramfs does not call fw_getenv directly but instead has its own read function that doesn't read the redundant environment, so an easy workaround is to set the environment twice, and also in the clear-once.service call to clear the environment twice.

rfrandse, 2017-10-03:
<https://gerrit.openbmc-project.xyz/7043> clear-once.service: Call fw_setenv twice for openbmconce
Resolves  #2292 PrepareForUpdate: Support for redundant environment variables

anoo1, 2017-10-03:
Commits:
https://gerrit.openbmc-project.xyz/#/c/7041/ 
https://gerrit.openbmc-project.xyz/#/c/7042/ 
https://gerrit.openbmc-project.xyz/#/c/7043/

closed 2017-10-25 by geissonator