openbmc/openbmc #2293: xyz.openbmc_project.Network.IP.Create failed to restore network after new BMC code update

charleshofer, 2017-09-12:
```
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.9-151"
VERSION_ID="v1.99.9-151-g533d9f3"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.9-151"
BUILD_ID="v1.99.9"
```
Found here, http://ausgsa.ibm.com/~anoo/public/bmc-update/0911-2-obmc-phosphor-image-witherspoon.ubi.mtd.tar

Here's what I did:
1. Updated to the above image.
2. Set the MAC with `fw_setenv ethaddr 70:E2:84:14:20:30`
3. Rebooted the BMC
4. Attempted to restore network settings with `busctl call xyz.openbmc_project.Network /xyz/openbmc_project/network/eth0 xyz.openbmc_project.Network.IP.Create IP ssys xyz.openbmc_project.Network.IP.Protocol.IPv4 9.3.185.161 23 9.3.184.0`
5. Ran `systemctl restart systemd-networkd.service`

I cannot ping the BMC.

anoo1, 2017-09-13:
@ratagupt could you take a look?

charleshofer, 2017-09-13:
I noticed that the gateway setting in /etc/systemd/network/00-bmc-eth0.network was always blank after running the busctl command. Setting the gateway manually in that file seemed to fix things.

geissonator, 2017-09-13:
Phase 5 since can potentially block new bmc code update (network restore required)

ratagupt, 2017-09-14:
The one which is at IP create is network gateway..(for this network what is the gateway).
Might be you want to set the default gateway.

If you want to set the default gateway of the system.
busctl set-property xyz.openbmc_project.Network /xyz/openbmc_project/network/conf xyz.openbmc_project.Network.SystemConfiguration DefaultGateway s "<gatewayIP>"

ratagupt, 2017-09-14:
this is not a bug,Seems usage problem.

charleshofer, 2017-09-14:
Attempted to set the default gateway. Was a little unclear on the instructions, but tried this getting errors back.

```
root@witherspoon:~# busctl set-property xyz.openbmc_project.Network /xyz/openbmc_project/network/conf xyz.openbmc_project.Network.SystemConfiguration DefaultGateway 9.3.181.1
Failed to create bus message: Invalid argument
root@witherspoon:~# busctl set-property xyz.openbmc_project.Network /xyz/openbmc_project/network/conf xyz.openbmc_project.Network.SystemConfiguration DefaultGateway s 9.3.181.1
Unknown object '/xyz/openbmc_project/network/conf'.
```

ratagupt, 2017-09-15:
Ahh,seems the dbus path is config then conf..
```
GET
root@witherspoon:~# busctl get-property xyz.openbmc_project.Network /xyz/openbmc_project/network/config xyz.openbmc_project.Network.SystemConfiguration DefaultGateway
s "9.41.164.1"
SET
root@witherspoon:~# busctl set-property xyz.openbmc_project.Network /xyz/openbmc_project/network/config xyz.openbmc_project.Network.SystemConfiguration DefaultGateway s "9.41.164.1"
root@witherspoon:~# 
```

ratagupt, 2017-09-15:
Can you please close this defect?

closed 2017-09-15 by charleshofer