openbmc/openbmc #2294: ubi7 error: ubi_create_volume: not enough PEBs, only 339 available

geissonator, 2017-09-13:
Dean was doing some PNOR work, he updated a the PNOR on a BMC without a reboot ~20 times.  He eventually hit this failure and was unable to do any more PNOR updates.

```
Sep 13 01:34:21 witherspoon phosphor-gevent[7541]: ::80c9:c076:8e91:f402%1340386068 - - [2017-09-13 01:34:21] "POST /xyz/openbmc_project/software/a067fa4e/action/delete HTTP/1.1" 200 169 0.248339
Sep 13 01:34:26 witherspoon phosphor-gevent[7541]: ::80c9:c076:8e91:f402%1340386068 - - [2017-09-13 01:34:26] "POST /xyz/openbmc_project/software/a067fa4e/action/delete HTTP/1.1" 200 169 0.390900
Sep 13 01:34:26 witherspoon systemd[1]: Starting Remove UBIFS volume pnor-ro-a067fa4e...
Sep 13 01:34:26 witherspoon systemd[1]: Starting Remove UBIFS volume pnor-rw-a067fa4e...
Sep 13 01:34:27 witherspoon kernel: UBIFS (ubi7:1): un-mount UBI device 7
Sep 13 01:34:27 witherspoon kernel: UBIFS (ubi7:1): background thread "ubifs_bgt7_1" stops
Sep 13 01:34:27 witherspoon kernel[982]: [45458.280000] UBIFS (ubi7:1): un-mount UBI device 7
Sep 13 01:34:27 witherspoon kernel[982]: [45458.280000] UBIFS (ubi7:1): background thread "ubifs_bgt7_1" stops
Sep 13 01:34:27 witherspoon obmc-flash-bios[20837]: libubi: error!: cannot read "/sys/class/ubi/ubi7_0/type"
Sep 13 01:34:27 witherspoon obmc-flash-bios[20837]:         error 19 (No such device)
Sep 13 01:34:27 witherspoon obmc-flash-bios[20837]: ubirmvol: error!: cannot find UBI volume "pnor-rw-a067fa4e"
Sep 13 01:34:27 witherspoon obmc-flash-bios[20837]:           error 19 (No such device)
Sep 13 01:34:27 witherspoon systemd[1]: Started Remove UBIFS volume pnor-rw-a067fa4e.
Sep 13 01:34:28 witherspoon kernel: block ubiblock7_0: released
Sep 13 01:34:28 witherspoon kernel[982]: [45458.680000] block ubiblock7_0: released
Sep 13 01:34:28 witherspoon systemd[1]: Started Remove UBIFS volume pnor-ro-a067fa4e.
Sep 13 01:34:34 witherspoon phosphor-gevent[7541]: ::80c9:c076:8e91:f402%1340386068 - - [2017-09-13 01:34:34] "GET /xyz/openbmc_project/software/enumerate HTTP/1.1" 200 1265 0.473856
Sep 13 01:35:41 witherspoon dropbear[20870]: Failed loading /etc/dropbear/dropbear_dss_host_key
Sep 13 01:35:41 witherspoon systemd[1]: Started SSH Per-Connection Server (9.5.114.79:36665).
Sep 13 01:35:41 witherspoon dropbear[20870]: Failed loading /etc/dropbear/dropbear_ecdsa_host_key
Sep 13 01:35:41 witherspoon dropbear[20870]: Child connection from ::ffff:9.5.114.79:36665
Sep 13 01:35:47 witherspoon dropbear[20870]: PAM password auth succeeded for 'root' from ::ffff:9.5.114.79:36665
Sep 13 01:35:55 witherspoon phosphor-version-software-manager[1123]: Untaring
Sep 13 01:35:55 witherspoon dropbear[20870]: Exit (root): Disconnect received
Sep 13 01:35:56 witherspoon openpower-update-manager[1118]: Error in reading file
Sep 13 01:35:56 witherspoon openpower-update-manager[1118]: Software Object with the same version already exists
Sep 13 01:37:05 witherspoon phosphor-gevent[7541]: ::80c9:c076:8e91:f402%1340386068 - - [2017-09-13 01:37:05] "GET /xyz/openbmc_project/software/enumerate HTTP/1.1" 200 1938 0.178676
Sep 13 01:37:50 witherspoon systemd[1]: Starting Mount SquashFS image pnor-ro-4fdd83...
Sep 13 01:37:50 witherspoon systemd[1]: Starting Mount UBIFS volume pnor-prsv and pnor-rw-4fdd83...
Sep 13 01:37:51 witherspoon obmc-flash-bios[20877]: Volume ID 0, size 385 LEBs (25182080 bytes, 24.0 MiB), LEB size 65408 bytes (63.9 KiB), static, name "pnor-ro-4fdd83", alignment 1
Sep 13 01:37:51 witherspoon obmc-flash-bios[20879]: ubimkvol: error!: cannot UBI create volume
Sep 13 01:37:51 witherspoon obmc-flash-bios[20879]:           error 28 (No space left on device)
Sep 13 01:37:51 witherspoon kernel: ubi7 error: ubi_create_volume: not enough PEBs, only 211 available
Sep 13 01:37:51 witherspoon kernel: ubi7 error: ubi_create_volume: cannot create volume 7, error -28
Sep 13 01:37:51 witherspoon kernel[982]: [45662.500000] ubi7 error: ubi_create_volume: not enough PEBs, only 211 available
Sep 13 01:37:51 witherspoon kernel[982]: [45662.500000] ubi7 error: ubi_create_volume: cannot create volume 7, error -28
Sep 13 01:37:51 witherspoon obmc-flash-bios[20879]: mount: unknown filesystem type 'ubifs'
Sep 13 01:37:51 witherspoon obmc-flash-bios[20879]: /usr/sbin/obmc-flash-bios: error 32
Sep 13 01:37:51 witherspoon systemd[1]: obmc-flash-bios-ubimount@4fdd83.service: Main process exited, code=exited, status=32/n/a
Sep 13 01:37:51 witherspoon systemd[1]: Failed to start Mount UBIFS volume pnor-prsv and pnor-rw-4fdd83.
Sep 13 01:37:51 witherspoon systemd[1]: obmc-flash-bios-ubimount@4fdd83.service: Unit entered failed state.
Sep 13 01:37:51 witherspoon systemd[1]: obmc-flash-bios-ubimount@4fdd83.service: Failed with result 'exit-code'.
Sep 13 01:37:59 witherspoon phosphor-gevent[7541]: ::80c9:c076:8e91:f402%1340386068 - - [2017-09-13 01:37:59] "GET /xyz/openbmc_project/software/enumerate HTTP/1.1" 200 1941 0.563152
Sep 13 01:38:27 witherspoon kernel: UBIFS error (pid: 20901): cannot open "ubi7:pnor-rw-4fdd83", error -19
Sep 13 01:38:27 witherspoon kernel: block ubiblock7_0: created from ubi7:0(pnor-ro-4fdd83)
Sep 13 01:38:27 witherspoon kernel[982]: [45662.640000] UBIFS error (pid: 20901): cannot open "ubi7:pnor-rw-4fdd83", error -19
Sep 13 01:38:27 witherspoon kernel[982]: [45697.890000] block ubiblock7_0: created from ubi7:0(pnor-ro-4fdd83)
Sep 13 01:38:27 witherspoon systemd[1]: Started Mount SquashFS image pnor-ro-4fdd83.
Sep 13 01:38:27 witherspoon systemd[1]: org.open_power.Software.Host.Updater.service: Main process exited, code=dumped, status=11/SEGV
Sep 13 01:38:27 witherspoon systemd[1]: org.open_power.Software.Host.Updater.service: Unit entered failed state.
Sep 13 01:38:27 witherspoon systemd[1]: org.open_power.Software.Host.Updater.service: Failed with result 'core-dump'.
Sep 13 01:38:28 witherspoon systemd[1]: org.open_power.Software.Host.Updater.service: Service hold-off time over, scheduling restart.
Sep 13 01:38:28 witherspoon systemd[1]: Stopped OpenPower Software Update Manager.
Sep 13 01:38:28 witherspoon systemd[1]: Starting OpenPower Software Update Manager...
Sep 13 01:38:28 witherspoon openpower-update-manager[20918]: Error in reading file
Sep 13 01:38:28 witherspoon openpower-update-manager[20918]: Unable to restore priority from file.
Sep 13 01:38:28 witherspoon openpower-update-manager[20918]: Error in reading file
Sep 13 01:38:28 witherspoon systemd[1]: Started OpenPower Software Update Manager.
Sep 13 01:38:28 witherspoon systemd[1]: Starting Create a ubi volume for pnor patches...
Sep 13 01:38:29 witherspoon systemd[1]: Started Create a ubi volume for pnor patches.
Sep 13 01:38:42 witherspoon phosphor-gevent[7541]: ::80c9:c076:8e91:f402%1340386068 - - [2017-09-13 01:38:42] "GET /xyz/openbmc_project/software/enumerate HTTP/1.1" 200 1963 0.190374
Sep 13 01:39:14 witherspoon phosphor-gevent[7541]: ::80c9:c076:8e91:f402%1340386068 - - [2017-09-13 01:39:14] "POST /xyz/openbmc_project/software/4fdd83/action/delete HTTP/1.1" 200 169 0.298466
Sep 13 01:39:14 witherspoon systemd[1]: Starting Remove UBIFS volume pnor-ro-4fdd83...
Sep 13 01:39:14 witherspoon systemd[1]: Starting Remove UBIFS volume pnor-rw-4fdd83...
Sep 13 01:39:15 witherspoon systemd[1]: Started Remove UBIFS volume pnor-rw-4fdd83.
Sep 13 01:39:15 witherspoon kernel: block ubiblock7_0: released
Sep 13 01:39:15 witherspoon kernel[982]: [45746.020000] block ubiblock7_0: released
[... 117 lines cut ...]
Sep 13 01:46:25 witherspoon systemd[1]: org.open_power.Software.Host.Updater.service: Main process exited, code=dumped, status=11/SEGV
Sep 13 01:46:25 witherspoon systemd[1]: org.open_power.Software.Host.Updater.service: Unit entered failed state.
Sep 13 01:46:25 witherspoon systemd[1]: org.open_power.Software.Host.Updater.service: Failed with result 'core-dump'.
Sep 13 01:46:25 witherspoon systemd[1]: Starting Create a ubi volume for pnor patches...
Sep 13 01:46:25 witherspoon systemd[1]: org.open_power.Software.Host.Updater.service: Service hold-off time over, scheduling restart.
Sep 13 01:46:25 witherspoon systemd[1]: Stopped OpenPower Software Update Manager.
Sep 13 01:46:25 witherspoon systemd[1]: Starting OpenPower Software Update Manager...
Sep 13 01:46:25 witherspoon openpower-update-manager[21105]: Failed to read pnorTOC
Sep 13 01:46:25 witherspoon openpower-update-manager[21105]: Error in reading file
Sep 13 01:46:25 witherspoon openpower-update-manager[21105]: Failed to read version from pnorTOC
[... cut ...]
Sep 13 01:46:25 witherspoon openpower-update-manager[21105]: Error in reading file
Sep 13 01:46:25 witherspoon systemd[1]: Started OpenPower Software Update Manager.
Sep 13 01:46:25 witherspoon systemd[1]: Started Create a ubi volume for pnor patches.
```

```
[witherspoon]# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v1.99.9"
VERSION_ID="ibm-v1.99.9-0-r4-0-gf0746a0"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v1.99.9"
BUILD_ID="ibm-v1.99.9-0-r4"
```

geissonator, 2017-09-13:
```
[witherspoon]# ls /media/
pnor-prsv         pnor-ro-5f4ebc63  pnor-ro-8764baa3  pnor-rw-5f4ebc63  pnor-rw-8764baa3  pnor-rw-a067fa4e  pnor-rw-a5fe0e91
```

A bmc reboot and reflash did not fix this issue

geissonator, 2017-09-13:
We corrupted /dev/mtd7 (corrupted UBI in first 3 bytes) and did a BMC reboot thinking that would cause a clean up.  That caused a bunch of software services to core dump :(

I then manually ran ubiformat /dev/mtd7 and rebooted the BMC

We then got this when trying to load a known good image:
```
Sep 13 02:58:13 witherspoon systemd[1]: Starting Remove UBIFS volume pnor-rw-5f4ebc63...
Sep 13 02:58:13 witherspoon kernel[966]: [  282.050000] 00000000: 55 42 49 23 01 00 00 00 00 00 00 00 00 00 00 02 00 00 00 40 00 00 00 80 0b 50 77 78 00 00 00 00  UBI#...............@.....Pwx....
Sep 13 02:58:13 witherspoon systemd[1]: Starting Remove UBIFS volume pnor-ro-5f4ebc63...
Sep 13 02:58:13 witherspoon kernel[966]: [  282.050000] 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 5f a7 06 04  ............................_...
Sep 13 02:58:13 witherspoon kernel[966]: [  282.050000] ubi7 error: ubi_attach_mtd_dev: failed to attach mtd7, error -22
Sep 13 02:58:13 witherspoon obmc-flash-bios[1226]: grep: /sys/class/ubi/ubi?/subsystem/ubi*/name: No such file or directory
Sep 13 02:58:13 witherspoon systemd[1]: Started Remove UBIFS volume pnor-ro-5f4ebc63.
Sep 13 02:58:13 witherspoon obmc-flash-bios[1225]: grep: /sys/class/ubi/ubi?/subsystem/ubi*/name: No such file or directory
Sep 13 02:58:13 witherspoon systemd[1]: Started Remove UBIFS volume pnor-rw-5f4ebc63.
Sep 13 02:58:16 witherspoon phosphor-gevent[1214]: ::80f9:b676:8e91:f402%1340386068 - - [2017-09-13 02:58:16] "POST /xyz/openbmc_project/software/5f4ebc63/action/delete HTTP/1.1" 404 315 0.029897
Sep 13 02:58:24 witherspoon systemd[1]: Starting Attach the PNOR mtd device to UBI...
Sep 13 02:58:24 witherspoon phosphor-gevent[1214]: ::80f9:b676:8e91:f402%1340386068 - - [2017-09-13 02:58:24] "POST /xyz/openbmc_project/software/8764baa3/action/delete HTTP/1.1" 200 169 0.425774
Sep 13 02:58:24 witherspoon kernel: ubi7: default fastmap pool size: 100
Sep 13 02:58:24 witherspoon kernel: ubi7: default fastmap WL pool size: 50
Sep 13 02:58:24 witherspoon kernel: ubi7: attaching mtd7
Sep 13 02:58:24 witherspoon kernel[966]: [  293.240000] ubi7: default fastmap pool size: 100
Sep 13 02:58:24 witherspoon kernel[966]: [  293.240000] ubi7: default fastmap WL pool size: 50
Sep 13 02:58:24 witherspoon kernel[966]: [  293.240000] ubi7: attaching mtd7
Sep 13 02:58:24 witherspoon kernel: ubi7 error: scan_peb: bad image sequence number 189822840 in PEB 2, expected 1440815547
Sep 13 02:58:24 witherspoon kernel: Erase counter header dump:
Sep 13 02:58:24 witherspoon kernel:         magic          0x55424923
Sep 13 02:58:24 witherspoon kernel:         version        1
Sep 13 02:58:24 witherspoon kernel:         ec             2
Sep 13 02:58:24 witherspoon kernel:         vid_hdr_offset 64
Sep 13 02:58:24 witherspoon kernel:         data_offset    128
Sep 13 02:58:24 witherspoon kernel:         image_seq      189822840
Sep 13 02:58:24 witherspoon kernel:         hdr_crc        0x5fa70604
Sep 13 02:58:24 witherspoon kernel: erase counter header hexdump:
Sep 13 02:58:24 witherspoon kernel: 00000000: 55 42 49 23 01 00 00 00 00 00 00 00 00 00 00 02 00 00 00 40 00 00 00 80 0b 50 77 78 00 00 00 00  UBI#...............@.....Pwx....
Sep 13 02:58:24 witherspoon kernel: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 5f a7 06 04  ............................_...
Sep 13 02:58:24 witherspoon kernel: ubi7 error: ubi_attach_mtd_dev: failed to attach mtd7, error -22
Sep 13 02:58:24 witherspoon kernel[966]: [  293.310000] ubi7 error: scan_peb: bad image sequence number 189822840 in PEB 2, expected 1440815547
Sep 13 02:58:24 witherspoon kernel[966]: [  293.310000] Erase counter header dump:
Sep 13 02:58:24 witherspoon kernel[966]: [  293.310000]         magic          0x55424923
Sep 13 02:58:24 witherspoon kernel[966]: [  293.310000]         version        1
Sep 13 02:58:24 witherspoon kernel[966]: [  293.310000]         ec             2
Sep 13 02:58:24 witherspoon kernel[966]: [  293.310000]         vid_hdr_offset 64
Sep 13 02:58:24 witherspoon kernel[966]: [  293.310000]         data_offset    128
Sep 13 02:58:24 witherspoon kernel[966]: [  293.310000]         image_seq      189822840
Sep 13 02:58:24 witherspoon kernel[966]: [  293.390000]         hdr_crc        0x5fa70604
Sep 13 02:58:24 witherspoon kernel[966]: [  293.390000] erase counter header hexdump:
Sep 13 02:58:24 witherspoon kernel[966]: [  293.390000] 00000000: 55 42 49 23 01 00 00 00 00 00 00 00 00 00 00 02 00 00 00 40 00 00 00 80 0b 50 77 78 00 00 00 00  UBI#...............@.....Pwx....
Sep 13 02:58:24 witherspoon kernel[966]: [  293.390000] 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 5f a7 06 04  ............................_...
Sep 13 02:58:24 witherspoon kernel[966]: [  293.390000] ubi7 error: ubi_attach_mtd_dev: failed to attach mtd7, error -22
Sep 13 02:58:24 witherspoon systemd[1]: obmc-flash-bios-ubiattach.service: Main process exited, code=exited, status=255/n/a
Sep 13 02:58:24 witherspoon obmc-flash-bios[1249]: ubiattach: error!: cannot attach mtd7
Sep 13 02:58:24 witherspoon obmc-flash-bios[1249]:            error 22 (Invalid argument)
Sep 13 02:58:24 witherspoon systemd[1]: Failed to start Attach the PNOR mtd device to UBI.
Sep 13 02:58:24 witherspoon obmc-flash-bios[1249]: /usr/sbin/obmc-flash-bios: error 255
[... cut ...]
Sep 13 03:05:09 witherspoon phosphor-gevent[1214]: ::80f9:b676:8e91:f402%1340386068 - - [2017-09-13 03:05:09] "PUT /org/openbmc/control/flash/bmc/attr/preserve_network_settings HTTP/1.1" 200 169 0.297051
Sep 13 03:05:09 witherspoon phosphor-gevent[1214]: ::80f9:b676:8e91:f402%1340386068 - - [2017-09-13 03:05:09] "POST /login HTTP/1.1" 200 320 0.038467
Sep 13 03:05:09 witherspoon phosphor-gevent[1214]: ::80f9:b676:8e91:f402%1340386068 - - [2017-09-13 03:05:09] "GET /org/openbmc/control/flash/bmc HTTP/1.1" 200 380 0.088024
```

saqibkh, 2017-09-13:
This system already has three pnor versions "pnor-ro-5f4ebc63  pnor-ro-8764baa3  pnor-rw-5f4ebc63". The activation fails because there is no more space available for another version.

We recently added code to limit the number of active pnor versions to 2. 
If a user tries to upload/active 3rd version then the active version with highest priority gets deleted automatically to make space for the next version.

@geissonator we can close this issue.

geissonator, 2017-09-13:
```
andrews-mbp:openbmc-events andrewg$ python openbmc-sfw --server 9.5.166.110 list
/xyz/openbmc_project/software/5f4ebc63
{
    "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active", 
    "Priority": 0, 
    "Version": "open-power-witherspoon-v1.19-33-g491ddf0-dirty", 
    "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.Host", 
    "ExtendedVersion": "buildroot-2017.08-5-g08caaab,skiboot-v5.8-42-gbf1751cc6b4b,hostboot-ee53983,linux-4.13-openpower1-pcfc3a8f,petitboot-v1.5.1-p335166a,machine-xml-f8aaa73-pc78476e,occ-6f77ee7,hostboot-binaries-836385d,capp-ucode-9c73e9f,sbe-8d90ab2-pb50b7c9", 
    "Path": "", 
    "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None"
}
/xyz/openbmc_project/software/8de2cc42
{
    "Path": "", 
    "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active", 
    "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None", 
    "Version": "ibm-v1.99.9-0-r4-0-gf0746a0", 
    "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.BMC"
}
/xyz/openbmc_project/software/8764baa3
{
    "Activation": "xyz.openbmc_project.Software.Activation.Activations.Invalid", 
    "Version": "", 
    "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.Host", 
    "ExtendedVersion": "", 
    "Path": "", 
    "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None"
}
```

mdmillerii, 2017-09-13:
seems like we want a cleanup to force rw volumes based on existing ro ones?

geissonator, 2017-09-15:
Looks like this hit on our CI system last night as well:
```
Sep 14 18:49:48 witherspoon kernel[958]: [  201.260000] block ubiblock7_0: created from ubi7:0(pnor-ro-8764baa3)
Sep 14 18:49:49 witherspoon obmc-flash-bios[1431]: ubimkvol: error!: cannot UBI create volume
Sep 14 18:49:49 witherspoon obmc-flash-bios[1431]:           error 28 (No space left on device)
Sep 14 18:49:49 witherspoon obmc-flash-bios[1431]: mount: unknown filesystem type 'ubifs'
Sep 14 18:49:49 witherspoon obmc-flash-bios[1431]: /usr/sbin/obmc-flash-bios: error 32
Sep 14 18:49:49 witherspoon systemd[1]: obmc-flash-bios-ubimount@8764baa3.service: Main process exited, code=exited, status=32/n/a
Sep 14 18:49:49 witherspoon systemd[1]: Failed to start Mount UBIFS volumes pnor-ro-8764baa3, pnor-rw-8764baa3 and pnor-prsv.
Sep 14 18:49:49 witherspoon kernel: ubi7 error: ubi_create_volume: not enough PEBs, only 211 available
Sep 14 18:49:49 witherspoon kernel: ubi7 error: ubi_create_volume: cannot create volume 7, error -28
Sep 14 18:49:49 witherspoon kernel[958]: [  201.780000] ubi7 error: ubi_create_volume: not enough PEBs, only 211 available
Sep 14 18:49:49 witherspoon systemd[1]: obmc-flash-bios-ubimount@8764baa3.service: Unit entered failed state.
Sep 14 18:49:49 witherspoon kernel[958]: [  201.780000] ubi7 error: ubi_create_volume: cannot create volume 7, error -28
Sep 14 18:49:49 witherspoon systemd[1]: obmc-flash-bios-ubimount@8764baa3.service: Triggering OnFailure= dependencies.
Sep 14 18:49:49 witherspoon systemd[1]: obmc-flash-bios-ubimount@8764baa3.service: Failed with result 'exit-code'.
Sep 14 18:49:49 witherspoon systemd[1]: Starting Remove UBIFS volume pnor-rw-8764baa3...
Sep 14 18:49:49 witherspoon systemd[1]: Starting Remove UBIFS volume pnor-ro-8764baa3...
Sep 14 18:49:50 witherspoon systemd[1]: Started Remove UBIFS volume pnor-ro-8764baa3.
Sep 14 18:49:50 witherspoon kernel: UBIFS error (pid: 1442): cannot open "ubi7:pnor-rw-8764baa3", error -19
Sep 14 18:49:50 witherspoon kernel: block ubiblock7_0: released
```

CI does a delete of all PNOR images before running so there shouldn't have been anything on here.

geissonator, 2017-09-18:
@gkeishin will help us verify this doesn't recreate with latest code.

gkeishin, 2017-09-18:
Wrote a quick hack to do multiple update PNOR without reboot
```
${LOOP_COUNT}     ${12}
${IMAGE_FILE_PATH_1}  /esw/san5/gkeishin/PNOR/witherspoon.pnor.squashfs_GOOD.tar
${IMAGE_FILE_PATH_2}  /esw/san5/gkeishin/PNOR/witherspoon-sequoia.pnor.squashfs_LATEST.tar

*** Test Cases ***

Multiple Host Code Update Without Reboot
    [Documentation]  Do a PNOR code update by uploading image on BMC via REST
    ...  for multiple iteration without rebooting post update.

    Repeat Keyword  ${LOOP_COUNT} times  Delete PNOR And Update

*** Keywords ***

Delete PNOR And Update
    [Documentation]  Delete PNOR and activate new PNOR image.
    Delete All PNOR Images
    Upload And Activate Image  ${IMAGE_FILE_PATH_2}
    Sleep  15 s
    Delete All PNOR Images
    Upload And Activate Image  ${IMAGE_FILE_PATH_1}
```

This is without poweron..  Trying out first with r4 build
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v1.99.9"
VERSION_ID="ibm-v1.99.9-0-r4-0-gf0746a0"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v1.99.9"
BUILD_ID="ibm-v1.99.9-0-r4"
root@witherspoon:~#
```

gkeishin, 2017-09-18:
The above test went good.. now running with power on and off post update with sample test hack

```
*** Test Cases ***

Multiple Host Code Update Without Reboot
    [Documentation]  Do a PNOR code update by uploading image on BMC via REST
    ...  for multiple iteration without rebooting post update.

    Repeat Keyword  ${LOOP_COUNT} times  Delete PNOR And Update

*** Keywords ***

Delete PNOR And Update
    [Documentation]  Delete PNOR and activate new PNOR image.
    Delete All PNOR Images
    Upload And Activate Image  ${IMAGE_FILE_PATH_2}
    Initiate Host Boot
    Initiate Host PowerOff
    Delete All Sessions
    Sleep  15 s
    Delete All PNOR Images
    Upload And Activate Image  ${IMAGE_FILE_PATH_1}
    Delete All Sessions
    Initiate Host Boot
    Initiate Host PowerOff
```

gkeishin, 2017-09-18:
Looked good. I didn't run 20 IPL but 15 round of it...I had to stop to run other tests. but probably there are other issues with the delta which we are not aware where Dean was doing like patching PNOR as part of it..

geissonator, 2017-09-18:
Closing this one out based on George testing

closed 2017-09-18 by geissonator