openbmc/openbmc #2295: mboxd operation failed internally

gkeishin, 2017-09-13:
```
Aug 28 19:35:08 witherspoon mboxd[1164]: The operation failed internally.
Aug 28 19:35:18 witherspoon phosphor-log-manager[958]: Failed to find metadata
Aug 28 19:35:18 witherspoon phosphor-fru-fault-monitor[1140]: /xyz/openbmc_project/logging/entry/1

Entry : /xyz/openbmc_project/logging/entry/1
{
    "AdditionalData": [
        "_PID=1164"
    ],
    "Id": 1,
    "Message": "xyz.openbmc_project.Common.Error.InternalFailure",
    "Resolved": 0,
    "Severity": "xyz.openbmc_project.Logging.Entry.Level.Error",
    "Timestamp": 1503948918429,
    "associations": []
}

```

`Pid info`
```
{
        "__CURSOR" : "s=0482b1f029404728b7f6f78c67da7a39;i=2c2;b=6b4e9f75ff44485fa7d4770946350fe3;m=561f177;t=557d5646d78c8;x=e39dafc490c5622c",
        "__REALTIME_TIMESTAMP" : "1503948908099784",
        "__MONOTONIC_TIMESTAMP" : "90304887",
        "_BOOT_ID" : "6b4e9f75ff44485fa7d4770946350fe3",
        "_MACHINE_ID" : "d77bdfede38e4991bae5bd832d415646",
        "_HOSTNAME" : "witherspoon",
        "_UID" : "0",
        "_GID" : "0",
        "_SYSTEMD_SLICE" : "system.slice",
        "_TRANSPORT" : "journal",
        "_CAP_EFFECTIVE" : "3fffffffff",
        "PRIORITY" : "3",
        "CODE_FILE" : "/usr/include/phosphor-logging/log.hpp",
        "CODE_LINE" : "100",
        "CODE_FUNC" : "helper_log",
        "MESSAGE" : "The operation failed internally.",
        "TRANSACTION_ID" : "3394759793",
        "SYSLOG_IDENTIFIER" : "mboxd",
        "_PID" : "1164",
        "_COMM" : "mboxd",
        "_EXE" : "/usr/sbin/mboxd",
        "_CMDLINE" : "mboxd --flash 64M --window-size 1M",
        "_SYSTEMD_CGROUP" : "/system.slice/mboxd.service",
        "_SYSTEMD_UNIT" : "mboxd.service",
        "_SYSTEMD_INVOCATION_ID" : "3a0a701ca500417496b796b7000b123b",
        "_SOURCE_REALTIME_TIMESTAMP" : "1503948908097382"
}
```

`FS info`
```
root@witherspoon:~# df
Filesystem           1K-blocks      Used Available Use% Mounted on
dev                     215208         0    215208   0% /dev
tmpfs                   216120     14328    201792   7% /run
/dev/mtdblock4           14848     14848         0 100% /run/initramfs/ro
/dev/mtdblock5            4096       304      3792   7% /run/initramfs/rw
cow                       4096       304      3792   7% /
tmpfs                   216120         0    216120   0% /dev/shm
tmpfs                   216120         0    216120   0% /sys/fs/cgroup
tmpfs                   216120         0    216120   0% /tmp
tmpfs                   216120        84    216036   0% /var/volatile
tmpfs                   216120         0    216120   0% /usr/local
tmpfs                   216120         0    216120   0% /var/lib/systemd/coredump
ubi7:pnor-rw-8764baa3
                         13816       256     12816   2% /media/pnor-rw-8764baa3
ubi7:pnor-rw-73731af2
                         13816       284     12788   2% /media/pnor-rw-73731af2
ubi7:pnor-prsv             972        32       872   4% /media/pnor-prsv
ubi7:pnor-rw-a5fe0e91
                         13816       284     12784   2% /media/pnor-rw-a5fe0e91
/dev/ubiblock7_5         20096     20096         0 100% /media/pnor-ro-8764baa3
ubi7:pnor-patch          13816        16     13056   0% /usr/local/share/pnor
root@witherspoon:~#
```
`Driver info`
```

root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.9-168"
VERSION_ID="v1.99.9-168-gc180a26"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.9-168"
BUILD_ID="v1.99.9"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-b95f53ea200a6573035a25ab6ddebe917635eeb6 #1 Wed Sep 13 02:20:54 UTC 2017 armv6l GNU/Linux
root@witherspoon:~# uptime
 19:38:11 up 4 min,  load average: 1.13, 2.04, 1.01
root@witherspoon:~# cat /etc/timestamp
20170913022303
root@witherspoon:~# cat /media/pnor-ro-8764baa3/VERSION
IBM-witherspoon-sequoia-ibm-OP9_v1.18_1.54
        op-build-v1.7-1321-g6eef393-dirty
        buildroot-2017.05-163-g8035745
        skiboot-v5.8-rc1-pd9d2012
        hostboot-c68be97
        linux-4.12.8-openpower1-p919e663
        petitboot-v1.5.1-p848cad7
        machine-xml-c1e49a6-pbb9f626
        occ-854999a
        hostboot-binaries-67a15fd
        capp-ucode-9c73e9f
        sbe-4a62126-p180ac2a
root@witherspoon:~#
```

- [x] Journald 
[journald_elog_on_standby.txt](https://github.com/openbmc/openbmc/files/1298069/journald_elog_on_standby.txt)

dkodihal, 2017-09-13:
Don't see anything useful in the journal from mboxd, apart from

Aug 28 19:35:08 witherspoon mboxd[1164]: The operation failed internally.

gkeishin, 2017-09-13:
Yeah.. It just have that entry and the errorlog, so added the pretty print for that elog. Turns out other folks also not able to get meaningful info out of it https://github.com/openbmc/openbmc/issues/2297

dkodihal, 2017-09-13:
@gkeishin so you see this only once after a BMC code update, and not on host boots?

gkeishin, 2017-09-13:
yeah.. for now, I saw only during post code update and not on boot...

gkeishin, 2017-09-13:
This is persistent and recreated on the second try, @dkodihal is taking a stab at it.

dkodihal, 2017-09-13:
@gkeishin @ratagupt I see the InternalFailure, but no other metadata in the journal. pnor_partition.cpp, and likely other files, have instances where they log InternalFailure with no other metadata. That makes debug impossible.

First thing to do @ratagupt is to make sure we, in a patch, add metadata to the journal with all InternalFailure elogs, and then recreate this.

One thing I noticed was that at BMC standby, I initially did not see the /var/lib/phosphor-software-manager directory. When is that and the syminks created @anoo1 or @saqibkh? On the way to BMC standby, or later?

saqibkh, 2017-09-13:
@dkodihal The symlinks are created on the way to HOST poweron.